v4l2loopback/v4l2loopback #288: modprobe: ERROR: could not insert 'v4l2loopback': Exec format error

unicornofdeath, 2020-04-20:
### Step 1: Read this

Thanks for your feedback. It is invaluable for making `v4l2loopback` a better
software.

To help us making the most of your feedback (so we can e.g. fix bugs more quickly), please make sure to provide the information requested in this template.
Also make sure to remove any non-relevant parts (so we can focus on the essential problem).

Please keep in mind that the development of `v4l2loopback` is done by volunteers.
They are spending their spare time to give you a hopfully nice product and to help you if you have troubles - for free.

Please look through the list of issues (*open* and **closed** ones alike), to see whether you problem has been reported before. Probably you can find a solution to your problem without having to create a new ticket.

#### Remove Cruft

Please *remove* these instructions (and other non-relevant information) from your report.
If your report looks like a copy of the template, it might get closed immediately.

#### Title
Please chose an apropriate title: "*does not work*" or "*found a bug*" are way too generic.
Try to find a one-liner that says what is not working (e.g. "*module fails to load*").

Also try not to use relative terms.
E.g. "*fails to build with latest kernel*" is bad, because the latest kernel at the time you create the bug report might not be the latest kernel when the problem is being worked on.

#### Accessibility
Sometimes pictures say more.
However, mostly they prevent the use of advanced tools (like "search" or "copy&paste").
And always they prevent people who don't use graphical browser to access the tracker from reading your content.
So, to make the web a better place, we ask you to post *text* rather than *screenshots of text* whenever feasible (pretty much always).

### Step 2: Describe your environment

  * `v4l2loopback` version: _____

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: _____

          uname -a

  * Distribution (+version): _____

          lsb_release -a
  
### Step 3: Describe the problem:

#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

  ```
  // TODO(you): code here to reproduce the problem
  ```

wick-ed, 2020-04-21:
Hi @unicornofdeath 

was the same for me (Ubuntu 18.04).
[This issue's comment](https://github.com/umlaeute/v4l2loopback/issues/260#issuecomment-598653025) gave me the solving hint.
Just did `make CC=/usr/bin/gcc` instead of a blank `make`.
After that the modprobe command worked. Using the loopback to stream from OBS. :)

unicornofdeath, 2020-04-21:
I have tried that and still get same error

umlaeute, 2020-04-22:
> I get this error

which one (probably the one from the title; but it makes a bug-report *much* easier to read if you don't have to jump through the page for each line)

>  I have tried to build from source I have download with apt-get I have tried everything and still get error

sorry, i don't understand this sentence.

> I get when installed with apt-get

this doesn't look bad (as in: there's no error here).

anyhow: *what* did you install with `apt-get`?

> I have tried that and still get same error

the point is you *must* use the same compiler as was used to build your kernel.
this *might* be `/usr/bin/gcc` (which is really just an alias to "some" compiler), it *might* be `/usr/bin/gcc-8` (which is a specific compiler, but there's no guarantee that this is the one used to build the kernel)?

you might also consider adding information (as per the [template](https://github.com/umlaeute/v4l2loopback/blob/master/.github/ISSUE_TEMPLATE/bug_report.md#step-2-describe-your-environment) which you seem to have ignoed), such as the kernel you are actually using, and which distribution.

closed 2020-04-22 by umlaeute