v4l2loopback/v4l2loopback #290: Compilation error

nskmda, 2020-04-21:
make delivers

`Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/nskmda/bin/virtual_cam/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-176-generic'
  CC [M]  /home/nskmda/bin/virtual_cam/v4l2loopback/v4l2loopback.o
/home/nskmda/bin/virtual_cam/v4l2loopback/v4l2loopback.c: In function ‘vidioc_querycap’:
/home/nskmda/bin/virtual_cam/v4l2loopback/v4l2loopback.c:701:11: error: ‘struct video_device’ has no member named ‘device_caps’
  dev->vdev->device_caps = cap->device_caps = cap->capabilities = capabilities;
           ^
/home/nskmda/bin/virtual_cam/v4l2loopback/v4l2loopback.c: In function ‘init_vdev’:
/home/nskmda/bin/virtual_cam/v4l2loopback/v4l2loopback.c:2088:6: error: ‘struct video_device’ has no member named ‘device_caps’
  vdev->device_caps  =
      ^
/home/nskmda/bin/virtual_cam/v4l2loopback/v4l2loopback.c:2092:6: error: ‘struct video_device’ has no member named ‘device_caps’
  vdev->device_caps  |= V4L2_CAP_VIDEO_M2M;
      ^
scripts/Makefile.build:291: recipe for target '/home/nskmda/bin/virtual_cam/v4l2loopback/v4l2loopback.o' failed
make[2]: *** [/home/nskmda/bin/virtual_cam/v4l2loopback/v4l2loopback.o] Error 1
Makefile:1457: recipe for target '_module_/home/nskmda/bin/virtual_cam/v4l2loopback' failed
make[1]: *** [_module_/home/nskmda/bin/virtual_cam/v4l2loopback] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-176-generic'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2
`
kernel 
```
uname -r
4.4.0-176-generic
```

headers seem 2 b installed:

`
nskmda@dmitry-desktop:~/bin/virtual_cam/v4l2loopback$ ls -l /usr/src/linux-headers-$(uname -r)
total 1244
drwxr-xr-x 3 root root    4096 Mar 20 12:47 arch
lrwxrwxrwx 1 root root      32 Feb 28 02:00 block -> ../linux-headers-4.4.0-176/block
lrwxrwxrwx 1 root root      32 Feb 28 02:00 certs -> ../linux-headers-4.4.0-176/certs
lrwxrwxrwx 1 root root      33 Feb 28 02:00 crypto -> ../linux-headers-4.4.0-176/crypto
lrwxrwxrwx 1 root root      40 Feb 28 02:00 Documentation -> ../linux-headers-4.4.0-176/Documentation
lrwxrwxrwx 1 root root      34 Feb 28 02:00 drivers -> ../linux-headers-4.4.0-176/drivers
lrwxrwxrwx 1 root root      35 Feb 28 02:00 firmware -> ../linux-headers-4.4.0-176/firmware
lrwxrwxrwx 1 root root      29 Feb 28 02:00 fs -> ../linux-headers-4.4.0-176/fs
drwxr-xr-x 4 root root    4096 Mar 20 12:47 include
lrwxrwxrwx 1 root root      31 Feb 28 02:00 init -> ../linux-headers-4.4.0-176/init
lrwxrwxrwx 1 root root      30 Feb 28 02:00 ipc -> ../linux-headers-4.4.0-176/ipc
lrwxrwxrwx 1 root root      33 Feb 28 02:00 Kbuild -> ../linux-headers-4.4.0-176/Kbuild
lrwxrwxrwx 1 root root      34 Feb 28 02:00 Kconfig -> ../linux-headers-4.4.0-176/Kconfig
drwxr-xr-x 2 root root    4096 Mar 20 12:47 kernel
lrwxrwxrwx 1 root root      30 Feb 28 02:00 lib -> ../linux-headers-4.4.0-176/lib
lrwxrwxrwx 1 root root      35 Feb 28 02:00 Makefile -> ../linux-headers-4.4.0-176/Makefile
lrwxrwxrwx 1 root root      29 Feb 28 02:00 mm -> ../linux-headers-4.4.0-176/mm
-rw-r--r-- 1 root root 1254673 Feb 28 02:01 Module.symvers
lrwxrwxrwx 1 root root      30 Feb 28 02:00 net -> ../linux-headers-4.4.0-176/net
lrwxrwxrwx 1 root root      34 Feb 28 02:00 samples -> ../linux-headers-4.4.0-176/samples
drwxr-xr-x 7 root root    4096 Mar 20 12:47 scripts
lrwxrwxrwx 1 root root      35 Feb 28 02:00 security -> ../linux-headers-4.4.0-176/security
lrwxrwxrwx 1 root root      32 Feb 28 02:00 sound -> ../linux-headers-4.4.0-176/sound
lrwxrwxrwx 1 root root      30 Feb 28 02:00 spl -> ../linux-headers-4.4.0-176/spl
lrwxrwxrwx 1 root root      32 Feb 28 02:00 tools -> ../linux-headers-4.4.0-176/tools
lrwxrwxrwx 1 root root      33 Feb 28 02:00 ubuntu -> ../linux-headers-4.4.0-176/ubuntu
lrwxrwxrwx 1 root root      30 Feb 28 02:00 usr -> ../linux-headers-4.4.0-176/usr
lrwxrwxrwx 1 root root      31 Feb 28 02:00 virt -> ../linux-headers-4.4.0-176/virt
lrwxrwxrwx 1 root root      30 Feb 28 02:00 zfs -> ../linux-headers-4.4.0-176/zfs
`

nskmda, 2020-04-21:
I tried both - the master & release branches - with the same result.

umlaeute, 2020-04-22:
which "release branches"? (there are only release-tags; and since you used the plural i wonder which you were trying)

closed 2020-04-22 by umlaeute

nskmda, 2020-04-22:
v0.12.15.
sorry, my bad. release tag...

nskmda, 2020-04-22:
here's my system info: 

`Linux ***** 4.4.0-176-generic #206-Ubuntu SMP Fri Feb 28 05:02:04 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`