v4l2loopback/v4l2loopback #293: segmentation fault

blshkv, 2020-04-23:
Gentoo linux with kernel version: 5.5.19
v4l2loopback-0.12.5

```6 R12: 00005565733e5390
[   47.813563] R13: 0000000000000000 R14: 0000556574b3a450 R15: 0000000000000000
[   47.813565] ---[ end trace 00ec8fa657f50351 ]---
[   47.813568] v4l2loopback: failed video_register_device()
[   47.813573] general protection fault: 0000 [#1] PREEMPT SMP PTI
[   47.813576] CPU: 1 PID: 2819 Comm: modprobe Tainted: G        W  O    T 5.5.19-pentoo #1
[   47.813577] Hardware name: LENOVO 2330A54/2330A54, BIOS G2ETB7WW (2.77 ) 09/24/2019
[   47.813581] RIP: 0010:kernfs_find_ns+0xf/0xc0
[   47.813583] Code: 0f 0b e9 6d ff ff ff e8 0f c5 df ff 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 41 55 48 85 d2 49 89 f5 41 54 0f 95 c1 55 53 <0f> b7 47 70 48 89 d3 66 83 e0 20 0f 95 c2 38 d1 75 56 4c 8b 67 48
[   47.813584] RSP: 0018:ffffa92d01c87bb8 EFLAGS: 00010246
[   47.813585] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[   47.813586] RDX: 0000000000000000 RSI: ffffffffc04edaa4 RDI: 6b6b6b6b6b6b6b6b
[   47.813587] RBP: 6b6b6b6b6b6b6b6b R08: 0000000000000338 R09: 0000000000000001
[   47.813588] R10: 0000000000000000 R11: 0000000000000001 R12: ffffffffc04edaa4
[   47.813589] R13: ffffffffc04edaa4 R14: b49799d70c3f781e R15: 0000000000000000
[   47.813591] FS:  00007fab7c8ad740(0000) GS:ffff9dce96240000(0000) knlGS:0000000000000000
[   47.813592] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   47.813593] CR2: 00007fffdff276f0 CR3: 00000001e470a005 CR4: 00000000001606e0
[   47.813594] Call Trace:
[   47.813596]  kernfs_remove_by_name_ns+0x2d/0x90
[   47.813598]  init_module+0x206a89/0x20709a [v4l2loopback]
[   47.813601]  init_module+0xaa6/0x1000 [v4l2loopback]
[   47.813603]  ? 0xffffffffc02e5000
[   47.813604]  do_one_initcall+0x54/0x230
[   47.813606]  do_init_module+0x59/0x210
[   47.813608]  load_module+0x254e/0x2830
[   47.813611]  ? __do_sys_finit_module+0xbb/0x120
[   47.813613]  __do_sys_finit_module+0xbb/0x120
[   47.813615]  do_syscall_64+0x43/0x110
[   47.813617]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   47.813618] RIP: 0033:0x7fab7c9af909
[   47.813620] Code: 00 00 00 75 05 48 83 c4 18 c3 e8 22 8b 01 00 66 90 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 57 95 0c 00 f7 d8 64 89 01 48
[   47.813621] RSP: 002b:00007fffdff2a718 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   47.813622] RAX: ffffffffffffffda RBX: 0000556574b3ac20 RCX: 00007fab7c9af909
[   47.813623] RDX: 0000000000000000 RSI: 00005565733e5390 RDI: 0000000000000003
[   47.813624] RBP: 0000000000040000 R08: 0000000000000000 R09: 0000000000000000
[   47.813625] R10: 0000000000000003 R11: 0000000000000246 R12: 00005565733e5390
[   47.813626] R13: 0000000000000000 R14: 0000556574b3a450 R15: 0000000000000000
[   47.813628] Modules linked in: v4l2loopback(O+) ctr ccm ipv6 nf_defrag_ipv6 usbhid uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videodev videobuf2_common dm_mod dax x86_pkg_temp_thermal kvm_intel iwldvm mac80211 kvm irqbypass efivars vboxnetadp(O) vboxnetflt(O) serio_raw i2c_i801 iwlwifi vboxdrv(O) sdhci_pci cqhci sdhci mmc_core cfg80211 xhci_pci e1000e ehci_pci ehci_hcd xhci_hcd thinkpad_acpi ledtrig_audio rfkill efivarfs
[   47.813640] ---[ end trace 00ec8fa657f50352 ]---
[   47.813643] RIP: 0010:kernfs_find_ns+0xf/0xc0
[   47.813644] Code: 0f 0b e9 6d ff ff ff e8 0f c5 df ff 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 41 55 48 85 d2 49 89 f5 41 54 0f 95 c1 55 53 <0f> b7 47 70 48 89 d3 66 83 e0 20 0f 95 c2 38 d1 75 56 4c 8b 67 48
[   47.813645] RSP: 0018:ffffa92d01c87bb8 EFLAGS: 00010246
[   47.813646] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[   47.813647] RDX: 0000000000000000 RSI: ffffffffc04edaa4 RDI: 6b6b6b6b6b6b6b6b
[   47.813648] RBP: 6b6b6b6b6b6b6b6b R08: 0000000000000338 R09: 0000000000000001
[   47.813649] R10: 0000000000000000 R11: 0000000000000001 R12: ffffffffc04edaa4
[   47.813650] R13: ffffffffc04edaa4 R14: b49799d70c3f781e R15: 0000000000000000
[   47.813652] FS:  00007fab7c8ad740(0000) GS:ffff9dce96240000(0000) knlGS:0000000000000000
[   47.813653] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   47.813654] CR2: 00007fffdff276f0 CR3: 00000001e470a005 CR4: 00000000001606e0

```

any ideas?

umlaeute, 2020-04-25:
duplicate.
fixed.

blshkv, 2020-04-26:
Can you point to duplicate and release a new version?

umlaeute, 2020-04-26:
hmm, actually #257 

i only realized that you were already using  `v0.12.5`. are you sure about this?

blshkv, 2020-04-26:
my bad, it was an error on my side. I really have installed v0.12.5 this time, and it does not crash.
Thanks!

closed 2020-04-26 by blshkv