v4l2loopback/v4l2loopback #296: modprobe: ERROR: could not insert 'v4l2loopback': Exec format error

unicornofdeath, 2020-04-26:
From `bash currentversion.sh ` I get `0.12.5`

Kernel version is `Linux rick-pc 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u2 (2019-08-08) x86_64 GNU/Linux`

Distribution info
```
No LSB modules are available.
Distributor ID:	PureOS
Description:	PureOS
Release:	9.0
Codename:	amber
```

I have installed v4l2loopback with these commands
```
git clone https://github.com/umlaeute/v4l2loopback.git
cd v4l2loopback
make
sudo make install
sudo depmod -a
```

Then I run `sudo modprobe v4l2loopback` and get `modprobe: ERROR: could not insert 'v4l2loopback': Exec format error`

From `sudo dmesg  | grep -i v4l2loopback` i get:
```
[  429.680391] v4l2loopback: loading out-of-tree module taints kernel.
[  429.680424] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
```
Module is installed in `/lib/modules/4.19.0-5-amd64/extra`
And I have installed headers for kernel

umlaeute, 2020-04-26:
duplicate of #260, #238, #208, #147
i'm pretty sure you have either the headers for the *wrong* kernel, or the wrong toolchain for building kernel modules.

none of these problems are related to `v4l2loopback`. unfortunately i do not have the resources to debug any of them.

closed 2020-04-26 by umlaeute

unicornofdeath, 2020-04-26:
I have read all of those and tried them and still get this message

weichm, 2020-04-26:
Having the same problem. @umlaeute says this is not related to `v4l2loopback`. I think so, too. So we have to look elsewhere for a solution. To that aim, I posted a question on stackexchange: https://github.com/umlaeute/v4l2loopback/issues/296

kripton, 2020-04-26:
What usually helps here is to get the kernel sources (should be easy via apt in your case) and the kernel config (should also be easy from your currently running kernel). Them compile the kernel including v4l2loopback and you should be good. Because then you can be sure that the v4l2loopback matches the kernel sources and its all built with the same toolchain

unicornofdeath, 2020-04-26:
I have tried installing with apt and still get the same error

weichm, 2020-04-26:
I think I found the solution, see  https://unix.stackexchange.com/questions/582637/modprobe-error-could-not-insert-v4l2loopback-exec-format-error

The problem is that you need an unsigned kernel, if I am right. I would be glad if someone could comment on the need for an usigned kernel for this module.

weichm, 2020-04-26:
I think `v4l2loopback-dkms` is pretty useless, as long as the standard kernel is signed.

umlaeute, 2020-04-26:
it's as useless as any other `dkms` module.

if your system is configured to not allow compiling kernel-modules by the system administrator, then you cannot use `dkms`. this is totally unrelated to `v4l2loopback` (in the sense, that `v4l2loopback` like any other out-of-tree module cannot do anything about it)

having said that, i'm running Debian with a stock kernel (right now this is `linux-image-5.5.0-2-amd64`) which appears to be signed (as opposed to the (`linux-image-5.5.0-2-amd64-unsigned` kernel) , and I don't have any problems.

so i doubt that you are right when you say:
> that you need an unsigned kernel

weichm, 2020-04-26:
Thank your for the clarification. You are right, I do not have any problems running on `4.19.0-8-amd64` having installed `v4l2loopback-dkms`. Before it was `4.19.0-2-amd64` and it did not run. I don't know what the problem was, but I do not mind anymore.

unicornofdeath, 2020-04-28:
This worked for me 

https://unix.stackexchange.com/questions/582637/module-x86-modules-skipping-invalid-relocation-target

Here are commands
```
sudo apt-get install linux-image-4.19.0-6-amd64-unsigned linux-headers-4.19.0-6-amd64
cd v4l2loopback/
make clean
make
sudo make install
sudo depmod
sudo modprobe v4l2loopback
```

umlaeute, 2020-04-28:
@unicornofdeath wow: how does this differ from the [README](https://github.com/umlaeute/v4l2loopback)?

i'm seriously baffled.

weichm, 2020-04-28:
This line is different.
`sudo apt-get install linux-image-4.19.0-6-amd64-unsigned linux-headers-4.19.0-6-amd64`.
The commands given in README did not work for `4.19.0-2`, that's why I amended the issue.
Sorry for the inconvenience.

umlaeute, 2020-04-28:
*that* line is indeed not given, because it just happens to work right *now* with your *specific release* of your *specific distribution*.
it won't work with the next release of your distribution.
it won't work with a different distribution.

but anyhow, thanks for all the additional (and same-same) information. hopefully people with the same-same problem will stumble upon it and read it before opening the same bug-report again :-)

unicornofdeath, 2020-04-29:
> @unicornofdeath wow: how does this differ from the [README](https://github.com/umlaeute/v4l2loopback)?
> 
> i'm seriously baffled.

> @unicornofdeath wow: how does this differ from the [README](https://github.com/umlaeute/v4l2loopback)?
> 
> i'm seriously baffled.

whole apt install line solves problems

unicornofdeath, 2020-04-29:
> This line is different.
> `sudo apt-get install linux-image-4.19.0-6-amd64-unsigned linux-headers-4.19.0-6-amd64`.
> The commands given in README did not work for `4.19.0-2`, that's why I amended the issue.
> Sorry for the inconvenience.

Try to Install for you kernel version