v4l2loopback/v4l2loopback #299: Incompatibility with kernel 5.6.x

Yamakuzure, 2020-04-27:
## environment
  * `v4l2loopback` version: 0.12.5
  * kernel version: linux-5.6.5-gentoo
  * Distribution (+version): Gentoo
  
## Steps to reproduce:
  1. make

## Observed Results:
````
v4l2loopback-0.12.5/v4l2loopback.c:137:47: warning: ‘struct timeval’ declared inside parameter list will not be visible outside of this definition or declaration
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                               ^~~~~~~
````
And many followup errors of type:
  `error: dereferencing pointer to incomplete type ‘struct timeval’`
  
## Expected Results:
  * Succesful build
  
## Other information:
It seems that there have been quite some changes towards ktime_t. However, I was able to compile the module with the following patch, but am pretty sure that there is a much more efficient (and correct) way to fix this. Honestly, I don't even know whether the module works with this...
````
--- a/v4l2loopback-0.12.5/v4l2loopback.c        2020-04-27 11:12:31.550824848 +0200
+++ b/v4l2loopback-0.12.5/v4l2loopback.c    2020-04-27 11:31:25.550780536 +0200
@@ -16,7 +16,16 @@
 #include <linux/version.h>
 #include <linux/vmalloc.h>
 #include <linux/mm.h>
-#include <linux/time.h>
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 6, 0)
+# include <uapi/linux/time.h>
+# include <uapi/linux/module.h>
+# include <uapi/linux/videodev2.h>
+# include <uapi/linux/v4l2-common.h>
+# define timeval __kernel_v4l2_timeval
+  typedef __kernel_old_time_t time_t;
+#else
+# include <linux/time.h>
+#endif
 #include <linux/module.h>
 #include <linux/videodev2.h>
 #include <linux/sched.h>
````

umlaeute, 2020-04-27:
it seems like your `v0.12.5` does not match mine (or the one in this repository):

line 1375, where you get an error looks completely different: https://github.com/umlaeute/v4l2loopback/blob/v0.12.5/v4l2loopback.c#L137

and the problem you described has been fixed in d7b657802c109ece5195208db26059921d2d6ebe, which was included in the `v0.12.4` release.

closed 2020-04-27 by umlaeute

Yamakuzure, 2020-04-27:
You are right. The Gentoo ebuild I updated used a snapshot pre-0.12.1 instead of a regular release download. I totally overlooked that.

Sorry for the noise!