v4l2loopback/v4l2loopback #301: loopbackdevice only shown in firefox when privacy.resistfingerprinting=true

dadosch, 2020-04-27:
### environment

  * `v4l2loopback` version: 0.12.5
  * kernel version:  5.4.35-1-lts
  * Distribution (+version): Arch Linux

### the problem:
Ok, so it might sound really silly: loopbackdevices created by v4l2loopback are only shown in firefox when privacy.resistfingerprinting is set to true. All other cameras are shown independently from that setting. ` exclusive_caps=1` doesn't resolve it. setting a `card_label` doesn't resolve it.

#### Steps to reproduce:

  1. load v4l2loopback module
  2. open any video source, for example OBS and let it output to the video device
  3. open firefox, be sure to set privacy.resistfingerprinting=false.
  
#### Observed Results:

If you go to a website which requests the camera, all cameras except the loopback device are shown.
  
#### Expected Results:

the loopback device should be shown just as the other cameras

#### Notes:
I'm not too sure this is actually a bug of v4l2loopback… Maybe anyone else has experienced the same?

dadosch, 2020-04-27:
OK, I don't know why that happened, but it seems to be solved when using the latest v4l2sink for obs. Then delete all obs settings. Then it will work in firefox without any flaws…

I never experienced such a strange bug, wherever it was in…

closed 2020-04-27 by dadosch

lights0123, 2020-09-11:
I'm experiencing this same issue on Arch Linux, but using a custom driver (pyfakewebcam). Could it be related to https://bugzilla.mozilla.org/show_bug.cgi?id=1372073?