v4l2loopback/v4l2loopback #302: Depmod skipped on install

gerben-b, 2020-04-29:
* `v4l2loopback` version: v4l2loopback version: 0.12.5
  * kernel version: 5.4.0-26-generic #30-Ubuntu
  * Distribution (+version): _____
No LSB modules are available.
Distributor ID: Ubuntu
Description: Ubuntu 20.04 LTS
Release: 20.04
Codename: focal

When I run make install, I get this output:
__________
_make -C /lib/modules/`uname -r`/build M=/home/gerben/ComputerVision/v4l2loopback-master modules_install
make[1]: Map '/usr/src/linux-headers-5.4.0-28-generic' wordt binnengegaan
  INSTALL /home/gerben/ComputerVision/v4l2loopback-master/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-28-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Map '/usr/src/linux-headers-5.4.0-28-generic' wordt verlaten

SUCCESS (if you got 'SSL errors' above, you can safely ignore them)_
___________

#### Steps to reproduce:

  1. make (no errors)
  2. sudo make install (above error)
  
#### Observed Results:

Depmod is skipped en v4l2loopback is not installed.  When I first installed it (yesterday) it worked okay, today in the afternoon it no longer worked for no reason I can pinpoint.
#### Expected Results:

I expected this to work and create virtual v4l2 devices.  
#### Relevant Code:
None.

umlaeute, 2020-04-29:
first:
- please put the invocation and output of make into a code-section (starting and ending with `~~~`), so they get properly formatted and are visually distinct from the surrounding text.
- please run make with the environment `LANG=C`; i don't speak dutch.

then:

- where do you see an error? i don't see any
  if you do get an error, please be sure to include it your ticket.

finally: `depmod` is called (and eventually skipped) by the build-infrastructure from your kernel-headers; are you sure you got everything in place? (e.g. the `System.map` must be provided by your kernel)

Dlutz159, 2020-05-03:
I seem to be having this same issue. I am running a fresh install of Ubuntu 20.04 running kernel 5.4.0-28-generic. Hopefully I can jump in here but if I should open a separate issue please let me know.

**Output of sudo make**
~~~$sudo make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/USER/Documents/_git/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
~~~

**Output of sudo make install**
~~~$sudo make install
make -C /lib/modules/`uname -r`/build M=/home/USER/Documents/_git/v4l2loopback modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
  INSTALL /home/USER/Documents/_git/v4l2loopback/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-28-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'

SUCCESS (if you got 'SSL errors' above, you can safely ignore them)
~~~

The error here (I think) is "missing System.map file. Skipping depmod" This leads to issues with:

**sudo modprobe v4l2loopback**
`modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/5.4.0-28-generic
`

gerben-b, 2020-05-03:
> first:
>
>     * please put the invocation and output of make into a code-section (starting and ending with `~~~`), so they get properly formatted and are visually distinct from the surrounding [...]

I don't understand any of this but it doesn't seem to address the problem.

A day later I tried it again and 'magically' the System.map file was found and it worked. All in all I tried to install this in several linux installations and I got it working in only one and even then not reliably. I tried to analyze the make file but could't find a reference to the System.map file. My guess is that this v4l2loopback repository in itself is working alright, but it needs work to make it reliable.

umlaeute, 2020-05-04:
> Also, when make install said that System.map was missing, I checked and the file was there.

are you sure? are you sure that the *correct* `System.map` was there?

you said:
>  kernel version: 5.4.0-26-generic #30-Ubuntu

and your in your build logs you have:

> make[1]: Map '/usr/src/linux-headers-5.4.0-28-generic' wordt binnengegaan

so it seems were compiling the module for a different kernel then the one you were using.

> A day later I tried it again

what happened in the meantime?
did you by chance do a reboot (thus synching the running kernel with the kernel headers)?

## also...

> I don't understand any of this but...

it seems your English is well enough, so i guess the problem is that i fail to break down my rather technical concerns to "normal people".
i'd like to improve this situation. but i need more specific feedback than "i don't understand".
esp. the my 2nd question ("where do you see an error?") seems plain enough, that i don't understand what you don't understand.

gerben-b, 2020-05-05:
I am sure that System.map was there, I have only one kernel and I could build and run it before and also afterwards. When it didn't work I had Microsoft teams installed. Teams is still there and now v4l2loopback ' magically works again. I did a reboot once which didn't solve the problem and I dis one the next day and then it worked again.
The error you didn't see is this "Warning: modules_install: missing 'System.map' file. Skipping depmod.". It is an error disguised as a warning.

I am sorry that I can't be more specific. I think the install process is perhaps too sensitive. Currently it has been working for days without a problem, I have no idea why it failed in between.

gerben-b, 2020-05-06:
I ran sudo apt dist-upgrade, got a new kernel and than it failed again. I guess you are right and it has to do with kernels. However, a reboot did not help. I guess, more documentation is necessary here.

closed 2020-05-06 by gerben-b

dbear496, 2021-07-29:
FWIW I ran into this same problem today. v4l2loopback has been working for me for a long time, but I did an update yesterday, and I guess that broke it. I have done two reboots since the update, but v4l2loopback still won't install. I checked the build logs against `uname -r`, and the kernel versions are the same: `5.8.0-63-generic`. It sounds like the problem mysteriously vanished for @gerben-b, so I guess I will wait a few reboot cycles to see if the problem goes away for me. I will also try doing a `git pull` to see if a newer version of v4l2loopback fixes the issue.

dbear496, 2021-07-29:
`git pull` didn't fixe the issue. However, I tried installing v4l2loopback with an older kernel (`5.8.0-59-generic`) and it worked fine. The make logs still showed `Warning: modules_install: missing 'System.map' file. Skipping depmod.`, but it still seemed to work. Though it would be nice for this to work with the newer kernel as well.

dbear496, 2021-07-29:
Turns out that my issue is #208, and I was able to fix it by following that thread. I also have `Warning: modules_install: missing 'System.map' file. Skipping depmod.` in my make logs, so that was confusing me and making me think it was this issue.