v4l2loopback/v4l2loopback #305: modprobe: ERROR: could not insert 'v4l2loopback_dc': Bad address

grodrum, 2020-05-04:
* `v4l2loopback` version: 0.12.5

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: Linux ubuntu 5.4.0-28-generic #32-Ubuntu SMP Wed Apr 22 17:40:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

          uname -a

  * Distribution (+version): 20.04 LTS

          lsb_release -a
  
### Step 3: Describe the problem:

#### Steps to reproduce:

  1. Installed v4l2loopback in different ways: using `sudo apt install`, then removed it using `sudo apt purge` and installed it using `git clone ... && make && make install`
  2. Run `sudo modprobe v4l2loopback-dc`

#### Observed Results:

  modprobe: ERROR: could not insert 'v4l2loopback_dc': Bad address

  `journalctl -f`

тра 05 01:04:57 ubuntu sudo[152846]:   spacer : TTY=pts/1 ; PWD=/home/spacer ; USER=root ; COMMAND=/usr/sbin/modprobe v4l2loopback-dc
тра 05 01:04:57 ubuntu sudo[152846]: pam_unix(sudo:session): session opened for user root by (uid=0)
тра 05 01:04:57 ubuntu kernel: ------------[ cut here ]------------
тра 05 01:04:57 ubuntu kernel: WARNING: CPU: 0 PID: 152847 at drivers/media/v4l2-core/v4l2-dev.c:863 __video_register_device+0x39e/0x600 [videodev]
тра 05 01:04:57 ubuntu kernel: Modules linked in: v4l2loopback_dc(OE+) v4l2loopback(OE) videodev vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) bnep ccm bluetooth ecdh_generic ecc binfmt_misc mt7601u mac80211 joydev input_leds cfg80211 libarc4 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core snd_usb_audio snd_seq_midi coretemp snd_usbmidi_lib snd_seq_midi_event snd_hwdep mc kvm_intel snd_rawmidi snd_pcm kvm snd_seq hp_wmi sparse_keymap wmi_bmof serio_raw snd_seq_device snd_timer snd soundcore tpm_infineon mei_me mei mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs xor zstd_compress raid6_pq libcrc32c dm_mirror dm_region_hash dm_log hid_generic usbhid hid i915 gpio_ich i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops lpc_ich psmouse ahci libahci drm e1000e wmi video
тра 05 01:04:57 ubuntu kernel: CPU: 0 PID: 152847 Comm: modprobe Tainted: G        W  OE     5.4.0-28-generic #32-Ubuntu
тра 05 01:04:57 ubuntu kernel: Hardware name: Hewlett-Packard HP Compaq 8000 Elite SFF PC/3646h, BIOS 786G7 v01.14 10/14/2015
тра 05 01:04:57 ubuntu kernel: RIP: 0010:__video_register_device+0x39e/0x600 [videodev]
тра 05 01:04:57 ubuntu kernel: Code: 0f 84 d6 01 00 00 41 8b 57 2c 85 d2 0f 85 a5 01 00 00 f0 41 80 8f e8 04 00 00 01 31 db 89 d8 5b 41 5c 41 5d 41 5e 41 5f 5d c3 <0f> 0b bb ea ff ff ff 89 d8 5b 41 5c 41 5d 41 5e 41 5f 5d c3 83 f8
тра 05 01:04:57 ubuntu kernel: RSP: 0018:ffffbb978181fc18 EFLAGS: 00010246
тра 05 01:04:57 ubuntu kernel: RAX: ffff9118c4a9e000 RBX: ffffffffc0bbd140 RCX: 0000000000000000
тра 05 01:04:57 ubuntu kernel: RDX: 00000000ffffffff RSI: 0000000000000000 RDI: ffff9118caffd000
тра 05 01:04:57 ubuntu kernel: RBP: ffffbb978181fc40 R08: ffffffffc0bbd140 R09: ffff9118d2803c00
тра 05 01:04:57 ubuntu kernel: R10: 0000000000000000 R11: ffff9118dbfd2000 R12: 0000000000000001
тра 05 01:04:57 ubuntu kernel: R13: 00000000ffffffff R14: ffffffffc0bbd158 R15: ffff9118caffd000
тра 05 01:04:57 ubuntu kernel: FS:  00007fbdda695540(0000) GS:ffff9118d3a00000(0000) knlGS:0000000000000000
тра 05 01:04:57 ubuntu kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
тра 05 01:04:57 ubuntu kernel: CR2: 000055c2dcab5d19 CR3: 00000000b9ade000 CR4: 00000000000406f0
тра 05 01:04:57 ubuntu kernel: Call Trace:
тра 05 01:04:57 ubuntu kernel:  init_module+0x369/0x1000 [v4l2loopback_dc]
тра 05 01:04:57 ubuntu kernel:  ? 0xffffffffc0ba0000                                                                                                                                         
тра 05 01:04:57 ubuntu kernel:  do_one_initcall+0x4a/0x1fa                                                                                                                                   
тра 05 01:04:57 ubuntu kernel:  ? _cond_resched+0x19/0x30                                                                                                                                    
тра 05 01:04:57 ubuntu kernel:  ? kmem_cache_alloc_trace+0x163/0x230                                                                                                                         
тра 05 01:04:57 ubuntu kernel:  do_init_module+0x62/0x250                                                                                                                                    
тра 05 01:04:57 ubuntu kernel:  load_module+0x10b8/0x1200                                                                                                                                    
тра 05 01:04:57 ubuntu kernel:  __do_sys_finit_module+0xbe/0x120                                                                                                                             
тра 05 01:04:57 ubuntu kernel:  ? __do_sys_finit_module+0xbe/0x120
тра 05 01:04:57 ubuntu kernel:  __x64_sys_finit_module+0x1a/0x20
тра 05 01:04:57 ubuntu kernel:  do_syscall_64+0x57/0x190
тра 05 01:04:57 ubuntu kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xa9
тра 05 01:04:57 ubuntu kernel: RIP: 0033:0x7fbdda7da70d
тра 05 01:04:57 ubuntu kernel: Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 53 f7 0c 00 f7 d8 64 89 01 48                                                                                                                                                    
тра 05 01:04:57 ubuntu kernel: RSP: 002b:00007ffd8a23a428 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
тра 05 01:04:57 ubuntu kernel: RAX: ffffffffffffffda RBX: 0000559a733fea80 RCX: 00007fbdda7da70d
тра 05 01:04:57 ubuntu kernel: RDX: 0000000000000000 RSI: 0000559a714ee358 RDI: 0000000000000003
тра 05 01:04:57 ubuntu kernel: RBP: 0000000000040000 R08: 0000000000000000 R09: 0000000000000000
тра 05 01:04:57 ubuntu kernel: R10: 0000000000000003 R11: 0000000000000246 R12: 0000559a714ee358
тра 05 01:04:57 ubuntu kernel: R13: 0000000000000000 R14: 0000559a733febb0 R15: 0000559a733fea80
тра 05 01:04:57 ubuntu kernel: ---[ end trace fe5e84655019f7ce ]---
**тра 05 01:04:57 ubuntu kernel: v4l2loopback: failed video_register_device()**
тра 05 01:04:57 ubuntu kernel: ------------[ cut here ]------------
тра 05 01:04:57 ubuntu kernel: kernfs: can not remove 'format', no directory
тра 05 01:04:57 ubuntu kernel: WARNING: CPU: 0 PID: 152847 at fs/kernfs/dir.c:1503 kernfs_remove_by_name_ns+0x8a/0xa0
тра 05 01:04:57 ubuntu kernel: Modules linked in: v4l2loopback_dc(OE+) v4l2loopback(OE) videodev vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) bnep ccm bluetooth ecdh_generic ecc binfmt_misc mt7601u mac80211 joydev input_leds cfg80211 libarc4 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core snd_usb_audio snd_seq_midi coretemp snd_usbmidi_lib snd_seq_midi_event snd_hwdep mc kvm_intel snd_rawmidi snd_pcm kvm snd_seq hp_wmi sparse_keymap wmi_bmof serio_raw snd_seq_device snd_timer snd soundcore tpm_infineon mei_me mei mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs xor zstd_compress raid6_pq libcrc32c dm_mirror dm_region_hash dm_log hid_generic usbhid hid i915 gpio_ich i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops lpc_ich psmouse ahci libahci drm e1000e wmi video                                                  
[... 114 lines cut ...]
тра 05 01:04:57 ubuntu kernel: Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 53 f7 0c 00 f7 d8 64 89 01 48                                                                                                                                                    
тра 05 01:04:57 ubuntu kernel: RSP: 002b:00007ffd8a23a428 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
тра 05 01:04:57 ubuntu kernel: RAX: ffffffffffffffda RBX: 0000559a733fea80 RCX: 00007fbdda7da70d
тра 05 01:04:57 ubuntu kernel: RDX: 0000000000000000 RSI: 0000559a714ee358 RDI: 0000000000000003
тра 05 01:04:57 ubuntu kernel: RBP: 0000000000040000 R08: 0000000000000000 R09: 0000000000000000
тра 05 01:04:57 ubuntu kernel: R10: 0000000000000003 R11: 0000000000000246 R12: 0000559a714ee358
тра 05 01:04:57 ubuntu kernel: R13: 0000000000000000 R14: 0000559a733febb0 R15: 0000559a733fea80
тра 05 01:04:57 ubuntu kernel: ---[ end trace fe5e84655019f7d1 ]---

  
#### Expected Results:

  Module could be loaded and I could install and use Droidcam
  
#### Relevant Code:

  ```
  sudo apt install linux-headers-`uname -r`
  sudo apt install v4l2loopback-dkms v4l2loopback-utils
  sudo modprobe v4l2loopback
  wget https://www.dev47apps.com/files/600/droidcam-64bit.tar.bz2 -O droidcam.tar.bz2
  tar xjvf droidcam.tar.bz2
  cd droidcam-64bit/
  sudo ./install
  ```

grodrum, 2020-05-04:
```
$ v4l2-ctl --list-devices
Dummy video device (0x0000) (platform:v4l2loopback-000):
        /dev/video0
```

```
$ sudo v4l2-ctl --device=/dev/video0 --all
Driver Info:
        Driver name      : v4l2 loopback
        Card type        : Dummy video device (0x0000)
        Bus info         : platform:v4l2loopback-000
        Driver version   : 5.4.30
        Capabilities     : 0x85208003
                Video Capture
                Video Output
                Video Memory-to-Memory
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x05208003
                Video Capture
                Video Output
                Video Memory-to-Memory
                Read/Write
                Streaming
                Extended Pix Format
Priority: 0
Video input : 0 (loopback: ok)
Video output: 0 (loopback in)
Format Video Output:
        Width/Height      : 0/0
        Pixel Format      : 'BGR4' (32-bit BGRA/X 8-8-8-8)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 0
        Colorspace        : sRGB
        Transfer Function : Default (maps to sRGB)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
        Flags             : 
Streaming Parameters Video Capture:
        Frames per second: 30.000 (30/1)
        Read buffers     : 2
Streaming Parameters Video Output:
        Frames per second: 30.000 (30/1)
        Write buffers    : 2

User Controls

                    keep_format 0x0098f900 (bool)   : default=0 value=0
              sustain_framerate 0x0098f901 (bool)   : default=0 value=0
                        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io 0x0098f903 (bool)   : default=0 value=0
```

grodrum, 2020-05-04:
```
~/Downloads/droidcam-64bit$ sudo ./install 
[sudo] password for spacer: 
Webcam parameters: '640' and '480'
Building v4l2loopback-dc.ko
make: Entering directory '/home/spacer/Downloads/droidcam-64bit/v4l2loopback'
make -C /lib/modules/`uname -r`/build M=`pwd`
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
make: Leaving directory '/home/spacer/Downloads/droidcam-64bit/v4l2loopback'
Moving driver and executable to system folders
Registering webcam device
insmod: ERROR: could not insert module /lib/modules/5.4.0-28-generic/kernel/drivers/media/video/v4l2loopback-dc.ko: Bad address
```
```
$ diff v4l2loopback/v4l2loopback-dc.ko /lib/modules/5.4.0-28-generic/kernel/drivers/media/video/v4l2loopback-dc.ko -s
Files v4l2loopback/v4l2loopback-dc.ko and /lib/modules/5.4.0-28-generic/kernel/drivers/media/video/v4l2loopback-dc.ko are identical
spacer@ubuntu:~/Downloads/droidcam-64bit$ ls /lib/modules/5.4.0-28-generic/kernel/drivers/media/video/
v4l2loopback-dc.ko
```

closed 2020-05-04 by grodrum

grodrum, 2020-05-05:
I've realized v4l2loopback module is working by using commands to verify it:

`ffmpeg -f x11grab -framerate 15 -video_size 1280x720 -i :0.0 -f v4l2 /dev/video0`
`ffplay /dev/video0`

Then I've found a different way to install droidcam: https://askubuntu.com/a/1235432/925971

WashingtonSteve, 2023-08-21:
I managed to solve the problems like this

`# Install necessary dependencies
sudo apt install git build-essential linux-headers-$(uname -r)

# Clone the repository
git clone https://github.com/umlaeute/v4l2loopback.git

# Navigate into the directory
cd v4l2loopback

# Compile and install
make && sudo make install

# Load the module
sudo modprobe v4l2loopback
`

darthdeus, 2023-11-15:
For anyone finding this from Arch, I had a similar issue with the AUR packages, but just manually building v4l2loopback and running droidcam via the github release binaries instead of AUR did the trick.