v4l2loopback/v4l2loopback #306: Unsigned v4l2loopback module leads to error

JJvdBerge, 2020-05-06:
When installing Iriun (Webcam) which has  v4l2loopback as a dependency I get an error:

`modprobe: ERROR: could not insert 'v4l2loopback': Operation not permitted
dpkg: error processing package iriunwebcam (--configure):
 installed iriunwebcam package post-installation script subprocess returned error exit status 1
Setting up libqt5opengl5-dev:amd64 (5.12.8+dfsg-0ubuntu1) ...
Errors were encountered while processing:
 iriunwebcam
` 
Some searching led me to [an issue in another project](https://github.com/alievk/avatarify/issues/43) which seems to be similar (unsigned module). Currently running Ubuntu 20.04 LTS (running kernel 5.4.0.29). Looking here (and in the FAQ) I couldn't see it as a filed issue so I thought I would submit one (as suggested in the last [post](https://github.com/alievk/avatarify/issues/43#issuecomment-616131111) of the previous link)

Ange1ique, 2020-05-11:
Hi, (I think) I had the same problem and found a solution for how to get the module signed. I also have Ubuntu 20.04 LTS (running kernel 5.4.0.29), with secure boot enabled (installing works with secure boot disabled, but I didn't like this option). I tried a lot before this, but I guess these steps made it work:
- uninstall v4l2loopback with: sudo apt-get remove --auto-remove v4l2loopback-dkms
- install v4l2loopback again with: sudo apt-get install v4l2loopback-dkms
You will get a message about 'Configuring Secure Boot', which offers you the opportunity to create a key for this module. Read carefully and continue with tab(s) and an enter. After setting the key/password and rebooting with the key/password:
- enter this in the terminal: sudo modprobe v4l2loopback (don't know if this is necessary, but this time I didn't get an error)
- install Iriun Webcam again with the deb file
Hope it works for you too!

SleutelMan, 2020-05-11:
hi @Ange1ique , thanks for the reply. That worked. I had done part of the process already, but hadn't read the instructions carefully enough it seems about the reboot and MOK enrollment. The combination of Secure Boot and Ubuntu is quite new to me, I used to run it the old-fashioned way so although I entered a key I forgot the enrollment process. 

It seems this issue is an user error and not an issue with v4l2loopback, so I'll close it. Thanks again! :-)

closed 2020-05-11 by SleutelMan

melvinroest, 2020-05-25:
@Ange1ique This should've ranked higher in Google when I was looking for this problem.

The whole `dmesg -c` and noticing that there's a kernel lockdown has been one big rabbit hole in retrospect. All I needed to do was simply select a different boot option. 

I really appreciate the answer!

nyxz, 2023-08-01:
In case anyone else comes here for the same issue but in Fedora realm - I got the same error in Fedora 36 with Secure boot enabled. I found the [Fedora doc for Secure boot](https://rpmfusion.org/Howto/Secure%20Boot?highlight=%28%5CbCategoryHowto%5Cb%29) that suggested the akmods are a convenient solution.
> With Fedora 36+, the akmods package have support to automatically sign locally built kmod with a self generated key. Such key must be imported into the EFI firmware (you must have right to access the EFI firmware).
>
> It's not required to disable secure boot (not even move to BIOS compatibility).

1. I installed `akmod-v4l2loopback` package (along with `v4l2loopback`)
    ```
    sudo dnf install akmod-v4l2loopback
    ```
2. Then I run
    ```
    /usr/sbin/kmodgenca
    ```
    although in my case this was unnecessary since on my system the `/etc/pki/akmods/certs/public_key.der` and `/etc/pki/akmods/private/private_key.priv` keys already existed so the command did nothing.
3. The I imported the key with:
    ```
    mokutil --import /etc/pki/akmods/certs/public_key.der
    ```
    and set a password for the key imported and rebooted the system. Then on load there would be a screen to load the MOK where you'll need to enter the password again. See the `/usr/share/doc/akmods/README.secureboot` for full guide.

After the total of 2 reboots I loaded my system and executed successfully:
```
modprobe v4l2loopback exclusive_caps=1
```

I hope this is useful to someone.

josh-001, 2023-10-17:
> Hi, (I think) I had the same problem and found a solution for how to get the module signed. I also have Ubuntu 20.04 LTS (running kernel 5.4.0.29), with secure boot enabled (installing works with [...]

Hi @Ange1ique , I removed like as you said but after trying to install again I am not getting any pop up like I did at first.can you /anyone help me this ?

josh-001, 2023-10-17:
> Hi @Ange1ique , I removed like as you said but after trying to install again I am not getting any pop up like I did at first.can you /anyone help me this ?

i resolved the issue !!

removed and installed again and in reboot I clicked enroll || MOK and entered the password I gave at the time of installation.after reboot I didn't got any error.
hope this one helps