v4l2loopback/v4l2loopback #308: Not rules for make ?

feydreva, 2020-05-16:
Hello, 

i am trying to isntall v4l2loopbak.
When I do a make install, i get errors that make did not find target for modules_install...
I do not know what to do from there...

>feydreva@Spychea  v4l2loopback]$ sudo make install
make -C /lib/modules/`uname -r`/build M=/home/feydreva/v4l2loopback modules_install
make[1] : on entre dans le répertoire « /usr/lib/modules/5.6.11-1-MANJARO/build »
make[1]: *** Aucune règle pour fabriquer la cible « modules_install ». Arrêt.
make[1] : on quitte le répertoire « /usr/lib/modules/5.6.11-1-MANJARO/build »
make: *** [Makefile:47 : install] Erreur 2

Kernel version : 
Linux Spychea 5.6.11-1-MANJARO #1 SMP PREEMPT Wed May 6 10:20:32 UTC 2020 x86_64 GNU/Linux

Distribution : 
LSB Version:    n/a
Distributor ID: ManjaroLinux
Description:    Manjaro Linux
Release:        20.0.1
Codename:       Lysia

thank you for you help.

umlaeute, 2020-06-01:
this looks a bit like an incomplete kernel-header installation.

umlaeute, 2020-06-18:
closing due to inactivity

closed 2020-06-18 by umlaeute