v4l2loopback/v4l2loopback #310: Multiple consumers can access the same loopback

justyn, 2020-05-22:
Hi

This isn't a problem or a bug report, but I wanted to raise this here for two reasons:

1) So there's a record for other people to know it's possible
2) To request an explanation of what is going on

I had turned to v4l2-loopback to share a webcam between three different applications (web browser, a videoconferencing app and ffmpeg to record). I originally set up with 3 loopbacks, one for each application.

But I discovered that it also works with a single loopback - all 3 consumers can read from the same loopback without causing a "device or resource busy" error.

It seems to work fine.

I don't really understand why it is possible with a loopback but not with the original webcam stream, so I was hoping someone here could explain.

For reference, here is the command I use to create the loopback:

`sudo modprobe v4l2loopback video_nr=30 exclusive_caps=1 card_label="loopback cam"`

And the command to stream the webcam at /dev/video5 into it:

`ffmpeg -f v4l2 -i /dev/video5 -codec copy -f v4l2 /dev/video30`

System info in case it's relevant:

Ubuntu 20.04
Kernel: 5.4.0-31-generic
package: v4l2loopback-dkms 0.12.3-1

I've also asked the question on unix.stackexchange [here](https://unix.stackexchange.com/questions/588118/why-can-multiple-consumers-access-a-single-v4l2-loopback-stream-from-a-webcam) 

Thanks!

justyn, 2020-05-22:
Thanks, is there any disadvantage to doing it this way rather than using individual loopbacks?

So far it seems perfectly stable.

umlaeute, 2020-05-25:
once we fix the queuing/dequeuing it *might* not be possible any more to do this.

however, until then, you should be totally safe (and it should be more performant this way)

justyn, 2020-05-26:
Thanks for the explanation.

Someone on Stack Exchange has [added some more detail too](https://github.com/pure-data/pure-data).

I hope you are able to keep this functionality in the future, since it is very useful for use cases like mine!

I'll close this issue now since my question is answered, thank you.

closed 2020-05-26 by justyn

umlaeute, 2020-05-26:
@justyn out of interest: could you fix the link to the stack-exchange page?

justyn, 2020-05-26:
Oh yes sorry, link is fixed now.

TadayukiOkada, 2021-09-30:
I tried opening one v4l2loopback device from multiple gst-launch command.
They did work and I saw multiple video windows.
However, when I killed one of the gst-launch command, other commands were also killed.
Is there any way to make killing one consumer not affect other consumers?

baptx, 2026-03-01:
I agree with @justyn, it would be useful to keep this feature but after upgrading from Mobian 12 to 13 on the PinePhone, I can see this feature was removed.
As @umlaeute wrote it should be more performant and I can confirm this because when using 2 devices with the command `sudo modprobe v4l2loopback devices=2` on a slower CPU like the PinePhone, I now have to limit the video recording script preview to 5 FPS to have a smooth recording. It seems to use more resources:
https://gist.github.com/baptx/7faaaf876ef38db040ce0e25bde41625
Can you add this feature back or is there already a workaround to enable it? The issue https://github.com/v4l2loopback/v4l2loopback/issues/566 could be reopened.

I also noticed another issue with v4l2loopback after the Mobian upgrade. Now we have to execute the modprobe command manually after boot, it is no longer possible to load v4l2loopback module automatically in files `/etc/modprobe.d/dkms.conf` and `/etc/modules-load.d/modules.conf`, otherwise there is an error "Error submitting a packet to the muxer: Invalid argument" when trying to record with ffmpeg.

baptx, 2026-03-03:
It looks like that is already what I am doing with ffmpeg in the script from my first comment in this issue.
But it is slower using 2 devices and to minimize power draw, I am already reducing framerate to 5 FPS for the video preview.
I cannot stop a stream because I need both a stream for the video preview and the recording.
I don't think pipewire would change the issue in my case since I need both streams without stopping them.
However the "-f xv display" option of ffmpeg could be an alternative to record a video with preview, then we don't need v4l2loopback at all:
https://superuser.com/questions/343586/how-to-display-and-capture-webcam-stream-at-the-same-time/1551806#1551806
When I tried this option a few years ago, there was a delay in the preview but it seems to be fixed when reducing the recording framerate on the PinePhone.