v4l2loopback/v4l2loopback #311: newbie needs help - dummy video stream choppy compared to video0

ram-sh, 2020-05-23:
Hi, i'm still finding my way around this and truly need your support!

using RPI4 4gb running 
`Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux`

i have a GoPro camera connected to the PI using [Cam Link](https://www.elgato.com/en/gaming/cam-link-4k) to its USB3:
```
v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
Type: Video Capture

[0]: 'YUYV' (YUYV 4:2:2)
Size: Discrete 1920x1080
Interval: Discrete 0.017s (59.940 fps)
```

i loaded the v4l2loopback module as root
`sudo modprobe v4l2loopback`

and the new video2 is created:

```v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12

Dummy video device (0x0000) (platform:v4l2loopback-000):
	/dev/video2

Cam Link: Cam Link (usb-0000:01:00.0-2):
	/dev/video0
	/dev/video1
```

changed the FPS to match the stream (tried 60 also)
```
v4l2loopback-ctl set-fps 59.94 /dev/video2
OK

```

and started ffmpeg to duplicate video0 to video2
`ffmpeg -re -f v4l2 -i /dev/video0 -f v4l2 -pix_fmt yuv420p /dev/video2`

my issue is when capturing the camera feed without the loopback the quality of the output is decent (considering my ffmpeg expertise) but when using the same ffmpeg command pointing to video2 the output is all choppy and messed up, the ffmpeg command video0:

``` 
ffmpeg -y -nostdin -re -f v4l2 -i /dev/video0 -f alsa -ac 2 -itsoffset -1 -i default -vcodec h264_omx -s 896x504 -r 60 -pix_fmt yuv420p -b:v 6M -minrate 6M -maxrate 6M -bufsize 100M -acodec ac3 -b:a 96k -ar 48000 -af aresample=async=1 -movflags +faststart -f mp4 /usr/ramDisk/h264_omx_2.mp4 
```

the output movie quality can be [shown here](https://youtu.be/7bvZeq0cpz0)

the same ffmpeg command using `-i /dev/video2` produced [this video](https://youtu.be/GXkic5j-gRE)

clearly i'm missing something here and would appreciate your help much!

==================================================
the ffmpeg version I use is:
```
ffmpeg version git-2020-05-21-d29c429 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib --extra-libs='-lpthread -lm' --arch=armel --enable-gmp --enable-gpl --enable-libaom --enable-libass --enable-libdav1d --enable-libdrm --enable-libfdk-aac --enable-libfreetype --enable-libkvazaar --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopus --enable-librtmp --enable-libsnappy --enable-libsoxr --enable-libssh --enable-libvorbis --enable-libvpx --enable-libzimg --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-mmal --enable-nonfree --enable-omx --enable-omx-rpi --enable-version3 --target-os=linux --enable-pthreads --enable-openssl --enable-hardcoded-tables
  libavutil      56. 46.100 / 56. 46.100
  libavcodec     58. 86.101 / 58. 86.101
  libavformat    58. 43.100 / 58. 43.100
  libavdevice    58.  9.103 / 58.  9.103
  libavfilter     7. 82.100 /  7. 82.100
  libswscale      5.  6.101 /  5.  6.101
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
Hyper fast Audio and Video encoder
```

ram-sh, 2020-05-24:
more info as i try to make it to work:

set the w/h to be the same as the source input:
`sudo modprobe v4l2loopback devices=1 video_nr=3 card_label="GoPro Cam" max_width=1920 max_height=1080 exclusive_caps=1`

checking the new dummy:
```
v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12

GoPro Cam (platform:v4l2loopback-000):
	/dev/video3

Cam Link: Cam Link (usb-0000:01:00.0-2):
	/dev/video0
	/dev/video1
```

update the FPS to match the camera:
```
v4l2loopback-ctl set-fps 59.94 /dev/video3
OK
```

running ffmpeg to duplicatre the stream from video0 to video3 by running:
`ffmpeg -re -f v4l2 -i /dev/video0 -f v4l2 -pix_fmt yuv420p /dev/video3`

it produced as an insane amount of duplicate frames right of the bat:
`frame=  771 fps= 60 q=-0.0 Lsize=N/A time=00:00:12.86 bitrate=N/A dup=516 drop=0 speed=0.998x `

could that be part of the problem?

mxa, 2020-05-31:
Stack Overflow has a tag v4l2loopback, maybe that's the better place for questions than the issue tracker?

https://stackoverflow.com/questions/tagged/v4l2loopback

umlaeute, 2020-07-01:
from the driver's perspective, the "framerate" field is really only **informal** - it doesn't do anything to enforce a given framerate.

so the problem you see here is that the producer (`ffmpeg`) simply ignores the framerate and writes the frames as fast as it can (which is apparently much faster).

the jumps in the frames are probably related to a somewhat buggy bufferhandling; try the latest and greatest version of `v4l2loopback` (or use `max_buffer=2`; but really get the latest `v4l2loopback`) to see if it helps (this doesn't fix the bugger buffer queuing, but gives it less chance to show).

umlaeute, 2020-07-01:
i just saw https://stackoverflow.com/questions/62394066/, which i think is *this* ticket on stackoverflow, and it has been answered and accepted.

therefore i'm closing this.

thanks for reporting back.

closed 2020-07-01 by umlaeute