v4l2loopback/v4l2loopback #314: how to set default settings at system start up?

ccoenen, 2020-06-02:
I am trying to set a few capture devices up with names on system start up. I can successfully set these from my terminal with the examples provided (i.e. `# modprobe v4l2loopback video_nr=3,4,7 card_label="device number 3","the number four","the last one"` from the docs).

what I have _not_ been able to achieve is the above setting but right after my computer boots. I always have to do this manually. I would appreciate an example if someone is willing to help me.

Bonus points if this also includes settings like timeout image or similar.

Thanks in advance :-)

johnteslade, 2020-06-02:
/etc/modprobe.d can do this.

See man modprobe.d for more or https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/6/html/deployment_guide/sec-Loading_a_Customized_Module-Persistent_Changes (the details apply to most distros).

umlaeute, 2020-06-05:
closing, as this is not an issue with `v4l2loopback`, but just general kernel-module handling

closed 2020-06-05 by umlaeute

ccoenen, 2020-06-07:
For future reference, here is what I do:

This is how I load the module on Ubuntu 20.04

```sh
$ cat /etc/modules-load.d/v4l2loopback.conf 
# enables video input and output as a webcam
v4l2loopback
```

This is how I configure the module:
```sh
$ cat /etc/modprobe.d/v4l2loopback.conf
options v4l2loopback video_nr=5,6,7 card_label="Loopback 5,Loopback 6,Loopback 7"
```