v4l2loopback/v4l2loopback #315: The device print messy codes via application when do internet streaming with  ffmpeg or OBS

alexunder, 2020-06-03:
## Describe my environment

  * `v4l2loopback` version: __0.12.5__

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: __4.13.0-45-generic__

          uname -a

  * Distribution (+version): ___ubuntu 16.04 LTS xenial

          lsb_release -a
  
### Step 3: Describe the problem:

#### Steps to reproduce:

  1. Using 
   
    modprobe v4l2loopback exclusive_caps=1

   to load the loopback driver.

  2. The application code using standard v4l2 interface to set the device:

 ```
    int dev_fd = open("/dev/video0", O_RDWR);
    if (dev_fd == -1) {
        fail("cannot open video device");
    }
    v.type = V4L2_BUF_TYPE_VIDEO_OUTPUT;
    if (ioctl(dev_fd, VIDIOC_G_FMT, &v) == -1){
        fail("cannot setup video device");
    }
    v.fmt.pix.width  = 1920;
    v.fmt.pix.height = 1080;
    v.fmt.pix.pixelformat = V4L2_PIX_FMT_RGB24;
    v.fmt.pix.sizeimage = 1920 * 1080 *3;
    v.fmt.pix.field = V4L2_FIELD_NONE;
    if (ioctl(dev_fd, VIDIOC_S_FMT, &v) == -1) {
        fail("cannot setup video device");
    }
```

  3. Using **write** system call to write video data to the loopback device.

  4. Using the command below to do internet streaming:

```
ffmpeg -f v4l2 -pix_fmt rgb24-s:v 1920x1080 -r 25 -i /dev/video0 -vcodec libx264 -f flv "rtmp://blablabla"
```
  
#### Observed Results:

 Then the streaming is successful, I can see the image from the internet. But after minutes, the image stops, then I can see the application generates many messy codes in terminal.
  
#### Expected Results:

The streaming work is running continuously.

umlaeute, 2020-06-05:
"messy codes"? please be more specific.

alexunder, 2020-06-09:
I am sorry that the weird behaviors happened due to my incorrect manipulations on the video device. Because I continued on reading the fd when the fd was shutdown. So I will close it right now.

closed 2020-06-09 by alexunder