v4l2loopback/v4l2loopback #316: document which programs it works with?

archenemies, 2020-06-04:
### Step 2: Describe your environment

  * `v4l2loopback` version: 0.12.5.r10.gd45a492-1

         $ sudo modprobe v4l2loopback
         $ sudo dmesg  | grep -i v4l2loopback
         (no output)

  * kernel version: 5.6.15-arch1-1 #1 SMP PREEMPT Wed, 27 May 2020 23:42:26 +0000 x86_64 GNU/Linux

  * Distribution (+version): Arch
  
### Step 3: Describe the problem:

I've managed to install the package and load the module, and write to the video device using ffmpeg as described [on superuser.com](https://superuser.com/questions/411897/using-desktop-as-fake-webcam-on-linux).

However, I am not able to read from the device using cheese or guvcview, which are the two applications that I know about that are able to read my laptop's webcam.

The device appears in dropdown menus but is not usable.

With cheese, there is no error message, it just selects a different device. With guvcview, I se errors about "Inappropriate ioctl for device". I'm attaching the full output if it is helpful: [guvcview-errors.txt](https://github.com/umlaeute/v4l2loopback/files/4727155/guvcview-errors.txt)

After several hours of debugging, and composing this issue, I discovered that all the applications I care about - Firefox (via https://www.onlinemictest.com/webcam-test/), Zoom - and Vlc, a media player which I had overlooked, which is able to do video mirroring, are all able to read the video device correctly:

    vlc -f v4l2:///dev/video2 :v4l2-standard= :input-slave=alsa://hw:2,0 :live-caching=300

So I'm not sure what the conclusion should be here, other than that I should have tried these other options sooner - but maybe it would have helped if the README.md mentioned that this tool works better with some programs than others?

Other than that, this seems like a very useful project and I'm grateful that it exists.

umlaeute, 2020-06-05:
the number of applications that support v4l2 is large.

i don't think anybody would like to compile a list and test each application whether it works with a given device.
and of course it doesn't end there: each new release of each application would need to be taken into account.

so i'm closing this as not practically doable.

closed 2020-06-05 by umlaeute

archenemies, 2020-06-05:
I wasn't asking for an exhaustive list, just a few examples in the README of how you can use it, and an acknowledgment that there are some unimplemented ioctls that make the virtual device distinguishable from a real one.

archenemies, 2020-06-05:
Otherwise people encounter "Inappropriate ioctl for device" and they don't know whether to submit a bug or try a new client program... Anyway, a few words about that would have saved me a lot of time.

umlaeute, 2020-06-05:
PRs are always welcome.

ccoenen, 2020-06-07:
I use it for two things:
gphoto2 puts my DSLR's video into a v4l2loopback camera, which is in turn picked up by OBS Studio, which in turn has a v4l2 sink (which is also loopback) which is what I use to make video conferences over Jitsi Meet or Big Blue Button.