v4l2loopback/v4l2loopback #317: Adding a rule in the Makefile to produce an installation tree (for package building)

downwater, 2020-06-17:
Usually, to install in a custom directory after compilation with a Makefile, the command line below is run:

    make install DESTDIR=/path/to/rootdir

Here, it seems that it's not possible, since the `install` target depends on `modules_install`. How about a new rule (`install-custom-root`?) which would copy the module as well as programs and manual pages to `DESTDIR`?

I want to generate a custom package (Slackware 64 14.2) from the sources.

(Tested with `v4l2loopback` version: [0.12.5](https://github.com/umlaeute/v4l2loopback/releases/tag/v0.12.5).)

umlaeute, 2020-06-18:
instead you should use the standard [`MODLIB`](https://www.kernel.org/doc/html/latest/kbuild/kbuild.html#modlib) or [`INSTALL_MOD_PATH`](https://www.kernel.org/doc/html/latest/kbuild/kbuild.html#install-mod-path)

closed 2020-06-18 by umlaeute

downwater, 2020-06-22:
Thanks, so the whole line becomes:

    make install-all PREFIX=/usr DESTDIR=/path/to/rootdir INSTALL_MOD_PATH=/path/to/rootdir

umlaeute, 2020-06-22:
only if you want the result to be:

~~~sh
/path/to/rootdir/lib/modules/$(uname -r)/extra/v4l2loopback.ko
/path/to/rootdir/usr/bin/v4l2loopback-ctl
/path/to/rootdir/usr/share/man/man1/v4l2loopback-ctl.1
~~~

(i'd rather discourage pasting "solution snippets" for poorly-defined problems; the line might be the solution to *your* problem, but the original description of the problem is not very accurate; so people might identify *their* problem as a dupe of yours and blindly paste your solution only to be suprprised of the outcome...)

downwater, 2020-06-22:
That's it, and now I have a perfectly working package. :-)
(Added also a `/sbin/depmod -a` in the installation script).