v4l2loopback/v4l2loopback #318: Can't build v4l2loopback on RaspberryPi

SatoshiKawabata, 2020-06-20:
I'm trying to build v4l2loopback on RaspberryPi.

My RaspberryPi environment.

```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

I cloned the repo and `make`.

```
git clone https://github.com/umlaeute/v4l2loopback.git
cd v4l2loopback
make
```

I got this error and build failed. 

```
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/pi/v4l2loopback modules
make[1]: *** /lib/modules/5.4.45-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```

I reffered #162 . I installed kernel headers by [raspberrypi-kernel-headers](https://www.raspberrypi.org/documentation/linux/kernel/headers.md) and tried `make` again. But the same error occurred. Please help me.

umlaeute, 2020-06-21:
you most likely installed the wrong headers (not those matching your exact kernel version).

closed 2020-06-21 by umlaeute

SatoshiKawabata, 2020-06-21:
@umlaeute How did you know so?
> you most likely installed the wrong headers

umlaeute, 2020-06-21:
because that's the only reason why the given directory does not exist.

what happens is:
1. v4l2loopback's Makefile determines the current (*running*) kernel by calling `uname -r`
2. it uses this value to change into the kernel-header directory:
~~~
make -C /lib/modules/`uname -r`/build M=/home/pi/v4l2loopback modules
~~~
3. in your case this fails (since this directory does not exist).

ergo: you don't have the kernel-headers for your currently running kernel installed.

either you installed the wrong headers (see the last paragraph in the documentation you linked to), or you are running the "wrong" kernel (e.g. because you did not reboot after installing a new kernel with matching headers)