v4l2loopback/v4l2loopback #319: Can we use this project

sediq-khan, 2020-06-23:
Hi,

We have an application written in Electron Js, which does not really access low-level functionalities. We would like to use your software to pick internal system audio and use it. I was wondering if your application is open source? or if we can use it for Windows and Mac. We would like to have it bundled with our application and install it as a pre-requisite using our installer. What do you think?

Thank you,
Sediq

kripton, 2020-06-23:
Hi, this project is not an "application". It's a module for the Linux kernel and as such it will most probably not work on Windows or macOS.
Besides that, if you're concerned about capturing `system audio`, I'd recommend using gstreamer and streaming it via a network socket (TCP or UDP) from gstreamer to your application. gstreamer might also work on Windows and macOS.
v4l2loopback is targeted mainly at video frames, I'm not sure if it supports audio.

umlaeute, 2020-06-23:
you might want to check the [COPYING](https://github.com/umlaeute/v4l2loopback/blob/master/COPYING) file for the license of this project.

apart from that, @kripton is right.
and no, it doesn't support audio.

closed 2020-06-23 by umlaeute