v4l2loopback/v4l2loopback #321: v4l2loopback required dependency for iriun webcam: doesn't work under kernel 5.7.5 and 5.7.6

DanielPrjct, 2020-06-26:
Hi! Arch Linux user here...

[daniel@hardwork ~]$ uname -a
Linux hardwork 5.7.6-arch1-1 #1 SMP PREEMPT Thu, 25 Jun 2020 00:14:47 +0000 x86_64 GNU/Linux

[daniel@hardwork ~]$ sudo dmesg  | grep -i v4l2loopback
v4l2loopback driver version 0.12.5 loaded

About one month ago, I've installed v4l2loopback-dkms-git 0.12.4.r0.g9bd5553-1 from AUR to satisfy iriunwebcam-bin dependencies; bluez and qt5-base iriun dependencies are previously installed in my system too (I'm using LXQt DE). Iriun Webcam used to run without issues.

After last kernel upgrade to linux 5.7.5.arch1-1, Iriun Webcam for Linux cann't connect to my phones by means wifi nor by USB cable.
[UPDATE] Upgraded kernel to linux 5.7.6.arch1-1, issue continues and there is not changes: iriun fails…

I've tried, after reboot, reinstalling v4l2loopback-dkms-git and iriunwebcam-bin but issue is not solved, although v4l2loopback dkms module is deleted, compiled and reinstalled again.

Iriun webcam window in PC shows a message: "Initialization failed. You may want to run: sudo modprobe v4l2loopback exclusive_caps=1".
But executing iriunwebcam after [daniel@hardwork ~]$ sudo modprobe v4l2loopback exclusive_caps=1 ; it doesn't solve this issue.
In the other hand, /etc/modprobe.d/iriunwebcam-options.conf has only one line: options v4l2loopback exclusive_caps=1 devices=1 card_label="Iriun Webcam","Iriun Webcam #2","Iriun Webcam #3","Iriun Webcam #4". So "v4l2loopback exclusive_caps=1" option is just included in iriun configuration…

I suspect that this issue is v4l2loopback module related, but I haven't knowledge on how to solve it.

Your help will be welcome! Best regards, DanielPrjct

EmilyMaryon, 2020-06-28:
I'm also having this problem, on version 5.7.4-arch1-1, and I have tried the same things as Daniel. Any tips would be much appreciated!

marcelkorpel, 2020-06-28:
And I also tested using
```
5.7.2-zen1-1-zen #1 ZEN SMP PREEMPT Wed, 10 Jun 2020 20:36:27 +0000 x86_64 GNU/Linux
```
but experienced the same issue.

@DanielPrjct Can you confirm everything works all right when using a kernel <5.7 with the same versions of v4l2loopback and iriun webcam? I don't have anything older than 5.7.2 at hand, so I can't test. It might be that it's not the kernel, but any of the other applications.

DanielPrjct, 2020-06-28:
@Marcel:
I'm sorry, I havn't any system but Arch Linux w/ 5.7.2 kernel.
Soon I will try virtualizing Debian stable under VirtualBox, and will
inform if iriun and v4l2loopback DKMS module works well on it.
Best regards, Daniel

marcelkorpel, 2020-06-28:
Wait, are you saying that you do have a working setup using Linux 5.7.2? I can't get it working with that version.

DanielPrjct, 2020-06-28:
@Marcel:
Not, I can NOT run iriun webcam with v4l2loopback DKMS under kernel 5.7.2
(nor 5.7.1).
Perhaps virtualizing another distro (Debian 10, Ubuntu 20.04) with a
previous kernel version (4.19.x), will let me test if this issue is kernel
related...

mavr76, 2020-06-29:
Same problem in Ubuntu 19.10.

I find that v4l2loopback in ubuntu log something related to module sign:
`sudo dmesg  | grep -i v4l2loopback
[    3.818240] v4l2loopback: loading out-of-tree module taints kernel.
[    3.818286] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
[    3.818657] v4l2loopback driver version 0.12.2 loaded
`

umlaeute, 2020-06-29:
@mavr76 your problem is unrelated (and a duplicate of #306). please remove your comment from this thread (or i will do it), to keep the issue concise (not every "doesn't work here" is the same as *my* "doesn't work")

umlaeute, 2020-06-29:
just as a data-point: the module works nicely with linux-5.7.0 on Debian/sid:

~~~
$ uname -a
Linux umlautS 5.7.0-1-amd64 #1 SMP Debian 5.7.6-1 (2020-06-24) x86_64 GNU/Linux
~~~

i haven't tried iriun, just with a simple GStreamer-pipeline.

umlaeute, 2020-06-29:
so i did a few more tests.

i grabbed iriun's "Webcam for Ubuntu", ran it and low and behold i got the error:
>  Initialization failed. You may want to run: sudo modprobe v4l2loopback exclusive_caps=1

inspecting the problem a bit, it seems that iriunwebcam fails to detect the camera *unless* the loopback device is labeled `Iriun Webcam`.
So besides setting the `exclusive_flags` (which is actually *not* required by iriunwebcam, but i guess it is provided as a courtesy for people who want to use the loopback-device in their browser), you really **must** set the `card_label="Iriun Webcam"`

i'm inclined to close this as a problem on Iriun's side.
(though i haven't actually installed their app on my mobile to see whether it really works; i get as far as "No phone available" which is past the point of "Initialization failed").

if the problem is indeed related to v4l2oopback (and/or it's support for kernel-5.7.x) let's re-open this (or a new ticket)

@DanielPrjct  wrote:
> I suspect that this issue is v4l2loopback module related, [...]

i would be very interested how you come to this conclusion.
since they are making actual money (it seems) with a closed-source application, i would suggest to first try their resources on solving this problem.

closed 2020-06-29 by umlaeute

sankarshandev98, 2021-10-03:
try to disable secure boot option on your bios menu.

gregovilardo, 2021-11-17:
Hi i has the same problem in manjaro, i solve it with 'sudo pacman -Syu $KERNEL-headers' as it is here https://forum.manjaro.org/t/root-tip-get-iriun-webcam-working-on-manjaro/33738, but it seems this header problems is only in manjaro not for plain arch but maybe it serves you

DrI-T, 2022-02-13:
Don't know if it will help some of you ...
Every time I run iriunwebcam; I have do a "cleanup" of v4l2loopback device running:
```
sudo rmmod v4l2loopback; sudo modprobe v4l2loopback exclusive_caps=1
```
otherwise the second time iriun doesn't work

NathanPW12, 2022-02-15:
> Não sei se vai ajudar alguns de vocês... Toda vez que eu corro iriunwebcam; Eu fiz uma "limpeza" do dispositivo v4l2loopback em execução:
> 
> ```
> sudo rmmod v4l2loopback; sudo modprobe v4l2loopback exclusive_caps=1
> ```
> 
> caso contrário, a segunda vez que o iriun não funciona

Thank youuuuuuuuuuuuu!

geraldohomero, 2022-06-26:
To make it easy for us i made this guide on how to set a bash script 

> Don't know if it will help some of you ... Every time I run iriunwebcam; I have do a "cleanup" of v4l2loopback device running:
> 
> ```
> sudo rmmod v4l2loopback; sudo modprobe v4l2loopback exclusive_caps=1
> ```
> 
> otherwise the second time iriun doesn't work

To make it easy for us i made this guide on how to set a bash script for the Iriun webcam

https://github.com/geraldohomero/automate_iriunwebcam_Linux

BenBE, 2022-06-26:
> Don't know if it will help some of you ... Every time I run iriunwebcam; I have do a "cleanup" of v4l2loopback device running:
> 
> ```
> sudo rmmod v4l2loopback; sudo modprobe v4l2loopback exclusive_caps=1
> ```
> 
> otherwise the second time iriun doesn't work

This strongly hints at issues #470, which gets resolved by applying the patches from #477 (bugfix) and #485 (resolving a compatibility issue reported in #484).

marcosportes, 2024-05-02:
The problem I had with v4l2loopback on IriumWebcam downloaded from AUR was resolved with the following commands:

1. sudo pacman -Syu dkms base-devel --needed
2. sudo pacman -S $(ls /boot | awk -F "-" '/^linux/ { print $1"-headers" }')\n
3. sudo pacman -S v4l2loopback-dkms
4. restart
5. sudo pamac install irunwebcam-bin

marcosportes, 2024-05-02:
I use Manjaro Linux. The problem I had with v4l2loopback on IriumWebcam downloaded from AUR was resolved with the following commands:

1. sudo pacman -Syu dkms base-devel --needed
2. sudo pacman -S $(ls /boot | awk -F "-" '/^linux/ { print $1"-headers" }')\n
3. sudo pacman -S v4l2loopback-dkms
4. restart
5. sudo pamac install irunwebcam-bin