v4l2loopback/v4l2loopback #322: Add support for split output/capture devices.

normanr, 2020-07-02:
Being able to attach the consumer before the producer is not possible with `exclusive_caps=1`. If the consumer if picky about the device capabilities, then it's not possible to revert to `exclusive_caps=0`. So this feature request proposal is to create device pairs instead of single devices, eg:

with exclusive_caps=0:
`/dev/video0 Capabilities: Video Capture, Video Output`

with exclusive_caps=1:
`/dev/video0 Capabilities: Video Output`
and once a producer is attached
`/dev/video0 Capabilities: Video Capture`

with this proposal (which would add a new flag or change exclusive_caps to an int and maybe use the value 2):
`/dev/video0 Capabilities: Video Output`
`/dev/video1 Capabilities: Video Capture`
in this mode `video_nr` and `card_label` would be `devices` * 2 in length with pairs of values being consumed for each output/capture device pair and a given device's capabilities would still be exclusive, but would not change once it was created.

This would also make sense if the linux kernel implemented support for Video4Linux device namespaces (similar to how network namespaces have `veth`, Video4Linux has `v4l2loopback`). You might want to run the producer and consumer in different device namespaces, so having two different devices is required for that.

umlaeute, 2020-07-03:
i find split devices ugly as hell.
however, it *does* solve a few problems, so i might consider it.

the actual API how to create such devices, is probably going to be coupled to the "dynamic device management" (see #282). in any case its not something that will be tackled before the dynamic device management works properly.

normanr, 2020-07-04:
Acknowledged. That makes a lot of sense. I opened this issue instead of commenting on the one for dynamic device management to avoid (what would otherwise have been) an off topic request.