v4l2loopback/v4l2loopback #323: [develop] dynamic device management does not play nicely with other video-devices

umlaeute, 2020-07-02:
now that there is dynamic device management in the develop-branch (resolving #282 ), problem start to emerge...

e.g. on my laptop (that has a real builtin webcam that is using `/dev/video0` and `/dev/video1`), i get kernel-oopses when running `utils/v4l2loopback query`), even if i'm not trying to access the non-loopback devices!

~~~
[59698.344241] v4l2loopback driver version 0.12.5 loaded
[59710.175415] BUG: unable to handle page fault for address: 00007ffc68b9c2c0
[59710.175422] #PF: supervisor read access in kernel mode
[59710.175425] #PF: error_code(0x0001) - permissions violation
[59710.175428] PGD 80000002fc92f067 P4D 80000002fc92f067 PUD 2da68d067 PMD 309664067 PTE 80000003af50b867
[59710.175438] Oops: 0001 [#2] SMP PTI
[59710.175444] CPU: 1 PID: 110557 Comm: v4l2loopback Tainted: G      D W  OE     5.7.0-1-amd64 #1 Debian 5.7.6-1
[59710.175447] Hardware name: LENOVO 20HMS3YW00/20HMS3YW00, BIOS R0IET42W (1.20 ) 08/09/2017
[59710.175458] RIP: 0010:v4l2loopback_control_ioctl+0xaf/0x190 [v4l2loopback]
[59710.175463] Code: 41 5d c3 48 8d 74 24 04 48 89 ef e8 bb f8 ff ff 4c 63 e0 85 c0 78 be 4c 63 64 24 04 eb b7 49 c7 c4 ea ff ff ff 48 85 ed 74 ab <48> 63 75 00 48 c7 c7 60 c1 75 c1 e8 a1 90 ef c2 48 89 c3 48 85 c0
[59710.175466] RSP: 0018:ffffbe0482d83ec8 EFLAGS: 00010206
[59710.175469] RAX: 0000000000000000 RBX: 0000000000004c82 RCX: 00007ffc68b9c2c0
[59710.175472] RDX: ffff97f453270fc0 RSI: 0000000000004c82 RDI: ffffffffc175c140
[59710.175474] RBP: 00007ffc68b9c2c0 R08: 0000000000000003 R09: 0000000000000000
[59710.175475] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffffffffffea
[59710.175477] R13: 00007ffc68b9c2c0 R14: 0000000000000003 R15: ffff97f457509f00
[59710.175480] FS:  00007f5b1979b540(0000) GS:ffff97f49f480000(0000) knlGS:0000000000000000
[59710.175482] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[59710.175484] CR2: 00007ffc68b9c2c0 CR3: 0000000236da0001 CR4: 00000000003606e0
[59710.175487] Call Trace:
[59710.175498]  ksys_ioctl+0x87/0xc0
[59710.175503]  __x64_sys_ioctl+0x16/0x20
[59710.175509]  do_syscall_64+0x52/0x180
[59710.175514]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[59710.175518] RIP: 0033:0x7f5b196cb457
[59710.175521] Code: 00 00 90 48 8b 05 39 7a 0c 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 09 7a 0c 00 f7 d8 64 89 01 48
[59710.175524] RSP: 002b:00007ffc68b9c2a8 EFLAGS: 00000202 ORIG_RAX: 0000000000000010
[59710.175527] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007f5b196cb457
[59710.175528] RDX: 00007ffc68b9c2c0 RSI: 0000000000004c82 RDI: 0000000000000003
[59710.175530] RBP: 00007ffc68b9c310 R08: 1999999999999999 R09: 0000000000000000
[59710.175532] R10: fffffffffffff5cf R11: 0000000000000202 R12: 0000560be5a67110
[59710.175534] R13: 00007ffc68b9c480 R14: 0000000000000000 R15: 0000000000000000
[59710.175538] Modules linked in: v4l2loopback(OE) sctp xt_nat xt_tcpudp veth xt_conntrack xt_MASQUERADE nf_conntrack_netlink xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 br_netfilter bridge stp llc overlay vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc snd_seq_dummy snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device ctr ccm bnep nf_tables nfnetlink cpufreq_powersave cpufreq_userspace cpufreq_conservative btusb btrtl btbcm btintel bluetooth drbg ansi_cprng ecdh_generic ecc intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel snd_hda_codec_hdmi kvm snd_soc_skl snd_soc_hdac_hda snd_hda_ext_core snd_soc_sst_ipc snd_soc_sst_dsp snd_soc_acpi_intel_match snd_soc_acpi snd_hda_codec_realtek irqbypass snd_soc_core snd_hda_codec_generic snd_compress ghash_clmulni_intel snd_hda_intel
[59710.175582]  snd_intel_dspcfg iwlmvm tps6598x typec mac80211 snd_hda_codec aesni_intel regmap_i2c libaes snd_hda_core crypto_simd snd_hwdep snd_pcm_oss libarc4 binfmt_misc cryptd glue_helper snd_mixer_oss intel_cstate nls_ascii intel_uncore nls_cp437 iwlwifi joydev mei_me vfat snd_pcm intel_rapl_perf efi_pstore fat wmi_bmof cfg80211 serio_raw pcspkr efivars intel_wmi_thunderbolt snd_timer mei iTCO_wdt iTCO_vendor_support watchdog thinkpad_acpi mmc_block intel_pch_thermal tpm_crb i2c_multi_instantiate nvram ledtrig_audio snd soundcore ac rfkill tpm_tis tpm_tis_core tpm acpi_pad evdev rng_core parport_pc ppdev lp parport nfsd loop auth_rpcgss dm_crypt nfs_acl lockd grace dm_mod sunrpc efivarfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 btrfs blake2b_generic zstd_decompress zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c crc32c_generic raid1 raid0 multipath linear md_mod i915 i2c_algo_bit drm_kms_helper cec
[59710.175635]  i2c_designware_platform rtsx_pci_sdmmc mmc_core i2c_designware_core nvme xhci_pci nvme_core t10_pi crc32_pclmul e1000e crc_t10dif psmouse xhci_hcd crc32c_intel drm crct10dif_generic ptp pps_core i2c_i801 crct10dif_pclmul crct10dif_common usbcore rtsx_pci intel_lpss_pci intel_lpss idma64 mfd_core usb_common wmi battery video button [last unloaded: v4l2loopback]
[59710.175657] CR2: 00007ffc68b9c2c0
[59710.175661] ---[ end trace f120bf02253018e3 ]---
[59710.352593] RIP: 0010:v4l2loopback_control_ioctl+0xaf/0x190 [v4l2loopback]
[59710.352600] Code: 41 5d c3 48 8d 74 24 04 48 89 ef e8 bb f8 ff ff 4c 63 e0 85 c0 78 be 4c 63 64 24 04 eb b7 49 c7 c4 ea ff ff ff 48 85 ed 74 ab <48> 63 75 00 48 c7 c7 60 c1 75 c1 e8 a1 90 ef c2 48 89 c3 48 85 c0
[59710.352602] RSP: 0018:ffffbe04837dbec8 EFLAGS: 00010206
[59710.352604] RAX: 0000000000000000 RBX: 0000000000004c82 RCX: 00007ffd58e53ac0
[59710.352605] RDX: ffff97f389432f40 RSI: 0000000000004c82 RDI: ffffffffc175c140
[59710.352606] RBP: 00007ffd58e53ac0 R08: 0000000000000003 R09: 0000000000000000
[59710.352607] R10: 0000000000000000 R11: 0000000000000000 R12: ffffffffffffffea
[59710.352608] R13: 00007ffd58e53ac0 R14: 0000000000000003 R15: ffff97f41f54c000
[59710.352609] FS:  00007f5b1979b540(0000) GS:ffff97f49f480000(0000) knlGS:0000000000000000
[59710.352610] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[59710.352611] CR2: 00007ffc68b9c2c0 CR3: 0000000236da0001 CR4: 00000000003606e0
~~~

normanr, 2020-07-04:
What device number are the dynamically added device's using? It looks like the default is 0, which would cause the loopback device to be installed over the real first device.

umlaeute, 2020-07-04:
hmm, no. actually we only pretend to use the device-number but instead use an internal number (which is a bug in its own right).

anyhow, this issue is mainly here so i don't forget about it.

umlaeute, 2020-07-04:
hmm, while a6bc4cb  was supposed to fix this, it seems there are sill problems (worked ok in VirtualBox, but not on bare metal)

normanr, 2020-07-05:
Can you reproduce the failure on demand? http://haneensa.github.io/2018/05/28/kerneldebug1/ seems to be a concise article on how to figure out which line of code is responsible (I have no idea if it's helpful or not, so apologies if it isn't helpful).

closed 2020-10-15 by umlaeute