v4l2loopback/v4l2loopback #325: v4l2scr -> v4l2sink  = error

costads, 2020-07-09:
### Step 2: Describe your environment

  * `v4l2loopback` version: _____

         v4l2loopback driver version 0.12.1 loaded

  * kernel version: _____

          Linux raspberrypi 4.19.118-v7l+

  * Distribution (+version): _____

No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

  

### Step 3: Describe the problem:

Trying to forward my rearcams signal on video0 to video5

#### Steps to reproduce:

gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video5
is working fine.
but when I try with:
gst-launch-1.0 -v v4l2src device=/dev/video0 ! v4l2sink device=/dev/video5 

I get this error:

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)

I have tried several other ways found online, but no matter what I get above error.

I have the same problem as well when I try with FFmpeg, f.ex.
ffmpeg -i /dev/video0 -f v4l2 -pix_fmt yuv420p /dev/video5

Then I get errors like:
Dequeued v4l2 buffer contains 687112 bytes, but 691200 were expected. Flags: 0x00002001.

So I am thinking the only thing in common is v4l2loopback, but ofcourse I am not sure.
I am a total newbie when it comes to all this Linux stuff

   
#### Expected Results:

  Video output on /video5

umlaeute, 2020-07-10:
it might be that the initial call with `videotestsrc` locked the device to some resolution/format that neither your ffmpeg-call nor the gstreamer-pipeline with `v4l2src` manage to provide.

so what's the output of the following command?
~~~sh
grep . /sys/devices/virtual/video4linux/video*/format
~~~

you could also try to insert a videoconverter:
~~~
gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! v4l2sink device=/dev/video5 
~~~

costads, 2020-07-10:
Thanks, seems like the videoconvert is helping on it, however, it doesn't work with the software I am using, resulting in a black screen on /video5

Do you have any idea about the FFmpeg error?
ffmpeg -i /dev/video0 -f v4l2 -pix_fmt yuv420p /dev/video5
Dequeued v4l2 buffer contains 687112 bytes, but 691200 were expected. Flags: 0x00002001.

It's working fine on RPI 3B+, but not on V4.

Just wondering if it is a FFmpeg or v4l2loopback error.

costads, 2020-07-10:
I borrowed another grabber, Hauppauge Live 2, and with that everything is working fine:
ffmpeg -i /dev/video0 -f v4l2 -pix_fmt yuv420p /dev/video5 

So I don't know if it is the stk1160 driver that doesn't play well, or how it's all connected.

closed 2020-07-16 by umlaeute