v4l2loopback/v4l2loopback #326: Problems with RPI4 kernel 5.4.51+

albertfaromatics, 2020-07-24:
### Step 2: Describe your environment

  * `v4l2loopback` version:  0.12.1

  * kernel version: 5.4.51+

  * Distribution (+version): Raspbian Buster (Rpi4)

### Step 3: Describe the problem:

After upgrading the kernel of the RPi to the newest version, and installing headers with rpi-source, the module fails to compile with error:

`
DKMS make.log for v4l2loopback-0.12.1 for kernel 5.4.51+ (armv7l)
Fri 24 Jul 18:22:26 BST 2020
Building v4l2-loopback driver...
make -C /lib/modules/5.4.51+/build M=/var/lib/dkms/v4l2loopback/0.12.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.51+'
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.c: In function ‘vidioc_qbuf’:
/var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.c:1509:4: error: implicit declaration of function ‘v4l2_get_timestamp’; did you mean ‘v4l2_get_subdevdata’? [-Werror=implicit-function-declaration]
    v4l2_get_timestamp(&b->buffer.timestamp);
    ^~~~~~~~~~~~~~~~~~
    v4l2_get_subdevdata
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:266: /var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.o] Error 1
make[1]: *** [Makefile:1709: /var/lib/dkms/v4l2loopback/0.12.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.51+'
make: *** [Makefile:43: v4l2loopback.ko] Error 2
`

Reverting back to 4.19.118+ fixed the issue

hutber, 2020-07-26:
I also had the same problem:

```
hutber@hutber:/var/www/v4l2loopback-master$ sudo modprobe v4l2loopback  && gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video2
ffmpeg version N-49161-g50e194e6e1-static https://johnvansickle.com/ffmpeg/  Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 6.3.0 (Debian 6.3.0-18+deb9u1) 20170516
  configuration: --enable-gpl --enable-version3 --enable-static --disable-debug --disable-ffplay --disable-indev=sndio --disable-outdev=sndio --cc=gcc-6 --enable-fontconfig --enable-frei0r --enable-gnutls --enable-gmp --enable-gray --enable-libaom --enable-libfribidi --enable-libass --enable-libvmaf --enable-libfreetype --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-librubberband --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libvorbis --enable-libopus --enable-libtheora --enable-libvidstab --enable-libvo-amrwbenc --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libdav1d --enable-libxvid --enable-libzvbi --enable-libzimg
  libavutil      56. 30.100 / 56. 30.100
  libavcodec     58. 53.101 / 58. 53.101
  libavformat    58. 28.101 / 58. 28.101
  libavdevice    58.  7.100 / 58.  7.100
  libavfilter     7. 55.100 /  7. 55.100
  libswscale      5.  4.101 /  5.  4.101
  libswresample   3.  4.100 /  3.  4.100
  libpostproc    55.  4.100 / 55.  4.100
Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.

*** Error ***              
This camera cannot capture previews.
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)
pipe:: Invalid data found when processing input
```

umlaeute, 2020-07-26:
@albertfaromatics that was fixed with 0.12.2 (please always try with the latest and greatest version of v4l2loopback)

@hutber i don't see how your problem is in *any* way related to not being able to compile the kernel module (please either clarify the relation, or remove your comment; or i will do the latter)

closed 2020-07-26 by umlaeute

albertfaromatics, 2020-07-27:
You are right. Tested and working

Was using apt install version of v4l2loopback-dkms, still downloading 0.12.1

Thanks @umlaeute

La-te, 2020-09-17:
@umlaeute As said, 0.12.1 is still what is coming from the Debian/Raspbian repositories. I can see that umlaeute@debian.org is the package maintainer. I don't really know how this whole process works. Who should be notified and who could work to get the working version to the repositories?

umlaeute, 2020-09-18:
> I can see that umlaeute@debian.org is the package maintainer

so? no clue how this is relevant here.

> Who should be notified and who could work to get the working version to the repositories?

honestly, i don't know who should be notified, most likely the Raspbian distribution.
Debian/buster was released with linux/4.19.0, which works nicely with the `v4l2loopback` package that ships with Debian/buster.
Raspbian updated the kernel they shipped with Raspbian/buster to something newer - obviously without checking whether the new kernel plays well with all the kernel-modules they provide (and if it doesn't whether there are newer versions of the kernel-modules packaged in Debian/testing).

I noticed that Debian/buster-backports now also ships with linux/5.7.0, so I uploaded a newer package of v4l2loopback to Debian/buster-backports.
Whether Raspbian/buster will pick up Debian/buster-backports, is beyond my knowledge.