v4l2loopback/v4l2loopback #329: mjpg-streamer "Unable to query buffer"

mikulik86, 2020-08-02:
Ihave a Raspberry Pi 4 B running Raspbian 10 Buster + the Raspberry Pi High Quality Camera. 
I need to use v4l2loopback to copy videofeed from the physical Camera (/dev/video0) to dummy device (/dev/video1) in order to access it by the mjpg-streamer and the v4l2rtspserver at the same time.

I load the v4l2loopback module:

`sudo modprobe v4l2loopback`

Then copy the video from /dev/video0 to /dev/video1:

`ffmpeg -f video4linux2 -input_format mjpeg -i /dev/video0 -vcodec copy -f v4l2 /dev/video1`

Launch the mjpg-streamer and get following error:

````
./mjpg_streamer -i "./input_uvc.so -d /dev/video1" -o "./output_http.so"
MJPG Streamer Version: git rev: 85f89a8c321e799fabb1693c5d133f3fb48ee748
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
Unable to query buffer: Invalid argument
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
````

Launching the mjpg-streamer on /dev/video0 works flawlessly.

I spent hours trying to find solution. Please help.

umlaeute, 2020-08-03:
i don't know `mjpg_streamer` and from your cmdline it seems to be all very self-compiled.
have you asked the `mjpg_streamer` authors for advise (after all, it is `mjpg_streamer` that fails, rather than `v4l2loopback`)?

mikulik86, 2020-08-04:
Actually asked, see: https://github.com/jacksonliam/mjpg-streamer/issues/264#issuecomment-667682275

I really don't understand programming, so i have no idea on which side the problem lies.
But you are my last resort, please help.

tachang, 2020-12-04:
FYI I've created a version that works with v4l2loopback https://github.com/openmiko/mjpg-streamer

umlaeute, 2020-12-05:
is that PR https://github.com/jacksonliam/mjpg-streamer/pull/280?

if so: cool, thanks

closed 2020-12-05 by umlaeute

tachang, 2020-12-05:
Correct.

jacksonliam, 2020-12-18:
Hi, @umlaeute  - maintainer of mjpg-streamer here.

Do you know why the simple `VIDIOC_QUERYBUF` as seen https://github.com/jacksonliam/mjpg-streamer/blob/925337f541ddbb061ade0bee5215299f7834d8ea/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c#L447

Would be failing on v4l2loopback? I believe you are returning `INVALID ARGUMENT`. FYI we use 4 buffers by default, I haven't looked in a debugger to see which of the 4 is failing.

All the video devices I've seen (physical or virtual) have no problems with this query. So im interested to find out if v4l2loopback is not working to spec, or if mjpg-streamer is doing something it shouldn't!

deftdawg, 2021-06-23:
> FYI I've created a version that works with v4l2loopback https://github.com/openmiko/mjpg-streamer

@tachang , I tried the `v4l2loopback` branch of your repo, it doesn't build (`#include <errno.h>` is missing), the plugin help command doesn't work (`./mjpg_streamer -i "input_v4l2loopback.so --help"`) and I'm looking at the help() function I'm not even sure it matches with the args... 

I'm trying to solve the problem of roating my old C270 180 degrees;  imaginging the solution is  `Logitech C270 (mjpeg/yuyv; doesn't support v42ctl vflip)` -> `ffmpeg filter:vflip` -> `v4l2loopback (only accepts raw)` -> `mjpg-streamer (doesn't like v4l2loopback)`... wondering if your plugin can help?