v4l2loopback/v4l2loopback #331: Stack trace in /var/log/syslog

darkpixel, 2020-08-06:
* `v4l2loopback` version: aba3067f81b343f4e80c588de895c1aeb0da4b76
  * kernel version: 5.4.0-42-generic
  * Distribution (+version): Linux Mint 20

I get the following stack trace when running ```sudo modprobe v4l2loopback exclusive_caps=1```
```
Aug  6 14:27:46 iridal kernel: [15487.846176] ------------[ cut here ]------------
Aug  6 14:27:46 iridal kernel: [15487.846180] WARNING: CPU: 6 PID: 143718 at drivers/media/v4l2-core/v4l2-ioctl.c:1069 v4l_querycap+0x8f/0xa0 [videodev]
Aug  6 14:27:46 iridal kernel: [15487.846180] Modules linked in: v4l2loopback(OE) tcp_diag inet_diag xt_nat xt_tcpudp veth xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bpfilter br_netfilter bridge stp llc rfcomm aufs cmac algif_hash algif_sk
cipher overlay af_alg bnep binfmt_misc zfs(PO) zunicode(PO) intel_rapl_msr zavl(PO) intel_rapl_common icp(PO) nls_iso8859_1 uvcvideo zcommon(PO) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 znvpair(PO) snd_hda_codec_realtek videobuf2_common spl(O) isst_if_common snd_hda_codec_generic zlua(PO) videodev ledtrig_audio snd_hda_codec_hdmi snd_hda_intel btusb sn
d_intel_dspcfg btrtl snd_hda_codec snd_usb_audio btbcm iwlmvm btintel snd_hda_core nfit snd_usbmidi_lib bluetooth snd_hwdep mc input_leds mac80211 x86_pkg_temp_thermal snd_seq_midi intel_powerclamp snd_seq_midi_event coretemp ecdh_generic snd_rawmidi libarc4 ecc snd_pcm kvm_intel snd_seq kvm iwlwifi
Aug  6 14:27:46 iridal kernel: [15487.846195]  snd_seq_device intel_cstate snd_timer intel_rapl_perf intel_wmi_thunderbolt snd ioatdma mei_me cfg80211 soundcore mei dca mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs xor zstd_compress hid_generic usbhid hid raid6_pq libcrc32c dm_mirror dm_region_hash dm_log uas usb_storage crct
10dif_pclmul amdgpu crc32_pclmul ghash_clmulni_intel amd_iommu_v2 gpu_sched i2c_algo_bit ttm mxm_wmi drm_kms_helper syscopyarea sysfillrect sysimgblt aesni_intel crypto_simd cryptd fb_sys_fops glue_helper drm e1000e ahci i2c_i801 libahci wmi [last unloaded: v4l2loopback]
Aug  6 14:27:46 iridal kernel: [15487.846205] CPU: 6 PID: 143718 Comm: chrome Tainted: P        W  OE     5.4.0-42-generic #46-Ubuntu
Aug  6 14:27:46 iridal kernel: [15487.846205] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./X299 Killer SLI/ac, BIOS P1.20 07/20/2017
Aug  6 14:27:46 iridal kernel: [15487.846208] RIP: 0010:v4l_querycap+0x8f/0xa0 [videodev]
Aug  6 14:27:46 iridal kernel: [15487.846208] Code: 00 00 80 48 b9 00 00 20 00 00 00 20 00 48 0b 4b 54 21 d6 39 f2 75 13 48 89 4b 54 5b 41 5c 41 5d 41 5e 41 5f 5d c3 0f 0b eb d0 <0f> 0b 48 89 4b 54 5b 41 5c 41 5d 41 5e 41 5f 5d c3 0f 1f 44 00 00
Aug  6 14:27:46 iridal kernel: [15487.846209] RSP: 0018:ffffaa91e929fc70 EFLAGS: 00010202
Aug  6 14:27:46 iridal kernel: [15487.846209] RAX: 0000000000000000 RBX: ffffaa91e929fd88 RCX: 8520800285208002
Aug  6 14:27:46 iridal kernel: [15487.846210] RDX: 0000000085008003 RSI: 0000000085008002 RDI: 0000000000000000
Aug  6 14:27:46 iridal kernel: [15487.846210] RBP: ffffaa91e929fc98 R08: 0000000000000000 R09: 0000000000ffff0a
Aug  6 14:27:46 iridal kernel: [15487.846211] R10: 0000000000000000 R11: 0000000000000000 R12: ffff9ea319bbf000
Aug  6 14:27:46 iridal kernel: [15487.846211] R13: ffff9ea641e3a400 R14: ffff9ea67a7e86a0 R15: ffffffffc1681380
Aug  6 14:27:46 iridal kernel: [15487.846212] FS:  00007f7771c92d80(0000) GS:ffff9ea77fd80000(0000) knlGS:0000000000000000
Aug  6 14:27:46 iridal kernel: [15487.846212] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Aug  6 14:27:46 iridal kernel: [15487.846213] CR2: 0000559cd2b41a20 CR3: 00000016b25d8001 CR4: 00000000003606e0
Aug  6 14:27:46 iridal kernel: [15487.846213] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
Aug  6 14:27:46 iridal kernel: [15487.846213] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
Aug  6 14:27:46 iridal kernel: [15487.846214] Call Trace:
Aug  6 14:27:46 iridal kernel: [15487.846217]  __video_do_ioctl+0x1a7/0x410 [videodev]
Aug  6 14:27:46 iridal kernel: [15487.846219]  video_usercopy+0x2b5/0x670 [videodev]
Aug  6 14:27:46 iridal kernel: [15487.846222]  ? v4l_s_fmt+0x670/0x670 [videodev]
Aug  6 14:27:46 iridal kernel: [15487.846225]  video_ioctl2+0x15/0x20 [videodev]
Aug  6 14:27:46 iridal kernel: [15487.846228]  v4l2_ioctl+0x4c/0x60 [videodev]
Aug  6 14:27:46 iridal kernel: [15487.846229]  do_vfs_ioctl+0x407/0x670
Aug  6 14:27:46 iridal kernel: [15487.846230]  ? putname+0x4a/0x50
Aug  6 14:27:46 iridal kernel: [15487.846231]  ksys_ioctl+0x67/0x90
Aug  6 14:27:46 iridal kernel: [15487.846231]  __x64_sys_ioctl+0x1a/0x20
Aug  6 14:27:46 iridal kernel: [15487.846233]  do_syscall_64+0x57/0x190
Aug  6 14:27:46 iridal kernel: [15487.846234]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Aug  6 14:27:46 iridal kernel: [15487.846234] RIP: 0033:0x7f7772ef337b
Aug  6 14:27:46 iridal kernel: [15487.846235] Code: 0f 1e fa 48 8b 05 15 3b 0d 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 0f 1f 44 00 00 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d e5 3a 0d 00 f7 d8 64 89 01 48
Aug  6 14:27:46 iridal kernel: [15487.846235] RSP: 002b:00007fffba336cc8 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
Aug  6 14:27:46 iridal kernel: [15487.846236] RAX: ffffffffffffffda RBX: 0000000000000017 RCX: 00007f7772ef337b
Aug  6 14:27:46 iridal kernel: [15487.846236] RDX: 00007fffba336d50 RSI: ffffffff80685600 RDI: 0000000000000017
Aug  6 14:27:46 iridal kernel: [15487.846237] RBP: 00007fffba336e30 R08: 0000000000000000 R09: 00007fffba336838
Aug  6 14:27:46 iridal kernel: [15487.846237] R10: 0000000000000000 R11: 0000000000000246 R12: 00003b9b0d78fae0
Aug  6 14:27:46 iridal kernel: [15487.846238] R13: 00003b9b0d7a03c0 R14: 00007fffba336cf0 R15: 00003b9b0d79ffc0
Aug  6 14:27:46 iridal kernel: [15487.846239] ---[ end trace b63a53508dbdf3b3 ]---

```

#### Steps to reproduce:

```
 aaron   master  ~  code  v4l2loopback  make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/tank/code/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-42-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'
 aaron   master  ~  code  v4l2loopback  sudo make install
make -C /lib/modules/`uname -r`/build M=/tank/code/v4l2loopback modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-42-generic'
  INSTALL /tank/code/v4l2loopback/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.44
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'

SUCCESS (if you got 'SSL errors' above, you can safely ignore them)

 aaron   master  ~  code  v4l2loopback  sudo depmod -a
 aaron   master  ~  code  v4l2loopback  sudo rmmod v4l2loopback
rmmod: ERROR: Module v4l2loopback is not currently loaded
 aaron   master  ~  code  v4l2loopback  1  sudo modprobe v4l2loopback exclusive_caps=1
 aaron   master  ~  code  v4l2loopback  
```

darkpixel, 2020-08-06:
I also get the same error when I use the Mint package:
```
 aaron   master  ~  code  v4l2loopback  make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/tank/code/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-42-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'
 aaron   master  ~  code  v4l2loopback  sudo make install
make -C /lib/modules/`uname -r`/build M=/tank/code/v4l2loopback modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-42-generic'
  INSTALL /tank/code/v4l2loopback/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.44
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'

SUCCESS (if you got 'SSL errors' above, you can safely ignore them)

 aaron   master  ~  code  v4l2loopback  sudo depmod -a
 aaron   master  ~  code  v4l2loopback  sudo rmmod v4l2loopback
rmmod: ERROR: Module v4l2loopback is not currently loaded
 aaron   master  ~  code  v4l2loopback  1  sudo modprobe v4l2loopback exclusive_caps=1
 aaron   master  ~  code  v4l2loopback  
```

umlaeute, 2020-08-09:
hmm, this looks suspiciously like #257 (and its dupes #279 and #294)
are you sure that you are actually loading aba3067 (or v4l2loopback>=0.12.5), and not some leftover installation?

darkpixel, 2020-08-09:
I may have been.  I purged all the v4l2loopback packages and OBS, then compiled everything from source.  It seems to be working now.

closed 2020-08-09 by darkpixel