v4l2loopback/v4l2loopback #332: Build fails on kernel 5.7.11-100.fc31.x86_64 with error "modprobe: ERROR: could not insert 'v4l2loopback'

rdlmda, 2020-08-09:
I'm able to compile and load the module on kernel 5.7.9-100.fc31.x86_64, but not on 5.7.11-100.fc31.x86_64.

### Step 2: Describe your environment

  * `v4l2loopback` version: irrelevant, trying to build from repo master latest

  * kernel version: 5.7.11-100.fc31.x86_64

  * Distribution (+version): Fedora 31
  
### Step 3: Describe the problem:

#### Steps to reproduce:

  1. `make && sudo make install`
```
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/user/software/webcam-loopback/v4l2loopback modules
make[1]: Entering directory '/usr/src/kernels/5.7.11-100.fc31.x86_64'
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/kernels/5.7.11-100.fc31.x86_64'
make -C /lib/modules/`uname -r`/build M=/home/user/software/webcam-loopback/v4l2loopback modules_install
make[1]: Entering directory '/usr/src/kernels/5.7.11-100.fc31.x86_64'
  INSTALL /home/user/software/webcam-loopback/v4l2loopback/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.7.11-100.fc31.x86_64
make[1]: Leaving directory '/usr/src/kernels/5.7.11-100.fc31.x86_64'

SUCCESS (if you got 'SSL errors' above, you can safely ignore them)
```
 2.  `sudo depmod -a` 
 3. `sudo modprobe v4l2loopback`
```
modprobe: ERROR: could not insert 'v4l2loopback': Exec format error
```
  
#### Observed Results:

  Module didn't load
  
#### Expected Results:

  Module loaded

umlaeute, 2020-08-09:
duplicate of #313, #260, #208

closed 2020-08-09 by umlaeute

rdlmda, 2020-08-09:
I managed to get it working by:
- Removing `/usr/lib/modules/5.7.11-100.fc31.x86_64/extra/v4l2loopback.ko`
- Removing `v4l2loopback.ko`, `v4l2loopback.o`, `v4l2loopback.mod`, `v4l2loopback.mod.o`, `v4l2loopback.mod.c`
- Performing `make clean` before the compile steps on README.

rdlmda, 2020-09-27:
I had the same issue today (after a kernel update) and this time just by running `make clean` I was able to fix it.

pailoro, 2020-11-20:
Same problem here:

Operating System: Ubuntu 20.04.1 LTS
Kernel: Linux 5.4.0-53-generic