v4l2loopback/v4l2loopback #333: apt install v4l2loopback-utils fails with "Bad return status for module build on kernel: 5.4.0-42-generic (x86_64)"

DaveDixon, 2020-08-13:
* `v4l2loopback` version: 0.12.3-1
  * kernel version: 5.4.0-42-generic (x86_64)
  * Distribution (+version): Ubuntu 20.04.1 LTS
  1. sudo add-apt-repository ppa:obsproject/obs-studio
  2. sudo apt-get install obs-studio
  3. sudo apt install v4l2loopback-utils
  
#### Observed Results:

Command line output
----------------------------
Loading new v4l2loopback-0.12.3 DKMS files...
Building for 5.4.0-42-generic
Building initial module for 5.4.0-42-generic
Error! Bad return status for module build on kernel: 5.4.0-42-generic (x86_64)
Consult /var/lib/dkms/v4l2loopback/0.12.3/build/make.log for more information.
dpkg: error processing package v4l2loopback-dkms (--configure):
 installed v4l2loopback-dkms package post-installation script subprocess returne
d error exit status 10
Errors were encountered while processing:
 v4l2loopback-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

/var/lib/dkms/v4l2loopback/0.12.3/build/make.log

#### Expected Results:

Expected sudo apt install v4l2loopback-utils to complete successfully
  
#### Relevant Code:

umlaeute, 2020-08-16:
this is a problem with your distribution, not with `v4l2loopback` upstream.

closed 2020-08-16 by umlaeute

DaveDixon, 2020-08-16:
I don't know what that means.

umlaeute, 2020-08-17:
what do you mean with "that"?

DaveDixon, 2020-08-17:
Thanks, @umlaeute, for getting back to me. I don't know what "problem with your distribution" means and how that's different from "v4l2loopback upstream". I have still don't know how to solve my problem. Thanks again.

umlaeute, 2020-08-17:
"your distribution" is Ubuntu.
"v4l2loopback upstream" is *this* repository.

so the problem needs to be fixed on the Ubuntu side (which is supposed to take care of shipping a `v4l2loopback-dkms` package that is compatible with the kernel they are shipping), and cannot be fixed here.

hence i'm closing this as "not for us".