v4l2loopback/v4l2loopback #334: Error on set-timeout-image: "/dev/video0 is not output device"

MarioMey, 2020-08-14:
### Environment

  * `v4l2loopback` version: [    4.048186] v4l2loopback driver version 0.12.1 loaded

  * kernel version: Linux debian 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2+deb10u1 (2020-06-07) x86_64 GNU/Linux

  * Distribution (+version):
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
  
### Problem:
I create virtual device in /dev/video0 and it works perfect.
`sudo modprobe v4l2loopback devices=1 video_nr=0 card_label="OBS Cam"`

 But, by doing this (following [this](https://github.com/umlaeute/v4l2loopback#setting-stream-timeout)), I get this:

```
mario@debian:~/Imágenes$ v4l2-ctl -d /dev/video0 -c timeout=1000
mario@debian:~/Imágenes$ v4l2loopback-ctl set-timeout-image /home/mario/Imágenes/vlcsnap-2020-06-30-00h07m01s251.png /dev/video0
Reading from file:///home/mario/Imágenes/vlcsnap-2020-06-30-00h07m01s251.png
Estableciendo el conducto a PAUSA …
ERROR: El conducto no quiere pausarse.
ERROR: del elemento /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: El dispositivo «/dev/video0» no es un dispositivo de salida.
Información adicional de depuración:
v4l2_calls.c(639): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x85208001
Estableciendo el conducto a NULL …
Liberando la tubería…
timeout: 1000
```

umlaeute, 2020-08-16:
are you sure `/dev/video0` is the loopback device? (if you have a real webcam, this would usually by `/dev/video0`, and the loopback device some other device like `/dev/video2`)

MarioMey, 2020-08-17:
Thanks for respond. I'm using Android Studio emulator. It takes video from the first video device: /dev/video0. So, I configure v4l2loopback to use /dev/video0 and cameras in 1, 2, etc.

ryleymcc, 2020-12-09:
If you encounter problems detecting your device with Chrome/WebRTC you can try 'exclusive_caps' mode:

sudo modprobe v4l2loopback exclusive_caps=1

This will enable 'exclusive_caps' mode that only reports CAPTURE/OUTPUT capabilities exclusively. The newly created device will announce OUTPUT capabilities only (so ordinary webcam applications (including Chrome) won't see it). As soon as you have attached a producer to the device, it will start announcing CAPTURE capabilities only (so applications that refuse to open devices that have other capabilities apart from capturing can open it too.)

Options that you provided when loading the module (e.g. via modprobe) cannot be easily changed on the fly. In order to change these options, you must first unload the module with rmmod (which will only work if no application is any longer accessing one of the loopback devices) and then load it again (with the new options).

MarioMey, 2020-12-09:
@ryleymcc thanks for that info, but issue is about something different. Even I didn't know how `exclusive_caps` works, I had it in my settings and I have no problems to output to Chrome.

azertyalex, 2022-04-01:
I seem to have the same issue
```bash
sudo v4l2loopback-ctl set-timeout-image /dev/video3 /home/amol/Pictures/alex.png 
v======================================================================v
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Device '/dev/video3' is not a output device.
Additional debug info:
v4l2_calls.c(636): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x5208000
Setting pipeline to NULL ...
Freeing pipeline ...
^======================================================================^
```

umlaeute, 2023-04-27:
i can reproduce this, when trying to set the timeout image without a running (output)stream.
```
sudo v4l2loopback-ctl set-timeout-image /dev/video2 test.png
Setting pipeline to PAUSED ...
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Could not open device '/dev/video2' for reading and writing.
Additional debug info:
../sys/v4l2/v4l2_calls.c(622): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
system error: Invalid argument
ERROR: pipeline doesn't want to preroll.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
^======================================================================^
$
```

the really bad thing is, that once the driver is in this state, it will refuse to accept any streams:

```sh
 gst-launch-1.0 videotestsrc ! videoconvert ! v4l2sink device=/dev/video2
Setting pipeline to PAUSED ...
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Could not open device '/dev/video2' for reading and writing.
Additional debug info:
../sys/v4l2/v4l2_calls.c(622): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
system error: Invalid argument
ERROR: pipeline doesn't want to preroll.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
$
```

apparently because it still tries (and fails) to set the timeout image on consecutive attempts to start streaming.

as of d12e7d7a861807ff2bb224f89c41e1922a3d2a62 (or so), we can no longer attach a 2nd producer if one is already attached, which currently makes it impossible to set the timeout image...

this is probably related to #541

movepointsolutions, 2023-08-30:
Hi everyone,
the timeout image logic was implemented by me, and this was my first kernel project.
I'm currently working on #146, and if something comes in the way I'll fix it.

stephematician, 2025-01-28:
Tagging that this is being addressed in #611

closed 2025-02-06 by umlaeute