v4l2loopback/v4l2loopback #336: fedora silverblue support

ppGodel, 2020-08-26:
Hi all, thanks for the good work you have been doing. I am uaing fedora silverblue and it do not handle dkms packages, the kernel modules ahould be addee to a container images to be used, there are a couple of examples to achieve this, like the old way to add nvidia drivers support to silverblue or wireguard. So my request is for a way to add v4l2loopback to the os image in silverblue.

umlaeute, 2020-08-26:
that's not solveable from the v4l2loopback development POV.

(having said that: if supporting silverblue is as simple as adding a configuration file (as with `dkms.conf`), i'd accept a PR that implements that)

closed 2020-08-26 by umlaeute

moacirrf, 2024-03-28:
Hi i created a solution to use this module on silverblue is not the best, but its working fine here.

[https://github.com/moacirrf/v4l2loopback-fedora-silverblue](https://github.com/moacirrf/v4l2loopback-fedora-silverblue)