v4l2loopback/v4l2loopback #340: make: *** [Makefile:43: v4l2loopback.ko] Error 2

Souravgoswami, 2020-09-02:
Hi, I want to run v4l2loopback on my Archlinux system.
System Info:
```
$ cat /proc/version
Linux version 5.8.5-xanmod1-1-xanmod (makepkg@archlinux) (clang version 10.0.1 , LLD 10.0.1) #1 SMP PREEMPT Wed, 02 Sep 2020 07:55:08 +0000
```

Unfortunately, whenever I run:

```
$ CC=clang HOSTCC=clang CXX=clang++ make
```

I get:
```
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/ramdisk/v4l2loopback modules
make[1]: Entering directory '/usr/lib/modules/5.8.5-xanmod1-1-xanmod/build'
  CC [M]  /ramdisk/v4l2loopback/v4l2loopback.o
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-mno-global-merge’
make[2]: *** [scripts/Makefile.build:281: /ramdisk/v4l2loopback/v4l2loopback.o] Error 1
make[1]: *** [Makefile:1760: /ramdisk/v4l2loopback] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.8.5-xanmod1-1-xanmod/build'
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```

umlaeute, 2020-09-02:
the error message indicates that you are actually using `gcc` rather than `clang`. which you could easily very by passing `V=1` as make arguments.

then: why are you passing the compiler as environment variables? you should use make arguments instead.
and of course: you are setting `HOSTCC` (?) and `CXX` (a C++-compiler).
but `v4l2loopback` is pure *C*, so you should pass `CC` instead.

~~~sh
make CC=clang
~~~

in any case: that's a general problem of passing the proper compiler to *any* kernel-module (note that the `v4l2loopback`-Makefile delegates the actual work to the module-makefile from your kernel), so I'm closing this as not-for-us.

Souravgoswami, 2020-09-02:
Thanks got it compiled and working!

closed 2020-09-02 by Souravgoswami