v4l2loopback/v4l2loopback #344: Make cam show up without producer

chaptergy, 2020-09-16:
Is it possible to make the camera show up without a producer attached? So just after creating the video with 
```
sudo modprobe v4l2loopback devices=1 video_nr=9 card_label="Webcam" exclusive_caps=1
```
I would like to be able to select it in Chrome, Skype and everywhere else you are able to select the cam once you attach a producer.

The reason is I would like to feed a cropped stream of my real webcam to this dummy webcam using ffmpeg, however I don't want ffmpeg to run all the time even when I'm not using the camera. That's why I use `inotify` to watch opens and closes of `/dev/video9` to start and end the ffmpeg process. But when the process is not running the camera won't show up in any software anyways, so I'm not able to select it to trigger the start of ffmpeg.

umlaeute, 2020-09-16:
not with `exclusive_caps`.

probably with a dual-device setup (not there yet).

closed 2020-09-16 by umlaeute

chaptergy, 2020-09-16:
But a feature is planned that would make this possible? The dual-device setup you were talking about?

umlaeute, 2020-09-18:
there's *something* that is being tackled (see #322).

however, since the loopback-device will not do any colorspace/format/dimension conversion, the *only* way this will ever work is if the device is setup to a given format (before the consumer accesses it), and this format won't change.

aramg, 2020-09-18:
FWIW the droidcam fork of v4l2loopback was modified to behave like that.
It defaults to YUV420 and the given width and height (default 640x480).
You should be able to pipe ffmpeg into it as long as the format and size are correct.

You dont need to install the client/app, just build and install the v4l2loopback portion,
https://www.dev47apps.com/droidcam/linux/
https://github.com/aramg/droidcam/tree/master/linux/v4l2loopback

umlaeute, 2020-09-19:
@aramg fantastic news; it would be even better if droidcam would not only *take* whatever open-source solution they find useful, but also would make a feeble attempt to *give back*.

aramg, 2020-09-20:
I have given back. I used this open-source solution to put together a useful & practical utility for people, that has benefited thousands of Linux users all around the world.

Sorry for not contributing back directly to this particular component. If time and interest were to present themselves at some point, I will. Droidcam has been a side project, spanning multiple platforms, there is only so much I am able to focus on in my spare time.