v4l2loopback/v4l2loopback #347: Cannot create /dev/video2 after created /dev/video1

KaeYan, 2020-09-20:
* `v4l2loopback` version: v4l2loopback driver version 0.9.1 loaded

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: Linux kaeyan-Nitro-NP515-51 4.15.0-112-generic #113~16.04.1-Ubuntu SMP Fri Jul 10 04:37:08 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

          uname -a

  * Distribution (+version): Ubuntu 16.04.6 LTS
          lsb_release -a
  
### Step 3: Describe the problem:

#### Steps to reproduce:

  1. sudo modprobe v4l2loopback video_nr=2,3,4
  2. ls /dev/video*
  
#### Observed Results:

  /dev/video0  /dev/video1
  
#### Expected Results:

  /dev/video0  /dev/video1   /dev/video2   /dev/video3   /dev/video4

umlaeute, 2020-09-20:
works for me.

please check with current v4l2loopback (at time of writing this is 0.12.5).

(closing for now; if problem persists with current v4l2loopback, i will re-open)

closed 2020-09-20 by umlaeute

umlaeute, 2020-09-21:
sidenote: any module parameters only take effect when actually *loading* the module. if you repeatedly run `modprobe v4l2loopback`, it will only load the module the first time. you need to unload the module first (e.g. using `rmmod`) before you can load it (again) with new parameters.