v4l2loopback/v4l2loopback #358: I cannot detect the webcam using any conferencing software or web browser

juaninse, 2020-10-15:
I have installed the loopback device using the following line

modprobe v4l2loopback devices=1 video_nr=10 card_label="LoopbackCam" exclusive_caps=1

I have verified that the device is there using 
 v4l2-ctl --list-devices

which returns

LoopbackCam (platform:v4l2loopback-000):
	/dev/video10

In OBS is use v4l2sink to send the output to /dev/video10, and there is no error reported.

However, not a single piece of software detects the LoopbackCam. Not webrowsers (neither firefox nor chromium) not linphone, not Cheese. Even after adding exclusive_caps=1 on the modprobe command, they still do not appear. 

I am using Ubuntu 20.04.

AethanFoot, 2020-10-15:
When I run modprobe i also have the option max_buffers=2 (not sure if that makes a difference), also you could check if this is an issues with v4l2loopback by using ffmpeg, something like this:
ffmpeg -f v4l2 -input_format mjpeg -framerate 30 -video_size 1920x1080 -i /dev/video0 -f v4l2 -pix_fmt yuv420p /dev/video10
which will loop your webcam into the virtual cam.

juaninse, 2020-10-15:
Thanks for the tip. I did try the max_buffers=2, as well as feeding the webcam into the virtual webcam with ffmpeg as you suggested.. No error was reported by ffmpeg doing so, however, the virtual webcam is not detected by any software still.

umlaeute, 2020-10-15:
check https://github.com/umlaeute/v4l2loopback/wiki/Colorspace-Issues

also:
which version of `v4l2loopback` are you using?

umlaeute, 2020-10-16:
i think you need to add much more information if you want to get any meaningful help.

- what's the version of `v4l2loopback`?

- use `v4l2-ctl` to query the device. something like `v4l2-ctl -d  /dev/video10 --all`
  - what is the output of the command when no consumers/producers are run?
  - what is the output of the command when you connect a producer (e.g. OBS)?
  - what is the output of the command when you try to access the device with a consumer?

- do you get anything in the kernel-logs and/or system logs (`dmesg`, `journalctl`)

- try with another (low-level) producer (`ffmpeg` actually fits the bill nicely, but sometimes it's a bit picky; i often prefer `GStreamer`)

- try with another (low-level) consumer (again: I find `GStreamer` is great for testing!)

PhilippKuntschik, 2020-11-01:
I was facing a similar issue with my fresh install over the weekend (xubuntu20.04.1):  ffplay (`ffplay /dev/video0`) was the only consumer able to access the video stream, others did not detect the device at all, or showed a cryptic error message ("VLC is unable to open the MRL 'file:///dev/video01'. Check the log for details.") without any help in the logs.

What solved my issue was to locally compile and install the current master branch as described in the README. 

I got on this trail after I found an verification failure with dmesg. Building the current master (0.12.5) got rid of that failure and finally allowed me to access the camera with any software.

```bash
sudo dmesg  | grep -i v4l2loopback
[  ] v4l2loopback: loading out-of-tree module taints kernel.
[  ] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
[  ] v4l2loopback driver version 0.12.3 loaded
# after local compilation and installation:
[ ] v4l2loopback driver version 0.12.5 loaded
```

rmonat, 2020-11-02:
@juaninse I have encountered an issue with a recent update in Ubuntu 20.04. It seems a fix in the ubuntu package broke the webrtc protocol used in browsers to detect webcams (more specifcally, the `V4L2_CAP_VIDEO_CAPTURE` has been unset by this fix). This has been mentioned by a user in launchpad, [see comments #10 and #11](https://bugs.launchpad.net/ubuntu/+source/v4l2loopback/+bug/1899633)

A workaround is to get back to the previous version :
```
sudo apt install v4l2loopback-dkms=0.12.3
```

With this, I'm able to use v4l2loopback-created webcams in chrome/firefox/...

emptythevoid, 2020-11-03:
I have the same problem.  I can confirm that installing the previous version resumes normal operations.  In my case, Zoom stopped picking up my OBS v4l2sink.

umlaeute, 2020-11-03:
closing this as it is a regression introduced by Ubuntu (thanks @rmonat for the hint)

closed 2020-11-03 by umlaeute