v4l2loopback/v4l2loopback #364: Can't install iridiunwebcam without old v4l2loopback dkms version.

samuk190, 2020-10-23:
when I try to install iridiunwebcam it try to install v4l2loopback dkms version, which is incompatible with latest kernel.
But I build and install from source, and it worked but everytime I do sudo apt-get upgrade or dist-upgrade it keeps trying to autoremove everything

samuk190, 2020-10-23:
I modified default .deb to not depends on v4l2loopback dms 
Download it here
https://github.com/samuk190/builtPackages-debian-
Don't forge to apt-get install qt5-default and apt-get install adb after that.

closed 2020-10-25 by umlaeute

samuk190, 2020-10-26:
@umlaeute  just to state here that the problem is xubuntu repo 20.04 is outdated and do no longer works with sudo apt-get install dkms-v4l2loopback
it should be updated because even on 20.04 version it have recent kernel which need the recent version of v4l2

umlaeute, 2020-10-27:
so that's a problem with the xubuntu repository, not with `v4l2loopback`.

you might want to report the problem there.

also note, that this has already been (erroneously) reported *here* in #333.

samuk190, 2020-10-27:
@umlaeute  but why applications that use v4l2 cannot detect installed dependency when you build and install from source? they always try to install the dependency that is on repository, instead of ignoring. same if you try to sudo apt-get install dkms-v4l2loopback

umlaeute, 2020-10-27:
because this is how package management systems (like apt) work.

in any case, this is *not an issue of the `v4l2loopback` module*, please let us not discuss this here.