v4l2loopback/v4l2loopback #365: Cannot build on Gentoo kernel 5.4.72

zioalex, 2020-10-27:
While building the v4l2loopback package I had the follow problem:
```
/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5 # make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5 modules
make[1]: Entering directory '/usr/src/linux-5.4.72-gentoo'
  CC [M]  /var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.o

  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "video_ioctl2" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "__video_register_device" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_ctrl_new_custom" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_ctrl_handler_init_class" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "video_device_release" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "video_device_alloc" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_device_register" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_ctrl_handler_free" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_device_unregister" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "video_unregister_device" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "video_devdata" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1642: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-5.4.72-gentoo'
make: *** [Makefile:43: v4l2loopback.ko] Error 2
````

So the module is build but it seems I cannot load it.
There is a similar problem in the #227 but I checked the videodev module and it is there:
```
 /var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5 # modinfo videodev
filename:       /lib/modules/5.4.72-gentoo-x86_64/kernel/drivers/media/v4l2-core/videodev.ko
alias:          char-major-81-*
license:        GPL
description:    Video4Linux2 core driver
author:         Alan Cox, Mauro Carvalho Chehab <mchehab@kernel.org>, Bill Dirks, Justin Schoeman, Gerd Knorr
depends:        mc,i2c-core
retpoline:      Y
intree:         Y
name:           videodev
vermagic:       5.4.72-gentoo-x86_64 SMP mod_unload modversions 

```

Any idea idea why I am having this issue?

umlaeute, 2020-10-27:
looks awfully like http://patchwork.ozlabs.org/project/buildroot/patch/20200411222542.64271-1-james.hilliard1@gmail.com/

could you actually *load* the `videodev` module (and run `depmod -a`, just in case)?

zioalex, 2020-10-27:
Yep same problem:
```
 /var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5 # modprobe videodev
 /var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5 # depmod -a
 /var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5 # make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5 modules
make[1]: Entering directory '/usr/src/linux-5.4.72-gentoo'
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "video_ioctl2" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "__video_register_device" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_ctrl_new_custom" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_ctrl_handler_init_class" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "video_device_release" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "video_device_alloc" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_device_register" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_ctrl_handler_free" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "v4l2_device_unregister" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "video_unregister_device" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
ERROR: "video_devdata" [/var/tmp/portage/media-video/v4l2loopback-0.12.5-r1/work/v4l2loopback-0.12.5/v4l2loopback.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[1]: *** [Makefile:1642: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-5.4.72-gentoo'
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```

umlaeute, 2020-10-27:
is this a self-configured kernel or one that is shipped with gentoo?

zioalex, 2020-10-28:
Yes, it is a self configured kernel.

umlaeute, 2020-10-28:
then you are missing something in your configuration (and i don't know what).

this is basically a duplicate of #338 so i'm closing this.
if you have more information to add, please add it to #338.

closed 2020-10-28 by umlaeute