v4l2loopback/v4l2loopback #367: Document the max. number of v4l2loopback devices or how to increase it

RokerHRO, 2020-11-02:
### Step 2: Describe your environment

  * `v4l2loopback` version: Version: 0.12.1-1

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: 4.19.0-10-amd64 #1 SMP Debian 4.19.132-1 (2020-07-24) x86_64 GNU/Linux

          uname -a

  * Distribution (+version): Debian 10.5

          lsb_release -a
  
### Step 3: Describe the problem:

I tried to load the v4l2loopback kernel module with more than 8 video devices. Only 8 devices are created. No error message is emitted.

This limit is not mentioned on the v4l2loopback documentation and also whether it is possible to increase that limit and if yes, how to increase it and if there are any disadvantages when set this limit to a far too high number etc.

umlaeute, 2020-11-05:
patches welcome

RokerHRO, 2020-11-05:
I'd do that if I were sure whether this limit exists in the v4l core itself or only in the v4l2loopback driver or somewhere else and whether it can be raised by simply changing a magic constant somewhere and what the disadvantages of such a change would be.

But at least _tell_ the users about this limit would be easy for everyone with write access to this project. :-)

umlaeute, 2020-11-09:
so far, it's documented in the source-code (only):

https://github.com/umlaeute/v4l2loopback/blob/99cc2968f2f3125ff16f67ff7087d55223433a02/v4l2loopback.c#L183-L191

however, the *new* [dynamic device management](https://github.com/umlaeute/v4l2loopback#dynamic-device-management) kind of obsoletes this, as you can (dynamically) create as many devices as you want to.

e.g. i just created 128 additional loopback devices using:

~~~sh
for i in $(seq 128); do
  v4l2loopback-ctl add
done
~~~

umlaeute, 2020-11-09:
to answer that question more specifically:

> I'd do that if I were sure whether this limit exists in the v4l core itself or only in the v4l2loopback driver or somewhere else and whether it can be raised by simply changing a magic constant somewhere (in the source or can it be changed at runtime?) 

there's a hard limitation in the kernel, where the minor device number is an unsigned 8bit value. so the maximum number of `/dev/video` devices is *256*.

> and what the disadvantages of such a change would be.

(iirc) the main reason for the hardcoded limit to 8 devices is to prevent the innocent user to shoot themselves in their knees.
it probably could be set to the absolute maximum (`256`) if this is really important.
i think that so far most people who need "a lot" of devices also need a way to dynamically allocate (and deallocate) them.

RokerHRO, 2020-11-10:
@umlaeute : Thank you for your detailed answers! :-)
So now when we have all the necessary information together, where can it be added to the documentation, so other users don't have to dig into the kernel sources (and misinterpreting e.g. the comment above the `MAX_DEVICES` constant as a hard compile-time constant that cannot be changed without recompilation etc.)?

umlaeute, 2020-11-11:
>  so other users don't have to dig into the kernel sources

well they don't really have to read the *kernel* source, just the *module* source.
but of course, even that might be too much.

> and misinterpreting e.g. the comment above the MAX_DEVICES constant as a hard compile-time constant

not sure i understand that. where' do you think there is a misinterpretation?

the `MAX_DEVICES` *is* the hard coded maximum of devices that can be created when loading the module.
if you want to change that, you indeed need to recompile the module.
using dynamic device managment simply bypasses this restriction.