v4l2loopback/v4l2loopback #368: v4l2loopback-ctl add -n "Virtual Camera" /dev/video2 doesn't work

PQCraft, 2020-11-04:
#### Title
v4l2loopback-ctl add -n "Virtual Camera" /dev/video2 outputs help instead of running

Also try not to use relative terms.
v4l2loopback-ctl does not think add is a command apparently

  * `v4l2loopback` version:

         (outputs nothing)

  * kernel version: _____

          `Linux PQC-Linux-Mint 5.4.0-52-generic #57-Ubuntu SMP Thu Oct 15 10:57:00 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

  * Distribution (+version): _____
```
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 20
Release:	20
Codename:	ulyana
```
  
### Step 3: Describe the problem:

#### Steps to reproduce:
```
# apt-get install v4l2loopback-utils
# sudo v4l2-ctl add -n "Virtual Camera" /dev/video2
```
#### Observed Results:

```
Usage:
    /usr/bin/v4l2loopback-ctl [command] [args]

Commands:
     set-fps <FPS> /dev/videoX  Set device fps (if format is already
                                negotiated).

     set-timeout-image <IMAGE> /dev/videoX  Set timeout on device, and 
                                            optionally the placeholder picture,
                                            instead of default null frame.

     set-caps <CAPS> /dev/videoX  Fix format for the device (using GStreamer
                                  caps), so it will be maintained until reset,
                                  regardless of future readers/writers.

Options:
    FPS      frames per second, either as integer ('30') or fraction ('50/2')
    IMAGE    image file ("no-input.png"), or 'videotestsrc'
    CAPS     GStreamer caps ("video/x-raw, format=UYVY, width=640, height=480"),
             or 'any'
    /dev/videoX  the v4l2-loopback device file you want to act on
```
  
#### Expected Results:

`made /dev/video2` or something like that

umlaeute, 2020-11-04:
you call

~~~sh
# sudo v4l2-ctl add -n "Virtual Camera" /dev/video2
~~~

this is calling `v4l2-ctl`, which is not supposed to be able to create devices.

assuming this is a typo, you get

~~~sh
# apt-get install v4l2loopback-utils
# sudo v4l2loopback-ctl add -n "Virtual Camera" /dev/video2
Usage:
    /usr/bin/v4l2loopback-ctl [command] [args]

Commands:
     set-fps <FPS> /dev/videoX
     set-timeout-image <IMAGE> /dev/videoX
     set-caps <CAPS> /dev/videoX  
#
~~~

why do you think that `v4l2loopback-ctl` can create devices? esp. why do you think that the *Linux Mint 20* version of `v4l2loopback-ctl` will be able to do that? ("Linux Mint" is notorious for being slow adapt packages; and even so, there has not been a release of `v4l2loopback` that supports dynamic device creation yet)

closed 2020-11-04 by umlaeute

PQCraft, 2020-11-05:
@umlaeute That is what readme.md says under Dynamic Device Management ($ sudo v4l2loopback-ctl add -n "loopy doopy" /dev/video7)

PQCraft, 2020-11-05:
I'm assuming this is just a typo due to your response.

umlaeute, 2020-11-05:
> @umlaeute That is what readme.md says under Dynamic Device Management ($ sudo v4l2loopback-ctl add -n "loopy doopy" /dev/video7)

which README?
if you are referring to [https://github.com/umlaeute/v4l2loopback/blob/master/README.md](https://github.com/umlaeute/v4l2loopback/blob/99cc2968f2f3125ff16f67ff7087d55223433a02/README.md), then you **must** use the `v4l2loopback-ctl` utility compiled from [these sources](https://github.com/umlaeute/v4l2loopback/tree/99cc2968f2f3125ff16f67ff7087d55223433a02/utils) and use the [**matching** `v4l2loopback` module](https://github.com/umlaeute/v4l2loopback/blob/99cc2968f2f3125ff16f67ff7087d55223433a02/v4l2loopback.c).

if you got the `v4l2loopback-utils` from somewhere else (e.g. *Linux Mint 20*), then you should read the README that comes with them (check `/usr/share/doc/v4l2loopback-utils/README.md.gz`).

PQCraft, 2020-11-05:
@umlaeute Oooh. Welp. I didn't realize that. Sorry for the trouble.