v4l2loopback/v4l2loopback #369: Can't auto remove!

juliocargnin, 2020-11-05:
It seem it got stuck in the system:

~$ sudo apt-get autoremove 

Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 12 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up v4l2loopback-dkms (0.12.3-1ubuntu0.1) ...
Removing old v4l2loopback-0.12.3 DKMS files...

------------------------------
Deleting module version: 0.12.3
completely from the DKMS tree.
------------------------------
Done.
Loading new v4l2loopback-0.12.3 DKMS files...
Building for 5.8.0-7625-generic
Building initial module for 5.8.0-7625-generic
ERROR (dkms apport): kernel package linux-headers-5.8.0-7625-generic is not supported
Error! Bad return status for module build on kernel: 5.8.0-7625-generic (x86_64)
Consult /var/lib/dkms/v4l2loopback/0.12.3/build/make.log for more information.
dpkg: error processing package v4l2loopback-dkms (--configure):
 installed v4l2loopback-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 v4l2loopback-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

The log file says:
`DKMS make.log for v4l2loopback-0.12.3 for kernel 5.8.0-7625-generic (x86_64)
Thu 05 Nov 2020 12:06:28 PM JST
Building v4l2-loopback driver...
make -C /lib/modules/5.8.0-7625-generic/build M=/var/lib/dkms/v4l2loopback/0.12.3/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-7625-generic'
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:137:47: warning: ‘struct timeval’ declared inside parameter list will not be visible outside of this definition or declaration
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                               ^~~~~~~
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c: In function ‘v4l2l_get_timestamp’:
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:147:4: error: dereferencing pointer to incomplete type ‘struct timeval’
  147 |  tv->tv_sec = (time_t)ts.tv_sec;
      |    ^~
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:147:16: error: ‘time_t’ undeclared (first use in this function); did you mean ‘ktime_t’?
  147 |  tv->tv_sec = (time_t)ts.tv_sec;
      |                ^~~~~~
      |                ktime_t
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:147:16: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:147:23: error: expected ‘;’ before ‘ts’
  147 |  tv->tv_sec = (time_t)ts.tv_sec;
      |                       ^~
      |                       ;
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c: In function ‘vidioc_qbuf’:
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:1527:24: error: passing argument 1 of ‘v4l2l_get_timestamp’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1527 |    v4l2l_get_timestamp(&b->buffer.timestamp);
      |                        ^~~~~~~~~~~~~~~~~~~~
      |                        |
      |                        struct __kernel_v4l2_timeval *
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:137:56: note: expected ‘struct timeval *’ but argument is of type ‘struct __kernel_v4l2_timeval *’
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                        ~~~~~~~~~~~~~~~~^~
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c: In function ‘v4l2_loopback_write’:
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:1954:22: error: passing argument 1 of ‘v4l2l_get_timestamp’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1954 |  v4l2l_get_timestamp(&b->timestamp);
      |                      ^~~~~~~~~~~~~
      |                      |
      |                      struct __kernel_v4l2_timeval *
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:137:56: note: expected ‘struct timeval *’ but argument is of type ‘struct __kernel_v4l2_timeval *’
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                        ~~~~~~~~~~~~~~~~^~
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c: In function ‘init_buffers’:
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:2059:23: error: passing argument 1 of ‘v4l2l_get_timestamp’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 2059 |   v4l2l_get_timestamp(&b->timestamp);
      |                       ^~~~~~~~~~~~~
      |                       |
      |                       struct __kernel_v4l2_timeval *
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:137:56: note: expected ‘struct timeval *’ but argument is of type ‘struct __kernel_v4l2_timeval *’
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                        ~~~~~~~~~~~~~~~~^~
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c: In function ‘init_vdev’:
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:2090:23: error: ‘VFL_TYPE_GRABBER’ undeclared (first use in this function); did you mean ‘VFL_TYPE_SUBDEV’?
 2090 |  vdev->vfl_type     = VFL_TYPE_GRABBER;
      |                       ^~~~~~~~~~~~~~~~
      |                       VFL_TYPE_SUBDEV
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c: In function ‘v4l2loopback_init_module’:
/var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.c:2441:44: error: ‘VFL_TYPE_GRABBER’ undeclared (first use in this function); did you mean ‘VFL_TYPE_SUBDEV’?
 2441 |   if (video_register_device(devs[i]->vdev, VFL_TYPE_GRABBER, video_nr[i]) < 0) {
      |                                            ^~~~~~~~~~~~~~~~
      |                                            VFL_TYPE_SUBDEV
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:290: /var/lib/dkms/v4l2loopback/0.12.3/build/v4l2loopback.o] Error 1
make[1]: *** [Makefile:1780: /var/lib/dkms/v4l2loopback/0.12.3/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-7625-generic'
make: *** [Makefile:43: v4l2loopback.ko] Error 2`

How can I fix this?

juliocargnin, 2020-11-05:
> It seem it got stuck in the system:
>
> ~$ sudo apt-get autoremove
>
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> 0 upgraded, 0 newly installed, 0 to remove [...]

I have fixed now by removing it with:
sudo apt-get purge --auto-remove v4l2loopback-dkms

umlaeute, 2020-11-05:
great that you found a solution!

---

however, please understand that you actually had problems with your package manager for your distribution.
while annoying for you, there is really *nothing* in that the "`v4l2loopback` upstream project" (*this* repository, where the driver is developed) can do about it.

for future reference: please report such issues with your distribution. reporting them here is useless and i will simply close such reports.

closed 2020-11-05 by umlaeute