v4l2loopback/v4l2loopback #370: Device '/dev/video1' is not a output device - While device seen as output by v4l2-ctl

didier-durand, 2020-11-05:
Hi there,

Please, let me know if additional info is needed: we'll be happy to provide it.

Is the disable due to the message below:  '_**v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel'**_  ?

Thanks for your help!

Didier

### Step 2: Configuration

  * `v4l2loopback` version: _____

          sudo dmesg  | grep -i v4l2loopback

 [  853.969836] v4l2loopback: loading out-of-tree module taints kernel.
[  853.969896] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
[  853.971377] v4l2loopback driver version 0.12.3 loaded

  * kernel version: _____

          uname -a

Linux microk8s-akri 5.4.0-1028-gcp #29-Ubuntu SMP Mon Oct 5 16:42:23 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

  * Distribution (+version): _____

          lsb_release -a

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

### Step 3: Describe the problem:

The following command fails with the message below 

sudo gst-launch-1.0 -v videotestsrc pattern=ball ! "video/x-raw,width=640,height=480,framerate=10/1" ! avenc_mjpeg ! v4l2sink device=/dev/video1 

ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: **Device '/dev/video1' is not a output device**
Additional debug info:
v4l2_calls.c(636): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x85208000
Setting pipeline to NULL ...
Freeing pipeline ...

**v4l2-ctl --all -d /dev/video1** produces: 

Driver Info:
	Driver name      : v4l2 loopback
	Card type        : Dummy video device (0x0000)
	Bus info         : platform:v4l2loopback-000
	Driver version   : 5.4.65
	Capabilities     : 0x85208002
		**Video Output**
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x85208000
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
Priority: 0
Video output: 0 (loopback in)
Format Video Output:
	Width/Height      : 0/0
	Pixel Format      : 'BGR4' (32-bit BGRA/X 8-8-8-8)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 0
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Streaming Parameters Video Capture:
	Frames per second: 30.000 (30/1)
	Read buffers     : 8
Streaming Parameters Video Output:
	Frames per second: 30.000 (30/1)
	Write buffers    : 8

User Controls

                    keep_format 0x0098f900 (bool)   : default=0 value=0
              sustain_framerate 0x0098f901 (bool)   : default=0 value=0
                        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io 0x0098f903 (bool)   : default=0 value=0

** lsmod** produces:

lsmod
Module                  Size  Used by
v4l2loopback           40960  0
videodev              225280  1 v4l2loopback
mc                     53248  1 videodev
btrfs                1253376  0
xor                    24576  1 btrfs
zstd_compress         167936  1 btrfs
raid6_pq              114688  1 btrfs
ufs                    81920  0
qnx4                   16384  0
hfsplus               110592  0
hfs                    61440  0
minix                  36864  0
ntfs                  106496  0
msdos                  20480  0
jfs                   188416  0
xfs                  1277952  0
ip6table_nat           16384  0
ip6_tables             32768  1 ip6table_nat
vxlan                  69632  0
ip6_udp_tunnel         16384  1 vxlan
udp_tunnel             16384  1 vxlan
ipt_rpfilter           16384  1
xt_set                 16384  4
xt_multiport           20480  56
iptable_raw            16384  1
ip_set_hash_ip         40960  1
ip_set_hash_net        49152  3
ip_set                 53248  3 ip_set_hash_ip,xt_set,ip_set_hash_net
xfrm_user              36864  1
xfrm_algo              16384  1 xfrm_user
veth                   28672  0
ipt_REJECT             16384  2
nf_reject_ipv4         16384  1 ipt_REJECT
overlay               114688  14
xt_nat                 16384  4
xt_tcpudp              20480  14
xt_addrtype            16384  7
ip_vs_sh               16384  0
ip_vs_wrr              16384  0
ip_vs_rr               16384  0
ip_vs                 155648  6 ip_vs_rr,ip_vs_sh,ip_vs_wrr
iptable_mangle         16384  1
xt_MASQUERADE          20480  3
xt_conntrack           16384  25
xt_mark                16384  56
iptable_nat            16384  1
nf_nat                 40960  4 ip6table_nat,xt_nat,iptable_nat,xt_MASQUERADE
aufs                  262144  0
xt_comment             16384  268
iptable_filter         16384  1
bpfilter               32768  0
br_netfilter           28672  0
bridge                176128  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
nls_iso8859_1          16384  1
dm_multipath           32768  0
scsi_dh_rdac           16384  0
scsi_dh_emc            16384  0
scsi_dh_alua           20480  0
crct10dif_pclmul       16384  1
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
aesni_intel           372736  0
virtio_net             53248  0
net_failover           20480  1 virtio_net
crypto_simd            16384  1 aesni_intel
psmouse               155648  0
failover               16384  1 net_failover
input_leds             16384  0
cryptd                 24576  2 crypto_simd,ghash_clmulni_intel
serio_raw              20480  0
glue_helper            16384  1 aesni_intel
sch_fq_codel           20480  3
drm                   491520  0
virtio_rng             16384  0
ip_tables              32768  4 iptable_filter,iptable_raw,iptable_nat,iptable_mangle
x_tables               40960  16 xt_conntrack,iptable_filter,xt_multiport,xt_tcpudp,xt_addrtype,xt_nat,xt_comment,xt_set,ip6_tables,ipt_REJECT,ipt_rpfilter,iptable_raw,ip_tables,xt_MASQUERADE,iptable_mangle,xt_mark
autofs4                45056  2

#### Steps to reproduce:

  1. Install Ubuntu 20.04 (here on Google GCE)
  2.sudo apt install -y linux-modules-extra-$(uname -r)
  3. sudo apt install -y v4l2loopback-dkms
  4.  sudo modprobe v4l2loopback exclusive_caps=1 devices=2 video_nr=1,2
  5. sudo gst-launch-1.0 -v videotestsrc pattern=ball ! "video/x-raw,width=640,height=480,framerate=10/1" ! avenc_mjpeg ! v4l2sink device=/dev/video1   (install gstream packages before)
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

 we would expect gst-launch to be able to stream on  /dev/video1 given the output of v4l2-ctl saying the 'Video Output' is in the capabilities.

didier-durand, 2020-11-06:
Additional info : the gst-launch above works fine if the sink is a filesink

sudo gst-launch-1.0 -v videotestsrc pattern=ball ! "video/x-raw,width=640,height=480,framerate=10/1" ! avenc_mjpeg ! filesink location=test.vid1

Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)10/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)10/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/avenc_mjpeg:avenc_mjpeg0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)10/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)10/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/avenc_mjpeg:avenc_mjpeg0.GstPad:src: caps = image/jpeg, parsed=(boolean)true, width=(int)640, height=(int)480, framerate=(fraction)10/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstFileSink:filesink0.GstPad:sink: caps = image/jpeg, parsed=(boolean)true, width=(int)640, height=(int)480, framerate=(fraction)10/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:09.649415847
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

ls -l test.vid1
-rw-r--r-- 1 root root 7639952 Nov  6 04:58 test.vid1

umlaeute, 2020-11-06:
your pipeline works nicely for me.

~~~sh
$ uname -a
Linux xenakis 5.9.0-1-amd64 #1 SMP Debian 5.9.1-1 (2020-10-17) x86_64 GNU/Linux
$ sudo dmesg  | grep v4l2
[848422.306793] v4l2loopback driver version 0.12.5 loaded
$ gst-launch-1.0 -v videotestsrc pattern=ball ! "video/x-raw,width=640,height=480,framerate=30/1" ! avenc_mjpeg ! v4l2sink device=/dev/video1 
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)Y444, width=(int)640, height=(int)480, framerate=(fraction)30/1, multiview-mode=(string)mono, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)Y444, width=(int)640, height=(int)480, framerate=(fraction)30/1, multiview-mode=(string)mono, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/avenc_mjpeg:avenc_mjpeg0.GstPad:sink: caps = video/x-raw, format=(string)Y444, width=(int)640, height=(int)480, framerate=(fraction)30/1, multiview-mode=(string)mono, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)Y444, width=(int)640, height=(int)480, framerate=(fraction)30/1, multiview-mode=(string)mono, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/avenc_mjpeg:avenc_mjpeg0.GstPad:src: caps = image/jpeg, parsed=(boolean)true, width=(int)640, height=(int)480, colorimetry=(string)2:4:7:1, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0.GstPad:sink: caps = image/jpeg, parsed=(boolean)true, width=(int)640, height=(int)480, colorimetry=(string)2:4:7:1, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:41.3 / 99:99:99.
~~~

and while the pipeline is running:

~~~sh
$ v4l2-ctl --all -d /dev/video1 
Driver Info:
	Driver name      : v4l2 loopback
	Card type        : Dummy video device (0x0000)
	Bus info         : platform:v4l2loopback-000
	Driver version   : 5.9.1
	Capabilities     : 0x85208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
Priority: 0
Video input : 0 (loopback: ok)
Video output: 0 (loopback in)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'JPEG' (JFIF JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 1228800
	Colorspace        : sRGB
	Transfer Function : sRGB
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Limited Range
	Flags             : 
Format Video Output:
	Width/Height      : 640/480
	Pixel Format      : 'JPEG' (JFIF JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 1228800
	Colorspace        : sRGB
	Transfer Function : sRGB
	YCbCr/HSV Encoding: ITU-R 601
	Quantization      : Limited Range
	Flags             : 
Streaming Parameters Video Capture:
	Frames per second: 30.000 (30/1)
	Read buffers     : 2
Streaming Parameters Video Output:
	Frames per second: 30.000 (30/1)
	Write buffers    : 2

User Controls

                    keep_format 0x0098f900 (bool)   : default=0 value=0
              sustain_framerate 0x0098f901 (bool)   : default=0 value=0
                        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io 0x0098f903 (bool)   : default=0 value=0
~~~

could you try with the latest release of `v4l2loopback-0.12.5`?

##### nitpicks
- in general, you shouldn't run gstreamer-pipelines as *root* (in general you shouldn't run any command as root that doesn't absolutely require it)
- i trust you that GStreamer works by itself, so there's not very much use in uploading a video-file from a pipeline that doesn't involve a loopback device (though of course the attempt to provide as much info as possible is appreciated)

didier-durand, 2020-11-06:
Thanks for your hints!

In the mean time, I replaced GStreamer by ffmpeg to stream to the sink and it works fine.

I will try GTSteamer with v4l2loopback-0.12.5 and let you know.

Didier

didier-durand, 2020-11-06:
Hi there

You had the right idea : I upgraded to v4l2loopback-0.12.5 and it works fine. See below.

To reach this,  I used 

sudo add-apt-repository "deb http://us.archive.ubuntu.com/ubuntu/ groovy universe multiverse"
sudo add-apt-repository "deb http://us.archive.ubuntu.com/ubuntu/ groovy-updates universe multiverse"

Then, sudo apt install -y v4l2loopback-dkms will install the required 0.12.5 in place of previous 0.12.3

It's fine for me because I am on a test system. But, I am on Ubuntu 20.04 LTS. So, groovy repos (=20.10) are not supposed to be used with 20.04. Am I correct? 

If yes, it may be an issue for shops using only LTS versions. The next one will be 22.04. Maybe quite far away for some folks.

Anyway, solved for me. Thanks a lot!

Didier

## dmesg output after upgrade

[  505.470735] v4l2loopback driver version 0.12.5 loaded

### GSteamer exec log 

**gst-launch-1.0 -v videotestsrc pattern=ball ! "video/x-raw,width=640,height=480,framerate=10/1" ! avenc_mjpeg ! v4l2sink device=/dev/video1** 

Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)10/1, multiview-mode=(string)mono, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)10/1, multiview-mode=(string)mono, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/avenc_mjpeg:avenc_mjpeg0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)10/1, multiview-mode=(string)mono, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)10/1, multiview-mode=(string)mono, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/avenc_mjpeg:avenc_mjpeg0.GstPad:src: caps = image/jpeg, parsed=(boolean)true, width=(int)640, height=(int)480, colorimetry=(string)2:4:7:1, framerate=(fraction)10/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0.GstPad:sink: caps = image/jpeg, parsed=(boolean)true, width=(int)640, height=(int)480, colorimetry=(string)2:4:7:1, framerate=(fraction)10/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:27.807473202
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

umlaeute, 2020-11-06:
you could have also just downloaded the [dkms-package](http://za.archive.ubuntu.com/ubuntu/pool/universe/v/v4l2loopback/v4l2loopback-dkms_0.12.5-1_all.deb) and installed it manually wit `dpkg -i` (this basically works, because the `v4l2loopback-dkms` package is very simple dependency-wise).

anyhow, glad it worked out for you.

closed 2020-11-06 by umlaeute

j2l, 2020-11-27:
"you could have also just downloaded the dkms-package and installed it manually wit dpkg -i (this basically works, because the v4l2loopback-dkms package is very simple dependency-wise)" didn't work for me on ubuntu 20.04
Thanks @didier-durand for finding a fix to this. @umlaeute I don't understand your LinusTorvald attitude but hey, I don't care, you didn't reply to me and I won't subscribe.

umlaeute, 2020-11-27:
>  @umlaeute I don't understand your LinusTorvald attitude

@j2l could you explain that a bit?
i agree that i might be often be snarky, but in this special case i really don't see how and why.
what i was suggesting ("download the dkms and install it manually"), was merely a way to circumvent the problem that adding a repository of another distro release to a ubuntu installation (in this case: adding "groovy" repos on a "focus" installation) is bound to give you a lot of problems.

umlaeute, 2025-10-23:
works nicely here:

```sh
$ uname -a
Linux xenakis 6.16.9+deb14-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.16.9-1 (2025-09-27) x86_64 GNU/Linux
$ gst-launch-1.0 --version
gst-launch-1.0 version 1.26.6
GStreamer 1.26.6
https://tracker.debian.org/pkg/gstreamer1.0
$ v4l2loopback-ctl --version
v4l2loopback-ctl v0.15.1
v4l2loopback sysfs v0.15.1
v4l2loopback module v0.15.1
```

I do see the error if i launch two producers for the same device simultaneously (which is forbidden, so the error is correct).

oscaracena, 2025-10-23:
> works nicely here:

So, it may be a matter of updating Gstreamer or the Linux Kernel. I'll try to check in a more recent OS. Thanks for your feedback!