v4l2loopback/v4l2loopback #371: ERROR: modpost: "__stack_chk_guard"

kovenko, 2020-11-05:
# uname -r
5.8.16-sunxi64

# make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/root/intercom/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.16-sunxi64'
  MODPOST /root/intercom/v4l2loopback/Module.symvers
ERROR: modpost: "__stack_chk_guard" [/root/intercom/v4l2loopback/v4l2loopback.ko] undefined!
scripts/Makefile.modpost:111: recipe for target '/root/intercom/v4l2loopback/Module.symvers' failed
make[2]: *** [/root/intercom/v4l2loopback/Module.symvers] Error 1
make[2]: *** Deleting file '/root/intercom/v4l2loopback/Module.symvers'
Makefile:1667: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.16-sunxi64'
Makefile:43: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2

v4l2loopback.c
add functon

unsigned long __stack_chk_guard;
void __stack_chk_guard_setup(void)
{
     __stack_chk_guard = 0xFFFFFFFFF;//provide some unique magic numbers
}

void __stack_chk_fail(void)
{
 /* Error message */
}

umlaeute, 2020-11-06:
which version of `v4l2loopback` are you trying to compile?

umlaeute, 2020-12-18:
closing due to inactivity.

closed 2020-12-18 by umlaeute