v4l2loopback/v4l2loopback #372: LoopBack kernel 5.8.16-2-MANJARO: Missing kernel headers.

PeterKW, 2020-11-05:
"Unable to install module v4l2loopback/0.12.5 for kernel 5.8.16-2-MANJARO: Missing kernel headers."

I haven't been able to find a fix that works for this issue yet unfortunately. It looks to me as though I have the correct headers for 5.8.* but I don't know very much about how headers work. Any other ways of getting loopback to work? 

I was attempting to install fakecam https://snapcraft.io/fakecam

PeterKW, 2020-11-05:
Ultimately, I would like to be able to use a virtual background in Microsoft Teams etc. which doesn't currently support virtual backgrounds on its Linux client unfortunately

xslendix, 2020-11-05:
you literally need to pacman -Syu linux-headers

closed 2020-11-06 by umlaeute

PeterKW, 2020-11-06:
Is this related to loopback?

fakecam
The fake cemera device is not accessible. Make sure you have installed and activated v4l2loopback-dkms. On debian and Ubuntu derivative distributions, this is done with the following command - for other systems check your provider's documentation and package repositories for a similar package:

sudo apt install v4l2loopback-dkms

The module must be configured with the following options:

devices=1 video_nr=20 card_label="fakecam" exclusive_caps=1

To do this now and get going straight away run:

modprobe -r v4l2loopback && modprobe v4l2loopback devices=1 video_nr=20 card_label="fakecam" exclusive_caps=1

This can be achieved by editing /etc/modprobe.d/fakecam.conf or /etc/modprobe.conf to add the following line:

options v4l2loopback devices=1 video_nr=20 card_label="fakecam" exclusive_caps=1

You can do this easily with the following command:

echo options v4l2loopback devices=1 video_nr=20 card_label="fakecam" exclusive_caps=1 | sudo tee -a /etc/modprobe.d/fakecam.conf

The module is not automatically reloaded on bootup so we need to add an entry into either /etc/modules or a new file in /etc/modules-load.d named fakecam.conf. It simply requires the word v4l2loopback on a new line in one of these places:

echo v4l2loopback | sudo tee -a /etc/modules-load.d/fakecam.conf

Once the configuration is set it will persist across reboots. If you haven't run the modprobe commands above then you should now either reboot, or run:

sudo modprobe -r v4l2loopback && sudo modprobe v4l2loopback

umlaeute, 2020-11-06:
i have no idea what you are getting at.

it *is* related in a broad sense (as is "fakecam", "Linux" and "computers in general).
but the problem can be easily solved by:
- reading the error message
- reading the documentation
- following the issue-template

that would have given you:

- as the error message says "missing kernel headers", you might have started by installing the kernel headers
- the readme says that you *must* have the headers for your exact kernel version installed
- the issue template tells you to
  - read the readme (see above)
  - check the issue tracker (there's a number of similar bug reports)

PeterKW, 2020-11-06:
Oh strange, I tried it a few times before and it didn't seem to do much differently. It seems to be working now so I guess I must have just been confused about it. Ach well

Thanks