v4l2loopback/v4l2loopback #373: Applications (Zoom, Skype, Chrome, Cheese and others) do not see the virtual camera v4l2loopback

melnikeg, 2020-11-07:
Hello dear community!
I installed the v4l2loopback driver following this [instruction](https://github.com/umlaeute/v4l2loopback).
But none of the programs on my Ubuntu 20.04 see the virtual device. The virtual camera does not appear in the device list. Although the command v4l2-ctl --list-device gives the output where the device is:
Dummy video device (0x0000) (platform: v4l2loopback-000):
        / dev / video2

CNF9055: Chicony USB 2.0 Camera (usb-0000: 00: 1a.0-1.4):
        / dev / video0
        / dev / video1

Please tell me what could be the problem.

umlaeute, 2020-11-07:
1. most likely you forgot to feed data to the loopback device
2. if not, then it's most likely a duplicate of #358 and thus a big introduced by the ubuntu packaging.

closed 2020-11-07 by umlaeute

melnikeg, 2020-11-08:
The problem was resolved by removing the v4l2loopback-dkms package
command:
sudo apt remove v4l2loopback-dkms
After that, the virtual camera was displayed in the list of devices in all applications.

umlaeute, 2020-11-08:
@melnikeg great you found a solution.
however, i think your advice to remove a random package is bad and dangerous if you don't provide context (people are known to just blindly copy cmdlines from the internet)

so, whoever reads this: removing the `v4l2loopback-dkms` is most likely **not** the solution  (unless you understand what you are doing).i think this is happening:
- Ubuntu-20.04 shipped a buggy version of `v4l2loopback-dkms`
   - afaik, this is being resolved via the next proposed-updates, so in the future the problem is likely non-existant
- the user installed the buggy module (discovering that it doesn't work)
- installing (and compiling) the `v4l2loopback` module from *source* (aka this repository) installs the new module *in-parallel* to the existing module
- the version installed by Ubuntu takes precedence (either that, or the user forgot to unload the (buggy) module (via `rmmod`) before loading the fixed version)
- fix
  - removing the Ubuntu package fixed a possible precedence issue
  - rebooting makes sure to unload any buggy version

melnikeg, 2020-11-08:
I agree that I have not fully understood what the essence of the matter is. I devoted many hours to solving my problem. But I never found a concrete answer. Therefore, I decided to share my solution, even if it is not completely professional, or generally unprofessional)))
Thanks for this link #358. This discussion gave me the idea that the problem is in v4l2loopback-dkms. As said in the comment https://github.com/umlaeute/v4l2loopback/issues/358#issuecomment-720519735
Anyway, the question is closed. The solution was found for me. I hope this topic will be useful to someone. At least it will show you the way for a solution. Thank you all for your participation!

camillescott, 2020-11-10:
I can confirm that I also had this issue on Ubuntu 20.04 LTS. Removing the provided `v4l2loopback-dkms` and installing from source fixed the problem.