v4l2loopback/v4l2loopback #374: RFE: offer multiple resolutions and ability to call script

bugsyb, 2020-11-07:
Would it be possible to add new functionality to v4l2loopback?

Idea is to behave closer to how typical webcam offers variation of resolution types and based on selection call script with parameters to provide appropriate feed (encoded, converted resolution, frame rate, etc.).

Same request has been posted recently at https://stackoverflow.com/questions/64722765/how-to-offer-multiple-resolutions-under-device-v4l2loopback

Copy/paste for convenience.
####

How to offer multiple resolutions under device (v4l2loopback)? Idea is that depending on what would be requested as read of stream, different command would be run. Background is that v42loopback is fed by other stream and is to be used from browser level and browser dynamically changes resolution (Jitsi) or user would be the one selecting it.

Therefore preferably, if i.e. 1920x1080 is requested, v4l2loopback device is fed with such stream by the source and if other, then parameters are adjusted. At the same time request to v4l2loopback device (/dev/video42 or any number) could be 640x480 and there's no reason for to feed the loopback device with FHD at that time.

This might be more of RFE as might require some code changes. The resolution would have to be passed over and calling script with parameters. On the same tune, type of stream could be requested, i.e. h264, VP8, etc. and obviously it is impossible to feed multiple streams like this to loopback device in parallel.

Due to bandwidth/cpu power it is not possible to provide continuous feed with multiple resources, hence the need.

umlaeute, 2020-11-07:
this is a duplicate of #197 (and the answer is still: no)

closed 2020-11-07 by umlaeute

bugsyb, 2020-11-07:
Thank you - haven't found that one, though it was slightly different - was about "sniffing" or in parallel getting different resolutions, but to you point https://github.com/umlaeute/v4l2loopback/issues/197#issuecomment-695928338, problem is with **no** communication down the path from v4l2loopback about resolution requested. Thought that the resolution requested once v4l2loopback gets request from consumer (and no one else uses device) could signal somehow that resolution/type requested. I'm mostly after resolution piece as idea is to offer h264.