v4l2loopback/v4l2loopback #375: H264 input stream via gstreamer fails (ffmpeg ok)

bugsyb, 2020-11-07:
Idea is to offer via v4l2loopback compressed H264 stream for webcam which does not support it natively and have full control over compression parameters as the one selected by browser are less than optimal.
Probably an error between keyboard and the screen but prefer to verify:
```
gst-launch-1.0  v4l2src device=/dev/video0 ! image/jpeg, width=1280, height=720, framerate=30/1, type=video ! jpegdec ! videoscale ! videoconvert ! x264enc tune=zerolatency ! v4l2sink device=/dev/video2
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Device '/dev/video2' is not a output device.
Additional debug info:
v4l2_calls.c(636): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x5208000
Setting pipeline to NULL ...
Freeing pipeline ...
```
Which in theory should be equivalent to:
`ffmpeg -y  -f video4linux2  -framerate 30 -video_size 1280x720  -input_format mjpeg -i /dev/video0 -c:v libx264 -f v4l2 /dev/video2
`
Just to confirm - what works is - straight output:
gst-launch-1.0 -v v4l2src device=/dev/video0 ! image/jpeg, width=1280, height=720, framerate=30/1, format=MJPG ! jpegdec ! xvimagesink

In effort to debug issue suspecting above-mentioned source of the issue, created two devices
`
modprobe v4l2loopback video_nr="2" 'card_label=test_virt' max_buggers=2 max_openers=64 devices=2 exclusive_caps=1,1
`
Whilst x264enc portion didn't work, simple copy worked nice:
`gst-launch-1.0 -v v4l2src device=/dev/video0 ! image/jpeg, width=1280, height=720, framerate=30/1, format=MJPG    ! v4l2sink device=/dev/video3`

Pushing it back through x264enc something is wrong, and suspicion is that it is more Gstreamer related, but also the flow seems ok - hence confusion - but it might be format, equally might be something missing what v4l2loopback requires to have set at v4l2sink level. 
Apologies if it falls more into gstreamer and if possible, any help would be appreciated.

```
gst-launch-1.0 -v v4l2src device=/dev/video0 ! image/jpeg, width=1280, height=720, framerate=30/1, format=MJPG ! jpegdec ! videoscale ! videoconvert ! x264enc tune=zerolatency ! v4l2sink device=/dev/video3
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)1280, height=(int)720, framerate=(fraction)30/1, format=(string)MJPG, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)1280, height=(int)720, framerate=(fraction)30/1, format=(string)MJPG, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstJpegDec:jpegdec0.GstPad:sink: caps = image/jpeg, width=(int)1280, height=(int)720, framerate=(fraction)30/1, format=(string)MJPG, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)1280, height=(int)720, framerate=(fraction)30/1, format=(string)MJPG, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstJpegDec:jpegdec0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstVideoScale:videoscale0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
Redistribute latency...
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstVideoScale:videoscale0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)1280, height=(int)720, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:src: caps = video/x-h264, codec_data=(buffer)0164001fffe1001d6764001facb200a00b7602dc0808150000030001000003003c8f18324801000568ebccb22c, stream-format=(string)avc, alignment=(string)au, level=(string)3.1, profile=(string)high, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:0:0, chroma-site=(string)mpeg2, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.281905689
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

umlaeute, 2020-11-08:
if it works with `ffmpeg` and fails with `GStreamer`, why do you think the problem is with `v4l2loopback`?

bugsyb, 2020-12-20:
Hope is/was that the problem is the link between keyboard and screen though to answer to your question - if using sink it works, but using the device it does not, hence it might be in how specific variables are taken and/or submit by calling party - thought there was already one situation where problem was on v4l2loopback side and how it took one and not the other.

It was more ask for a help and if it happens to be a bug, than maybe a direction and transparency in what has been discovered.

From GStreamer perspective, if it works with Sink, why shouldn't it work with other device? (yes, integration, but is it specific to device within same type?). Trying to avoid ping-pong as well.

For the moment I had to give up this path, as idea was that device would be used by Jitsi, but if as per other we cannot offer different resolutions to let Jitsi pick and call third party (i.e. GStreamer with parameters) then this is cul de sac for me :(

umlaeute, 2020-12-22:
> From GStreamer perspective, if it works with Sink, why shouldn't it work with other device? (yes, integration, but is it specific to device within same type?). Trying to avoid ping-pong as well.

because GStreamer elements are complex beasts. one element might accept what others don't (and `fakesink` will accept *everything*.
what's more, the streams an element might accept depend not only on the intrinsic capabilities of the element (e.g. "the `v4l2sink` element can write a `video/x-raw,format=YUY2,width=32768,height=32768` stream"), but also on the properties of the surrounding system (e.g. "the `/dev/video3` device has been configured with a maximum width of 8192 pixels"; therefore the `v4l2sink` element can **not** write a 32768x32768 frame to `/dev/video3` on this specific system).

also, please note that in your pipelines it seems that `v4l2src` is failing, rather than `v4l2sink`
and your first pipeline fails with 
> ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Device '/dev/video2' is not a output device.

and i think it's fair enought that `v4l2sink` should refuse to send data to a video-device that is not an output device.
If the device is indeed a loopback device, make sure that no other producer is writing to it (as you are using `exclusive_caps`, as soon as you are writing to the device, it will become a "capture device" only)

> It was more ask for a help and if it happens to be a bug, than maybe a direction and transparency in what has been discovered.

for general help, please use a forum like the one at [Unix & Linux](https://unix.stackexchange.com/), rather than this *bug tracker*.

and just for the record, this works nicely on my system:

~~~sh
$ gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoscale ! videoconvert ! x264enc ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=1280,height=720,framerate=30/1 ! v4l2sink device=/dev/video5
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[...]
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, codec_data=(buffer)017a001fffe10020677a001fbcd9405005bbff000300046a021a0c80000003008000001e478c18cb01000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)3.1, profile=(string)high-4:2:2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)3/4, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1
0:06:57.6
~~~

this is with GStreamer-1.18.2 and v4l2loopback-0.12.5 (as shipped with Debian/sid as of today).
obviously my loopback device is `/dev/video5`, and my input device (`/dev/video0`) doesn't produce JPEG-frames, so there's no `jpegdec` element.

closed 2020-12-22 by umlaeute

thilindaEdiri, 2023-09-28:
@umlaeute 
Hi,

I am trying to stream my USB cam video (/dev/video2) in H264 format to virtual camera (/dev/video10) using GStreamer. But i am getting following error. 

$ gst-launch-1.0 -v v4l2src device=/dev/video2 ! video/x-h264,width=640,height=480,framerate=30/1 ! v4l2sink device=/dev/video10 
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Device '/dev/video2' does not support 2:4:7:1 colorimetry
Additional debug info:
gstv4l2object.c(3849): gst_v4l2_object_set_format_full (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Device wants 2:4:5:1 colorimetry
Execution ended after 0:00:00.019780161
Setting pipeline to NULL ...
Freeing pipeline ...

- I am using GStreamer-1.16.3 and v4l2loopback-0.12.5

- My USB cam (Arducam USB Camera: Arducam USB) support H264 video format.

- Caps details of "gst-device-monitor-1.0" as follows (only for width=640,height=480):

caps : video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 20/1, 15/1, 10/1 };

- When I try to run your code (/dev/video0 device is for video/x-raw format);

gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoscale ! videoconvert ! x264enc ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! v4l2sink device=/dev/video10

- GStreamer pipeline gives following error in the middle of starting process.

ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:02.493410398
Setting pipeline to NULL ...
Freeing pipeline ...

I would appreciate a prompt response.

Thank you.