v4l2loopback/v4l2loopback #376: /dev/video0: not a video capture device (all of a sudden...)

grssnbchr, 2020-11-10:
_I honestly don't know whether this is a v4l2loopback` problem, but maybe someone can help me start debugging this nonetheless._

 * `v4l2loopback` version: v4l2loopback driver version 0.12.3 loaded

  * kernel version: Linux syg-5 5.4.0-52-generic #57-Ubuntu SMP Thu Oct 15 10:57:00 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

  * Distribution (+version): Ubuntu 20.04.1 LTS

**This morning I did a system update with `apt update && apt upgrade`. Since then, my `/dev/video0 `cannot be found anymore, neither in VLC nor in Google Meet or any other application that needs access to my webcam.** See this excerpt from below:

```
v4l2 debug: device Dummy video device (0x0000) using driver v4l2 loopback (version 5.4.65) on platform:v4l2loopback-000
v4l2 debug: with capabilities 0x85208000 (overall 0x85208000)
v4l2 error: not a video capture device
```

Background: I connected my Z50 mirrorless camera as a webcam via `gphoto2`: `gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 2 -f v4l2 /dev/video0`. This has worked fine up until this morning, so I suspect it has something to do with updated packages, although v4l2loopback or gphoto2 doesn't seem to have been updated. 

**Rebooting didn't help.**

Here's /dev/video0:

```
crw-rw----+ 1 root video 81, 0 Nov 10 15:30 /dev/video0
```

My user is in the `video` group, I made sure of that.

VLC reports these errors:

```
main debug: processing request item: v4l2:///dev/video0, node: Playlist, skip: 0
main debug: rebuilding array of current - root Playlist
main debug: rebuild done - 2 items, index 1
main debug: starting playback of new item
main debug: resyncing on v4l2:///dev/video0
main debug: v4l2:///dev/video0 is at 1
main debug: creating new input thread
main debug: Creating an input for 'v4l2:///dev/video0'
main debug: requesting art for new input thread
main debug: using timeshift granularity of 50 MiB
main debug: using default timeshift path
main debug: `v4l2:///dev/video0' gives access `v4l2' demux `any' path `/dev/video0'
main debug: creating demux: access='v4l2' demux='any' location='/dev/video0' file='/dev/video0'
main debug: looking for access_demux module matching "v4l2": 17 candidates
v4l2 debug: opening device '/dev/video0'
main debug: looking for meta fetcher module matching "any": 1 candidates
lua debug: Trying Lua scripts in /home/tgrossen/.local/share/vlc/lua/meta/fetcher
lua debug: Trying Lua scripts in /usr/lib/x86_64-linux-gnu/vlc/lua/meta/fetcher
lua debug: Trying Lua scripts in /usr/share/vlc/lua/meta/fetcher
main debug: no meta fetcher modules matched
main debug: looking for art finder module matching "any": 2 candidates
lua debug: Trying Lua scripts in /home/tgrossen/.local/share/vlc/lua/meta/art
lua debug: Trying Lua scripts in /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art
lua debug: Trying Lua playlist script /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art/00_musicbrainz.luac
v4l2 debug: device Dummy video device (0x0000) using driver v4l2 loopback (version 5.4.65) on platform:v4l2loopback-000
v4l2 debug: with capabilities 0x85208000 (overall 0x85208000)
v4l2 error: not a video capture device
v4l2 debug: opening device '/dev/video0'
lua debug: skipping script (unmatched scope) /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art/00_musicbrainz.luac
v4l2 debug: device Dummy video device (0x0000) using driver v4l2 loopback (version 5.4.65) on platform:v4l2loopback-000
v4l2 debug: with capabilities 0x85208000 (overall 0x85208000)
v4l2 error: not a radio tuner device
lua debug: Trying Lua playlist script /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art/01_googleimage.luac
main debug: no access_demux modules matched
main debug: creating access: v4l2:///dev/video0
main debug: (path: /dev/video0)
main debug: looking for access module matching "v4l2": 28 candidates
v4l2 debug: opening device '/dev/video0'
v4l2 debug: device Dummy video device (0x0000) using driver v4l2 loopback (version 5.4.65) on platform:v4l2loopback-000
v4l2 debug: with capabilities 0x85208000 (overall 0x85208000)
v4l2 error: not a video capture device
main debug: no access modules matched
main debug: dead input
lua debug: skipping script (unmatched scope) /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art/01_googleimage.luac
lua debug: Trying Lua playlist script /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art/02_frenchtv.luac
lua debug: skipping script (unmatched scope) /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art/02_frenchtv.luac
lua debug: Trying Lua playlist script /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art/03_lastfm.luac
lua debug: skipping script (unmatched scope) /usr/lib/x86_64-linux-gnu/vlc/lua/meta/art/03_lastfm.luac
lua debug: Trying Lua scripts in /usr/share/vlc/lua/meta/art
main debug: no art finder modules matched
main debug: changing item without a request (current 1/2)
main debug: nothing to play
  ```

Here's what's been changed according to  `cat /var/log/apt/history.log`:

```
Start-Date: 2020-11-10  08:35:49
Commandline: aptdaemon role='role-commit-packages' sender=':1.341'
Upgrade: netplan.io:amd64 (0.100-0ubuntu4~20.04.2, 0.100-0ubuntu4~20.04.3), libsystemd0:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), libsystemd0:i386 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), libasound2-data:amd64 (1.2.2-2.1ubuntu2, 1.2.2-2.1ubuntu2.1), zlib1g:amd64 (1:1.2.11.dfsg-2ubuntu1.1, 1:1.2.11.dfsg-2ubuntu1.2), zlib1g:i386 (1:1.2.11.dfsg-2ubuntu1.1, 1:1.2.11.dfsg-2ubuntu1.2), google-chrome-stable:amd64 (86.0.4240.183-1, 86.0.4240.193-1), qemu-system-x86:amd64 (1:4.2-3ubuntu6.7, 1:4.2-3ubuntu6.8), udev:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), alsa-utils:amd64 (1.2.2-1ubuntu1, 1.2.2-1ubuntu2), zlib1g-dev:amd64 (1:1.2.11.dfsg-2ubuntu1.1, 1:1.2.11.dfsg-2ubuntu1.2), libudev1:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), libudev1:i386 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), systemd-timesyncd:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), apport:amd64 (2.20.11-0ubuntu27.10, 2.20.11-0ubuntu27.11), chromium-browser:amd64 (1:85.0.4183.83-0ubuntu0.20.04.1, 1:85.0.4183.83-0ubuntu0.20.04.2), python3-apport:amd64 (2.20.11-0ubuntu27.10, 2.20.11-0ubuntu27.11), qemu-system-data:amd64 (1:4.2-3ubuntu6.7, 1:4.2-3ubuntu6.8), libnss-myhostname:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), qemu-utils:amd64 (1:4.2-3ubuntu6.7, 1:4.2-3ubuntu6.8), systemd-sysv:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), libpam-systemd:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), systemd:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), libnss-systemd:amd64 (245.4-4ubuntu3.2, 245.4-4ubuntu3.3), libasound2:amd64 (1.2.2-2.1ubuntu2, 1.2.2-2.1ubuntu2.1), libasound2:i386 (1.2.2-2.1ubuntu2, 1.2.2-2.1ubuntu2.1), libnetplan0:amd64 (0.100-0ubuntu4~20.04.2, 0.100-0ubuntu4~20.04.3), libvirt0:amd64 (6.0.0-0ubuntu8.4, 6.0.0-0ubuntu8.5), code:amd64 (1.50.1-1602600906, 1.51.0-1604600753), apport-gtk:amd64 (2.20.11-0ubuntu27.10, 2.20.11-0ubuntu27.11), linux-firmware:amd64 (1.187.3, 1.187.4), qemu-block-extra:amd64 (1:4.2-3ubuntu6.7, 1:4.2-3ubuntu6.8), python3-problem-report:amd64 (2.20.11-0ubuntu27.10, 2.20.11-0ubuntu27.11), qemu-system-common:amd64 (1:4.2-3ubuntu6.7, 1:4.2-3ubuntu6.8), libatopology2:amd64 (1.2.2-2.1ubuntu2, 1.2.2-2.1ubuntu2.1), qemu-system-gui:amd64 (1:4.2-3ubuntu6.7, 1:4.2-3ubuntu6.8)
End-Date: 2020-11-10  08:38:32
```

fa2k, 2020-11-10:
I think I have the same problem. v4l2loopback versions 12.3 and HEAD from github (91AC9A85B2E0DCA4F1005BE) tested.

I use OBS as a source for the video. It used to work in skype. Now almost nothing works: 

Loopback  device does not appear in list of outputs: Skype, Firefox, Chrome, Cheese, Zoom.
Same as above: VLC
Works (!?): Discord native app.

I have tried to change exclusive_caps to 1 and 2, change the resolution, output framerate from OBS, format YUY2, YUV420 and RGB32.

grssnbchr, 2020-11-10:
Since when has it stopped working for you? Do you think it's also because of recent updates to your system?

fa2k, 2020-11-10:
Some time within the last month, but I don't know exactly when it happened, because I thought I had messed up something. I could  get the apt log, but it wouldn't be much use since I don't know when, sorry.

umlaeute, 2020-11-10:
there's a known bug in the Ubuntu-package (see #370; which also has a solution)

since the title of *this* issue ("/dev/video0: not a video capture device [...]") is **very** similar to the title of *that* issue ("Device '/dev/vid eo1' is not a output device [...]") i would highly appreciate it if you could first check the issue tracker next time ;-)

closed 2020-11-10 by umlaeute

grssnbchr, 2020-11-11:
Thank you. I actually looked at that issue but thought it's a different one. Sorry for the circumstances.

umlaeute, 2020-11-11:
fair enough.
@grssnbchr could you still try to build `v4l2loopback` from sources (*this repository*)?
your apt-logs look unconspicious enough; but probably you did another update *before that one*?

@fa2k there's no revision "91AC9A85B2E0DCA4F1005BE" in this repository

umlaeute, 2020-11-11:
and it seems there's an updated `v4l2loopback-dkms` package in *focal-updates*. you should probably try this as well...

fa2k, 2020-11-11:
> @fa2k there's no revision "91AC9A85B2E0DCA4F1005BE" in this repository

That was the `srcversion` in modinfo, but could be related to the kernel, not the module. Anyway, I must have messed up the installation from source, because when I tried the linked solution in #370 it worked for me (version 12.5)! So I don't actually have the problem in this bug. [For anyone coming through here to #370, I'd like to emphasize umlaeute's last comment: It could have some unexpected consequences to follow the original solution, and update the universe and multiverse to the next Ubuntu version]