v4l2loopback/v4l2loopback #377: device caps not being set

omegadragon, 2020-11-12:
was comparing my pseye v4l2-ctl output vs v4l2loopback output
[pseye.txt](https://github.com/umlaeute/v4l2loopback/files/5530380/pseye.txt)
[loopback.txt](https://github.com/umlaeute/v4l2loopback/files/5530381/loopback.txt)
the pseye works in latest chromium and Firefox were v4l2loopback does not and i found v4l2loopback is not setting "Video Capture" in device caps and im assuming the 2 browsers are looking at that setting to see if its capable ignoring "Capabilities" or comparing them. 

v4l2-ctl --device=/dev/video0 --all     was the command i used to check.

umlaeute, 2020-11-12:
- try with current `master`
- *how* have you loaded the module?
- have you attached a producer to the loopback device?

omegadragon, 2020-11-13:
Thanks for your reply!
I got it working it was a old copy being persisted through reboots conflicting with the new one. so i just went manually purging v4l2loopback.ko till it didn't load the driver after a reboot and put a compiled master in as a test it worked!  so i deleted its .ko file and tried the Ubuntu dkms 12.5 one in a newer release and it seems to work too XD.

closed 2020-11-13 by omegadragon