v4l2loopback/v4l2loopback #378: Not working for everything except webcamoid

PQCraft, 2020-11-12:
I built everything from source and it refuses to work:
I did these steps in order to get OBS to output video
`sudo modprobe -rv v4l2loopback`
`sudo modprobe v4l2loopback devices=1 video_nr=2 card_label="VCAM" exclusive_caps=1`
It makes /dev/video2
`v4l2-ctl --list-devices` produces:
```
VCAM (platform:v4l2loopback-000):
	/dev/video2

VGA WebCam: VGA WebCam (usb-0000:00:12.0-1.4):
	/dev/video0
	/dev/video1

```
Then I go on zoom and the only option there is `VGA WebCam: VGA WebCam`.
Then I went to firefox and opened `https://www.onlinemictest.com/webcam-test/`
The 'Will you allow www.onlinemictest.com to use your camera?` dialog pops up and in the dropdown box, there is only one option: `VGA WebCam: VGA WebCam`
IDK what I did wrong and I am frustrated and confused.

PQCraft, 2020-11-12:
Forgot to add but after I ran `sudo modprobe v4l2loopback devices=1 video_nr=2 card_label="VCAM" exclusive_caps=1` I started OBS and v4l2sink. Everything on that side works though because when I open Webcamoid and manually make it read video off of /dev/video2 it works.

PQCraft, 2020-11-12:
It just doesn't work with anything I need

umlaeute, 2020-11-12:
sorry, but this is not a support forum, but an issue tracker.

it seems that you tested exactly two other applications, which hardly counts as "everything".
and if it's working with webcamoid, then the v4l2loopback device is obviously working.

there are known restrictions with many browsers, as to which colorspaces and resolutions are supported.

so i'm closing this.

closed 2020-11-12 by umlaeute

PQCraft, 2020-11-12:
Oof. Welp... it doesn't work so i'm just going to remove it as it's bloat at this point.