v4l2loopback/v4l2loopback #380: clarify supported kernel versions in the README

dpocock, 2020-11-15:
The README in the repository mentions that 5.0.0 "should work" and 4.0.0 "should work".  These statements may be a little bit too broad.  Can you please think about providing a more specific list of permutations that have been tested and confirmed to work?

For example, each time you make a release, could you include a note about the kernel version you were personally using before tagging the release?

For me, the following combination works:

Debian buster / kernel 4.19.132-2 / v4l2-loopback 0.12.1-1

Using the backports kernel 5.8.10-1 I observed the following:
- v4l2-loopback 0.12.1-1 from package on buster (stable) fails to compile
- building a backport of 0.12.5-1 package from sid, it compiles successfully, the module loads but it does not work.  I did not have time to look for the root cause but I may test it again in the next few weeks and try to provide more diagnostics.

As time goes by, more and more people will try to use the backports kernels.  The goal of having this module in the kernel source tree (issue #268) may help to make that easier.

umlaeute, 2020-11-15:
if you use the backports kernel, you probably should also use the [backports version of `v4l2loopback-dkms`](https://packages.debian.org/buster-backports/v4l2loopback-dkms), which should work nicely with the kernel from backports.

but about the original request: i don't think it is feasible to do an exhaustive list of supported kernel versions (and variants thereof!).
i don't even have the resources to provide a list of supported distribution releases (which would reduce the number of linux versions significantly).

otoh, if a distribution v4l2loopback, i think it is their responsibility to ship a version of `v4l2loopback` that works with the shipped kernel. (which is true for both Debian/buster and Debian/buster-backports - you just have to make sure that you install both the kernel and the v4l2loopback-dkms from the same).

anyhow: if you think a "specific list of permutations that have been tested and confirmed to work" is necessary and/or useful, i think the only way to go forward is to engage the community to collect this information.
i would suggest that you start such a matrix of supported kernel versions vs v4l2loopback versions on the https://github.com/umlaeute/v4l2loopback/wiki

umlaeute, 2025-04-02:
>  I meant if the building process on Fedora, which I mentioned, is documented somewhere

no it's not documented. i do not have access to Fedora machines (and am not knowledgeable about that ecosystem).

but feel free to provide instructions in the [wiki](https://github.com/v4l2loopback/v4l2loopback/wiki) :-)

(as a sidenote: "*`dkms build -m v4l2loopback -v 0.14.0` fails.*" is not very useful, as far as bug-reporting goes; typically you should provide the actual output. you probably omitted this, in order to not fully hijack an unrelated issue)