v4l2loopback/v4l2loopback #381: can't parse device using v4l2loopback-ctl set-caps /dev/video0 "UYVY:640x480"

lukelu0520, 2020-11-20:
Dear v4l2loopback friends,

We stuck on following the readme in the forcing format: `v4l2loopback-ctl set-caps /dev/video0 "UYVY:640x480"`

The error message is `can't parse device`. We double checked the installation of v4l2loopback-ctl and /dev/video* number, but can not solve the issue.

Did anyone get the same issue before? Hope to hear your expertise.

Best,
-- Luke

umlaeute, 2020-11-20:
please read the README that comes with the version of `v4l2loopback-ctl` that you are using.

the [documentation n the repository](https://github.com/umlaeute/v4l2loopback) refers to the version of the tools found in the repository.

in your case: the `v4l2loopback-ctl` utility has learned a couple of new tricks (that accompany some new tricks of the kernel module itself) which are now documented in the README, like simpler non-gstreamer caps.
the version of the tools you are using on your computer, is most likely too old to know these tricks..

as if 11.11.2020, there hasn't been a *release* of `v4l2loopback` and utils yet, that know how to deal with the new-style caps (and dynamic device management)

closed 2020-11-20 by umlaeute

robertlzj, 2022-06-29:
Hi, @umlaeute  I suffer the same problem.
It seems `v4l2loopback-ctl set-caps` is _another_ command with the same name, _besides_ this repository.
In its copyright from [man](https://manpages.org/v4l2loopback-ctl#copyright), shows it belongs to IOhannes m zmoelnig <zmoelnig@iem.at>, and Anton Novikov (random.plant@gmail.com).