v4l2loopback/v4l2loopback #382: unable to open control device '/dev/v4l2loopback': No such device

aquaticom, 2020-11-29:
### v4l2loopback version v0.12.5 (non-dkms)
Using v4l2loopback  compiled from github sources to be the latest version 
Built using instructions from the readme, but running _make install-all_ instead of _make install_ to install v4l2loopback-ctl utility alongside the module 
The commands ran were 

> $ make && sudo make install-all
> $ sudo depmod -a

This problem involves v4l2loopback-utils, they are also version v0.12.5, built from the same source.
### Kernel version 5.8.0-29-generic 31-Ubuntu
### Distribution Ubuntu 20.10 Groovy Gorilla, running as a virtualbox guest
### The problem:
Dynamic device management fails
_$ sudo v4l2loopback-ctl add -n "Webcam 1" /dev/video1_ fails with message:
unable to open control device '/dev/v4l2loopback': No such device

Steps to replicate:

- Build and install non-dkms v4l2loopback with its utilities on Ubuntu 20.10

- _$ sudo v4l2loopback-ctl add -n "Webcam 1" /dev/video1_

- It succeeds at first

- After some time and doing ??? (maybe rebooting) load the module again and run _$ sudo v4l2loopback-ctl add -n "Webcam 1" /dev/video1_

- It will give error message _unable to open control device '/dev/v4l2loopback': No such device_

Additional info: $ ls /dev/v4l2loopback shows a file by that name exists regardless if v4l2loopback module is loaded or not
 
The problem appeared a day or few after I built the module. At first v4l2loopback worked and I was able to dynamically make a v4l2loopback device. 

Working inside of Virtualbox, I cloned a working backup VM when I first built v4l2loopback and successfully added a dynamic video device with v4l2loopback-ctl. After I got the error message, I booted the backup clone and it worked the first time, but after a reboot and ??? (some unknown cause) gave the same error message. 

In between v4l2loopback working and breaking I was getting the output of _$ v4l2ctl --list-devices_  with a python script, using the subprocess module. 

> subprocess.run(['v4l2-ctl','--list-devices'], capture_output=True, check=True)

Maybe v4l2-ctl "locked" the /dev/v4l2loopback device file? The file now exists whether the module is loaded or not, but this may be unrelated. I've since rebooted multiple times and no running processes are interfering with v4l2loopback. 

In the time between v4l2loopback working and breaking, I also installed Atom through snap package manager. No kernel update happened in this time. I rebooted the machine a few times in this period. I haven't streamed or read any video to and from the v4l2loopback devices on this machine yet. 

I can't think of anything strange I could have done in this in-between time to cause this bug. I think it's spontaneous breakage.

a suggestion online on a Stack Overflow question about missing device files said to run 
$ sudo udevadm monitor to check udev events, it may help
Here is udevadm monitor output when I run modprobe v4l2loopback

> KERNEL[2025.310214] add      /module/mc (module)
> UDEV  [2025.311443] add      /module/mc (module)
> KERNEL[2025.311528] add      /bus/media (bus)
> UDEV  [2025.311873] add      /bus/media (bus)
> KERNEL[2025.315014] add      /module/videodev (module)
> UDEV  [2025.316473] add      /module/videodev (module)
> KERNEL[2025.316511] add      /class/video4linux (class)
> UDEV  [2025.316863] add      /class/video4linux (class)
> KERNEL[2025.317084] add      /module/v4l2loopback (module)
> UDEV  [2025.317448] add      /module/v4l2loopback (module)
> KERNEL[2025.317503] add      /devices/virtual/misc/v4l2loopback (misc)
> UDEV  [2025.318159] add      /devices/virtual/misc/v4l2loopback (misc)
> KERNEL[2025.318274] add      /devices/virtual/video4linux/video0 (video4linux)
> UDEV  [2025.319346] add      /devices/virtual/video4linux/video0 (video4linux)

And udevadm monitor output when I run modprobe -r v4l2loopback

> KERNEL[2209.615833] remove   /devices/virtual/video4linux/video0 (video4linux)
> UDEV  [2209.617216] remove   /devices/virtual/video4linux/video0 (video4linux)
> KERNEL[2209.617381] remove   /devices/virtual/misc/v4l2loopback (misc)
> UDEV  [2209.617769] remove   /devices/virtual/misc/v4l2loopback (misc)
> KERNEL[2209.618002] remove   /module/v4l2loopback (module)
> UDEV  [2209.618320] remove   /module/v4l2loopback (module)
> KERNEL[2209.618573] remove   /class/video4linux (class)
> UDEV  [2209.618894] remove   /class/video4linux (class)
> KERNEL[2209.619104] remove   /module/videodev (module)
> UDEV  [2209.619415] remove   /module/videodev (module)
> KERNEL[2209.619590] remove   /bus/media (bus)
> KERNEL[2209.619667] remove   /module/mc (module)
> UDEV  [2209.620086] remove   /bus/media (bus)
> UDEV  [2209.620645] remove   /module/mc (module)

I hope to resolve this to use v4l2loopback on Ubuntu 20.10 with Dynamic Device Management.

Thank you. ❤️

aquaticom, 2020-11-29:
### Update:
I did additional testing.
I made a new virtual machine running up to-date Ubuntu 20.10 with VirtualBox Guest additions, and made a snapshot of its clean state.

Then in a testing VM I first installed v4l2loopback-dkms and v4l2loopback-utils from the Ubuntu Repositories, and found a _different_ issue. 
### Ubuntu 20.10 repositories have an up to-date v4l2loopback-dkms v.0.12.5 kernel module and a outdated v4l2loopback-utils v.0.6 This mismatch needs to be fixed by Ubuntu Maintainers.

This version mismatch did not cause the problem I wrote this issue about.

Ignoring the version mismatch, I narrowed down my issue with the _unable to open control device '/dev/v4l2loopback': No such device_ error by doing snapshot testing:

In a clean Ubuntu 20.10 install, I built the v4l2loopback module and utils with _make install-all_ from the github source, and dynamic device management worked correctly. I saved a snapshot of the running VM, and hypothesized that rebooting is what causes the issue. This turned out correct.

I rebooted from the working state multiple times, and every time the module broke and gave the error this issue is about.

### v4l2loopback (non-dkms) v.0.12.5 on Ubuntu 20.10 works in the same session you build it, but breaks after shutdown or reboot. 

However, running modprobe -r v4l2loopback, then modprobe v4l2loopback in the same session the module was built in and loading and re-loading the module doesn't cause issues. Something breaks when you reboot or shutdown the system.

aquaticom, 2020-11-29:
### Update 2:
Tested VM's with Ubuntu 20.10 and Debian 10.6 _without_ vbox Guest Additions. 
Same procedure, same issue:
Works at first, then breaks after a reboot.

Not using Guest Additions or using Debian 10.6 (kernel 4.19) doesn't correct the behavior. Very weird.

umlaeute, 2020-11-29:
interesting. admittedly, the code has not been very well tested with reboots ;-)
otoh, i have of course rebooted my development machine (running Debian/sid) a couple of times, and i don't have such problems.

so: what's that `/dev/v4l2loopback` file you get after the reboot (which - according to your report - is clearly **not** a device-file created by the module as it should be)?

>      $ ls /dev/v4l2loopback 
> 
> shows a file by that name exists regardless if v4l2loopback module is loaded or not

for starters: what's the output of:

~~~sh
ls -l /dev/v4l2loopback
~~~

how do you load the loopback module?
are you sure that the correct version of the module is loaded?

umlaeute, 2020-11-29:
fired up my slightly outdated Debian/testing virtualbox VM:

~~~sh
$ uname -a
Linux debian 5.7.0-1-amd64 #1 SMP Debian 5.7.6-1 (2020-06-24) x86_64 GNU/Linux
~~~

- compiled the module, ran `make install-all`.
- autoload module:

~~~sh
$ uname -a
Linux debian 5.7.0-1-amd64 #1 SMP Debian 5.7.6-1 (2020-06-24) x86_64 GNU/Linux
$ make
[...]
$ sudo make install-all
[...]
$ echo v4l2loopback > sudo /etc/modules-load.d/v4l2loopback.conf
v4l2loopback
$ sudo depmod -a
$ sudo modprobe v4l2loopback
$ sudo v4l2loopback-ctl add -n "fudel" /dev/video11
/dev/video11
$ ls -lha /dev/video11 
crw-rw---- 1 root video 81, 3 Nov 29 18:19 /dev/video11
~~~

immediately after a reboot:

~~~sh
$ sudo v4l2loopback-ctl add -n "fudel" /dev/video11
/dev/video11
$ ls -l /dev/video11 /dev/v4l2loopback 
crw------- 1 root root 10, 61 Jul  3 12:02 /dev/v4l2loopback
crw-rw---- 1 root video 81, 3 Nov 29 18:19 /dev/video11
~~~

aquatic0m, 2020-11-30:
Locked myself out of my github account. Oops. Had to make a new one.

So with the module unloaded
_ls -l /dev/v4l2loopback_ shows: 
> crw------- 1 root root 10, 255 Nov 30 12:37 /dev/v4l2loopback

The creation timestamp 12:37 is the time I booted the system, but running _$ sudo dmesg | grep v4l2loopback_ shows nothing.

/dev/v4l2loopback seems to be always created on bootup, but the v4l2loopback driver doesn't get autoloaded

I load the module with _$ sudo modprobe v4l2loopback_
When I load it _$ sudo dmesg | grep v4l2loopback_ gives the output
> [ 1453.185292] v4l2loopback driver version 0.12.5 loaded

This makes me think I have the right version

also, _v4l2loopback-ctl -v_ shows
> v4l2loopback-ctl v0.12.5

I am not autoloading the module, maybe that makes a difference?

aquatic0m, 2020-11-30:
### Mostly figured out current issue in this comment

### I found a workaround and an important detail
After running _$ sudo rm -rf /dev/v4l2loopback_ with the module unloaded, 

$ sudo modprobe -r v4l2loopback
$ sudo rm -rf /dev/v4l2loopback
$ sudo modprobe v4l2loopback
$ sudo v4l2loopback-ctl add -n "Webcam 1" /dev/video1

> /dev/video1

Force-deleting /dev/v4l2loopback seems to solve this issue. What is this erroneous /dev/v4l2loopback file if not the device file?

**^ Success** 
Testing what happens when you unload the module in the same session:
$ sudo modprobe -r v4l2loopback
$ ls -l /dev/v4l2loopback

> ls: cannot access '/dev/v4l2loopback': No such file or directory

**/dev/v4l2loopback gets successfully deleted when you modprobe -r the module in the same session**
After this I tested loading and unloading the module a few times and running v4l2loopback-ctl add .. when it's loaded. Loading and unloading the module in the same session isn't a problem

### Testing rm -rf /dev/v4l2loopback and rebooting the system
$ sudo rm -rf /dev/v4l2loopback
$ ls -l /dev/v4l2loopback

> ls: cannot access '/dev/v4l2loopback': No such file or directory

$ reboot 

(system reboots)

Right after bootup:
$ ls -l /dev/v4l2loopback 

> crw------- 1 root root 10, 255 Nov 30 **13:42** /dev/v4l2loopback

The file's creation timestamp is the minute I turned on the system. The file wasn't there when I shut down the system. It gets created on bootup.

From here, if I run _$ sudo rm -rf /dev/v4l2loopback_ again, then _$ modprobe v4l2loopback_, I can successfully use Dynamic Device Management

### The issue disappears when enabling automatic module loading _after_ running rm -rf 
**(or running rm -rf from any state without an erroneous /dev/v4l2loopback file, like the state which you likely had on your Debian system when building the module. You then enabled automatic module loading in the first session. The error condition is rebooting a system without enabling automatic module loading.)**

So on a broken system:

- Unload v4l2loopback 
- rm -rf /dev/v4l2loopback
- Enable automatic module loading for v4l2loopback
(write v4l2loopback into _/etc/modules-load.d/v4l2loopback.conf_ )
- Reboot
- Now the started system has v4l2loopback loaded
- The /dev/v4l2loopback dev file is no longer **erroneous** 
- Dynamic device management now works
- Rebooting n-times from this point no longer creates an erroneous /dev/v4l2loopback, no further intervention is needed

### To replicate my error on any system:
- make install-all
- modprobe v4l2loopback
- reboot

The system is now broken because it does not autoload v4l2loopback.

umlaeute, 2020-11-30:
confirmed, with (for the sake of easier pasting the script):

~~~sh
make install-all
depmod -a
modprobe v4l2loopback
reboot
~~~

bjmuld, 2021-01-20:
I experienced this issue, but in my case, I had several successful reboots before the issue.  Also, automatic module loading was enabled from the start...

bjmuld, 2021-01-21:
@umlaeute I'm experiencing this issue persistently now, and it cannot be resolved with manual removal of `/dev/v4l2loopback` followed by a reboot.
Ideas?

closed 2021-01-27 by umlaeute

wangjia184, 2024-08-27:
I am encountering this issue if try to install via DKMS while `modprobe` works.

Ubuntu 22.04 Kernel 6.8

umlaeute, 2024-08-28:
> Ubuntu 22.04

which comes with v4l2loopback-0.12.7 (which does **not** come with dynamic device support nor a `/dev/v4l2loopback` device).

you can:
- ~~install a new version of Ubuntu~~ (actually no; it seems that Ubuntu has not synched with the latest and greatest v4l2loopback from Debian yet)
- install an up-to-date version of `v4l2loopback` from *this repository* (rather than your package manager)
- live with the limitations of the version you installed.

wangjia184, 2024-08-28:
@umlaeute  The error is from the version built from the main branch of this repostory.

umlaeute, 2024-08-28:
@wangjia184 
- which error (exaxctly)? 
- and what's your output of
   ```sh
  dmesg | grep v4l2loopback
   ```
- and what's the output of
   ```sh
  ls -l /dev/v4l2loopback
   ```

wangjia184, 2024-08-28:
@umlaeute v4l2loopback was loaded by DKMS successfully,  but `/dev/v4l2loopback` does not exist.  

I gave up DKMS and use modprobe instead now

wangjia184, 2024-08-30:
@umlaeute here is what I got 
```
$ ls -l /dev/v4l2loopback
ls: cannot access '/dev/v4l2loopback': No such file or directory
```

umlaeute, 2024-08-30:
Then you are not loading a recent (>=0.13) version of the `v4l2loopback` module, but some older one.